sinara-hw/Booster #2: Booster v1.0 Testing

hartytp, 2018-04-12:
@gkasprow Booster received and looking beautiful.

One observation before powering it one: it's *heavy*. I wonder if we could slightly reduce the weight without compromising performance. That might help with some mechanical issues.

gkasprow, 2018-04-12:
The wall thickness of machined stuff is 4 mm. We can go for thinner ones, but manufacturing precision would need to be far better, thus more expensive. This is linked to needed tolerances required by screws holding the cover.
We can gain slightly on the sinks - the radiators can be thinner and here some reduction could take place.

hartytp, 2018-04-12:
The weight isn't a huge deal - it's no heavier than any other boat anchor like an 8656b. Comment is just that if there are any easy places to shave off some weight we should go for it. Don't do it if it increases costs

hartytp, 2018-04-12:
The heatsink look like they could be a bit smaller without problems but I doubt that will save much money or weight and could result in higher fan speeds needed (more noise and wear)

hartytp, 2018-04-13:
@gkasprow what's the expected behavior of this unit at startup?
- Powering it on, the fans start off quite loud (I assume running at full speed) and all LEDs are off.
- The fan then spins down over a few seconds and completely stops, all LEDs remain off
- It doesn't respond to either of the buttons on the FP

hartytp, 2018-04-13:
Aah, maybe I was just a bit impatient. The standby button does give me green lights after a few seconds.

hartytp, 2018-04-13:
Fan control loop sounds like it might be oscillating a bit with no load on the amps. Gives my office a slightly haunted sound of fans gently moaning to themselves.

gkasprow, 2018-04-13:
@hartytp we can switch to software fan control loop. The MCU can read the temp and them set fixed PWM pattern. At the moment the temperature control is done entirely by the Maxim chip. It is known by its haunted behaviour. I like this expression :)
@wizath please have a look on standby button behaviour.

hartytp, 2018-04-16:
Small-signal S21:

- Measured for channel 0 using a stimulus power of -30dBm

[image]

hartytp, 2018-04-16:
And S11 (no S22 as I was using a 30dB attenuator to protect the VNA).

[image]

hartytp, 2018-04-16:
@gkasprow Are these what you expect?

Looks really nice: peak gain is approx 37.5dB at 100MHz, 36dB at 500MHz and still above 34dB at 750MHz.

gkasprow, 2018-04-16:
@hartytp we observed smilar measurements at WUT.
btw, what about firmware update.
At the moment to update the FW one need to install the jumper on the main board, but it is not really practical. Another idea is dedicated hidden switch on the front panel activated by a needle.
Yet another idea is to enter the bootloader using software command, but this need to be verified.
What method is the most convenient for you?

hartytp, 2018-04-16:
> @hartytp we observed smilar measurements at WUT.

Thanks for confirming @gkasprow.

47.5dB of gain seems a little high to me. Can we add some attenuation somewhere to reduce this to 40dB peak gain, please?

> Yet another idea is to enter the bootloader using software command, but this need to be verified.

This is definitely the most convenient. That way, we can write a script that automates the updating with minimal intervention from the user.

hartytp, 2018-04-23:
Small-signal S21 for all 8 channels using a -30dBm stimulus.

- Apart from channel ?, the channel-channel variation looks good. 
- Gain flatness looks good across the specified bandwidth, but a few resonances outside the specified bandwidth (locations vary significantly between channels).

[image]

[image]

hartytp, 2018-04-23:
Small-signal cross-talk:
- stimulus is -20dBm (approx +27dBm output)
- Drive channel 0 and look at RF output from channel *n*, with all unused outputs/inputs 50Ohm terminated.
- Data is normalized by dividing by the gain of the output channel (S21 from previous data set), so that numbers represent the equivalent input signal that would have to be applied to a channel to produce the measured output.

[image]

[image]

hartytp, 2018-04-23:
@gkasprow Are the numbers above consistent with your measurements?

hartytp, 2018-04-23:
@gkasprow Cross-talk looks a little worse than in the previous version. Is this expected?

hartytp, 2018-04-23:
Since the cross-talk seems to be the same on each channel, my guess is that the cross-talk is limited by the power distribution network or I2C/other control signals. Let's check that we've really filtered everything entering/leaving the RF enclosures properly on both sides of the connectors.

gkasprow, 2018-04-23:
In the nev revision I added LC filters on all supplies and additional grounding holes close to the DSUB connectors. This whould help.

hartytp, 2018-04-26:
Linearity/large-signal performance testing using a synth and spectrum analyzer (40dB input attenuator) with a simple python script to automate things. Looking at channel 0.

[image]

Comments:
- I see a 1dB glitch at around -8.7dBm input. This is reproducible and I do not believe it is an artefact of my measurement setup.
- Small signal gain is +47dB
- P1dB is 29.7dBm
- P3dB is 34.5dBm
- Need to fit to find IP3, will do tomorrow.

The glitch concerns me. Do you have any thoughts about that @gkasprow 

P1dB is also lower than we originally agreed, but I think that's okay. Essentially, this device just saturates quite softly. The AOMs I will drive with it are far more non-linear than Booster, that doesn't concern me.

gkasprow, 2018-04-26:
You can check the power supply at the module entry. Maybe it enters overcurent mode. There should be diagnostic output at the VUART - measurement of module current.

gkasprow, 2018-04-26:
U can check if the current is increasing with input power.

hartytp, 2018-04-26:
Thanks Greg.

Let me double check that there is nothing going wrong with my data collection/analysis first, then I'll do that.

hartytp, 2018-04-26:
@gkasprow Okay, I think this is real.

At lower powers (-15dBm input for example) everything behaves as I expect.

When I apply -9dBm input, the output power from the amplifier drifts by about 1dB over a second or so after I apply the RF. 

I do not see this if I remove the attenuators and connect the synth directly to the spectrum analyzer, so I suspect it is real.

Could you ask Maceij/Tomek if they can try to reproduce this behaviour, please?

gkasprow, 2018-04-26:
I already did.

hartytp, 2018-04-27:
IIP3=34dBm, measured with a single input tone.

[image]

hartytp, 2018-04-27:
So, the only visible spur is -95dBc at +-210kHz. I assume that's the SMPS.

That's already excellent. But, in any case, it will probably wind up a bit lower down once the extra power filtering is added for the next version.

hartytp, 2018-04-27:
No obvious change in the spectrum when I increase the input power to -8.5dBm.

gkasprow, 2018-04-27:
It looks like thermal or power supply issue.
Did you see such behaviour on 2 channel prototype?

hartytp, 2018-04-27:
Agreed, it's one or the other.

We couldn't test this on the previous prototype because there was a different issue with the power supply.

jordens, 2018-04-27:
That's a weird way to determine IP3. Wouldn't you usually just measure the powers at **one** small signal and then solve for IP3 using the (nicely visible) 30dB/10dB slope?
The way you do it and how you rely on the flattening of the 3rd harmonic, you'd probably want to plot a "power dependent IP3" (some "IP3 nonlinearity"). The way you do it (and if you'd plot a power dependent IP3), you are preventing meaningful calculations and design using the IP3 (those assume the 30dB/10dB slope).
That IP3 would then also be frequency dependent.
C.f. also how they characterized the USRP daughterboards.

hartytp, 2018-04-27:
@gkasprow FWIW, I don't think this is thermal.
- Below about -9dBm input, I do not see this behaviour at all, and above it I do. The transition between working and not working is very sharp and reproducible (see the graphs I plotted above). If this were a thermal issue then I would expect a more gradual transition (unless this is something odd like the fan controller causing issues)
- There is no hysteresis, which I would expect due to the finite thermal mass of things. If I switch from high to low powers, I go straight back to the right output power
- When I go to high power, the output power holds at the right level for 1s or so, then ramps to a new value over 1s or so. Again, that doesn't feel thermal...

hartytp, 2018-04-27:
> That's a weird way to determine IP3.

Yes! This was a really quick hack while I waited for something else, without too much thought. It's just preliminary testing, I'd take it all with a pinch of salt.

> Wouldn't you usually just measure the powers at one small signal and then solve for IP3 using the (nicely visible) 30dB/10dB slope?

I guess so. Usually, I wouldn't expect the slope of the third harmonics to change so much, even at relatively low input powers, which is why I didn't do that. The saturation behaviour of this amp is a bit different to the case I'm used to thinking about.

Anyway, doing it the way you suggest (which is probably correct), I get IIP3=+1.3dBm

[image]

gkasprow, 2018-04-27:
@hartytp Can you do simple measurement please?
Open the upper cover. Solder piece of wire and connect DMM. Observe voltage on the components I marked during transition
[image]

hartytp, 2018-04-27:
Okay, will do.

Do you think we can change the design a tiny bit, so that the screws holding the module lids can be removed without removing the whole module? ATM the lip at the front of the chassis blocks 3 screws.

hartytp, 2018-04-27:
And these recessed SMA nuts are a real PITA to remove!

gkasprow, 2018-04-27:
Could you look with a scope at this 5V? Is there just a drop or it starts oscillating?
I added all the issues to the todo list.

gkasprow, 2018-04-27:
the headers will be glued with thermal glue during soldering.

gkasprow, 2018-04-27:
the 5V regulator is 1.5A. I'm sure the module does not consume that much.
Could you measure if the 6.5V is there during the issue?
[image]

gkasprow, 2018-04-27:
@hartytp I think I know what is going on. There should be yet another capacitor at the LDO output. Now there is one after the 0.1Ohm resistor but this may be not enough.
Can you solder 10uF 0805 or 1206, at least 10V capacitor between regulator pin 3 and 4? I marked it on the drawing below.
[image]

hartytp, 2018-04-27:
> I added all the issues to the todo list.

Thanks Greg! They're all very small issues -- as I said, generally, I think the mechanics for this is beautifully done -- but they're worth fixing for the final version and I wanted to write them down before I forgot.

> @hartytp I think I know what is going on. There should be yet another capacitor at the LDO output. Now there is one after the 0.1Ohm resistor but this may be not enough.
Can you solder 10uF 0805 or 1206, at least 10V capacitor between regulator pin 3 and 4? I marked it on the drawing below.

Thanks for the suggestion.

I took that data in a hurry because I had to leave the lab. I'll retake it on Monday and try your suggestion.

hartytp, 2018-04-27:
PS when you have time, could you make sure that all the latest Altium files are uploaded? I have Altium viewer, so this will help debugging.

gkasprow, 2018-04-27:
all is here https://github.com/sinara-hw/Booster/releases/tag/rev1.0

gkasprow, 2018-04-28:
there will be metal spacers used that will help blocking the RF path between modules.

hartytp, 2018-04-30:
@gkasprow Please can we have some test points with silk screen labels on the bottom layer of the top PCB in the RF modules. This would be useful for probing the signals coming into these modules (e.g. putting a scope on the bias voltage/power measurements).

hartytp, 2018-04-30:
@gkasprow Out of curiosity, do we really need 2 separate PCBs in the RF modules? AFAICT, the top PCB has almost no components and basically just routes signals to the bottom (RF) PCB. Wouldn't it be cheaper/easier to just put the D-type connection directly on the RF PCB and remove the top PCB. This would also allow us to remove the board-board connectors (which are a bit fiddly to assemble at the moment) and generally make the whole module simpler and easier to assemble (particularly since the top PCB has some complex cutouts to make things fit together).

This might require additional layers on the RF PCB, but I assume that 1 board with more layers is cheaper than two boards + board to board connectors (particularly once assembly costs are included).

hartytp, 2018-04-30:
Anyway, I'm fine with keeping both boards if you don't think it's worth scraping one, I'm just curious about the current design.

gkasprow, 2018-04-30:
@hartytp well, everything is possible and every design can be optimized. Current design was done in such way to keep 2U enclosure possible with horizontal airflow. If we move the RF module to top, then the lid will need to evacuate the heat which would increase the thermal resistance a lot. Another problem is routing of control and supply lines over the RF board which would make it more expensive. As you wrote, it would have to be multi layer board.
Current construction makes replacement of modules relatively easy and several design choices were made to make it possible. 
Yet another idea is D-sub assembled on opposite side and plugged to the main board from the other side (bottom). This would require designing of the mechanics and electronics from scratch and another one or two prototype round - the main board would have to be lifted, supplies mounted upside down.
I'm not sure if it is worth an effort.
Current design is quite well debugged and most issues were identified in first revision. I'm just finishing fixing issues and second revision goes to production.
ACK for test points

hartytp, 2018-04-30:
>  I'm just finishing fixing issues and second revision goes to production.

👍 

You're right: the current situation is absolutely fine. Let's stick with it, since changes would require a redesign without reducing manufacturing costs enough to justify it.

> If we move the RF module to top, then the lid will need to evacuate the heat which would increase the thermal resistance a lot.

That's not quite what I meant. I was thinking of changing the mechanical design, so that the RF board has the D-type on it, but still mounts on the bottom of the enclosure, rather than the lid. I haven't looked at the CAD models, so maybe this doesn't work.

> Another problem is routing of control and supply lines over the RF board which would make it more expensive. As you wrote, it would have to be multi layer board.

Out of curiosity, is having a pair of 2-layer boards, along with accompanying board-board connectors and the required cutouts in the top board, cheaper than a single 4-layer board?

hartytp, 2018-05-01:
Looking on the D-type in the RF module with a scope probe (cover removed) I see 2.4V on the REFL power pin, with input at -8dBm and the output connected directly to a 10W 30dB attenuator. That seems like far too much reflected power for this configuration, doesn't it?

hartytp, 2018-05-01:
Forward power pin on the D-type while toggling the RF on/off.
[image]

hartytp, 2018-05-01:
aaah, I see the N8V0_CH is a negative bias for the FET. It's not actually a -8V regulator. Well, in that case, maybe -1.2V is correct.

gkasprow, 2018-05-01:
Yes, -1.2 is correct. Previous amplifier design required higher negative voltage.

gkasprow, 2018-05-01:
4 layer board is much more expensive than two 2-layer boards.

hartytp, 2018-05-01:
Looking at the D-type with a DVM:
- 28V rail is at 27.5V, with <=10mV changes as I turn the RF on/off (no transients that I could see)
- Forward power is about 3.44V, decreasing by about 100mV during the transients, offset with power off is about 125mV
- Preamp voltage is 4.885V, with a few mV change as I turn the RF on/off
- Reverse power is 2.5V, decreasing by 100mV during the transient, offset (RF off) is about 50mV
- FET bias is -1.218V with no obvious glitches

@gkasprow is there anything else you want me to check, or is this something that that would be better for Maciej to look at once your university reopens?

hartytp, 2018-05-01:
> 4 layer board is much more expensive than two 2-layer boards.

Okay, good to know.

hartytp, 2018-05-01:
FWIW, the other thing about the current 2-PCB arrangement that's a bit annoying is that one can't access/probe the RF PCB while it's assembled in the enclosure. This makes testing/debugging a bit trickier.

gkasprow, 2018-05-01:
To do debugging, you can connect only power supplies to the pins. It is simple circuit and not much is needed to make it working. You can also insert long goldpin header between and lift the upper board a few centimetres.

gkasprow, 2018-05-01:
You can also do simple trick - use 10 pin IDC with piece of ribbon cable and goldpin header inserted in the RF board. In this way you can place the upper board far away and have full access to the bottom board.

hartytp, 2018-05-01:
@gkasprow True. I'll do something like that.

hartytp, 2018-06-06:
Closing this issue, as Booster has had a few changes so I'll open a new issue for testing Booster v1.0.1 when I get my hands on it.

closed 2018-06-06 by hartytp