sinara-hw/Booster #6: Booster: front panel SMAs

hartytp, 2018-04-23:
The exposed thread on the FP SMAs seems a bit short. When I tighten up connectors against the FP, the connectors touch the nuts. When I loosen the connectors, the nuts often come off with them!

Probably needs a longer SMA thread and a star washer to hold them in place. A proper D-shaped cut-out might be better.

hartytp, 2018-04-23:
Also, having the nuts for the SMAs recessed into small pockets makes them a bit tricky to tighten/loosen.

hartytp, 2018-04-23:
I see, there is an internal SMA nut that should stop the connector from rotating with respect to the PCB. However, on at least the channel I'm looking at, there is no star washer on the SMA cable, which meant that it came loose very easily, and started rotating when I removed an external SMA connector.

gkasprow, 2018-04-23:
Well, I tried hard to find SMA with longer thread and it is nearly impossible.
That's why I made such cutouts to use what I managed to find on stock.

gkasprow, 2018-04-23:
I asked Technosystem, maybe they are able to get longer threads.

hartytp, 2018-04-23:
@gkasprow Okay, that makes sense.

The current situation still feels a bit mechanically dodgy, so I'd like it if we could do something to make it better.

One thing would be to remove the internal SMA nut and move the RF enclosure as close as possible to the chassis.

hartytp, 2018-04-23:
@gkasprow If it doesn't increase the cost too much, another option is to have cutouts in the chassis front pannel for the RF modules. That way, the face of the module is directly exposed to the user, so the SMA doesn't need to go through two pieces of metal.

That might be better/cheaper than finding exotic long SMAs.

gkasprow, 2018-04-23:
Yes, this can be done easily. But then we have to add some screws to attach the module to the front panel somehow.

gkasprow, 2018-04-23:
The module wall is very thin now. 0.75mm or so.

gkasprow, 2018-04-23:
not at all. The cutouts are already D-shaped. At least I designed them in such way.

hartytp, 2018-04-23:
> not at all. The cutouts are already D-shaped. At least I designed them in such way.

@gkasprow hmm well, on one channel, the flange-mount coax cable came loose when I removed an external SMA cable and was able to rotate by 360deg.

gkasprow, 2018-04-23:
We can also leave the cut-outs as they are and only add fixing screws. This would IMHO be more aesthetic than exposed module fronts.

hartytp, 2018-04-23:
> We can also leave the cut-outs as they are and only add fixing screws.

I'm not totally sure I follow you here. Currently, we have the following stack up:
- PCB mount SMA
- Module wall
- Star washer + nut
- Chassis wall (note that the module wall does not touch the chassis wall because of the nut + washer)
- Nut (no washer) recessed in cut-out

Do you mean changing this so
- that the module wall comes flush with the chassis wall
- having only one nut (tightening against the module wall)
- having a cut-out in the chassis wall that's big enough for the washer and nut to poke through?

That could work too.

-----

In the end, there are many ways of skinning a cat. My priorities are: cost, robustness, ease of connecting/removing external SMAs, machinability/ease of assembly/ease of module replacement. Aesthetics is good, but secondary.

- [ ] Anyway, you have much more experience than me with these kinds of things, so do what you think is best.

- [ ] But, please also make sure we have suitable (e.g. D-shaped) cutouts to prevent the cables from rotating

hartytp, 2018-04-28:
One other comment about these SMA cutouts: when I was removing modules from booster, I found it hard to replace channel 0. I think this was because the main PCB flexes a bit near the edges, so the PCB-mount D-type was at a slightly odd angle. If we have holes in the chassis front panel for each SMA then when we replace modules, the SMAs have to go in first and then the modules must be lowered/rotated down to mate with the main PCB. This gives very little ability to adjust the position of the module as it is lowered down, making it hard to connect it with the D-type. If there were a larger cutout, so then it would be easier to connect the modules with the main PCB.

This is essentially the same situation with Sayma/Allaki, where having SMA cutouts in the front panel makes it hard to assemble the whole units, and replacing with a larger cutout helps.

gkasprow, 2018-04-28:
OK, added to todo list.

gkasprow, 2018-04-28:
Do you mean larger cutouts for SMAs?

hartytp, 2018-04-28:
> Do you mean larger cutouts for SMAs?

For all these issues, I'd prefer not to dictate what design changes need to be be made. Let me tell you my experiences as a user/tester, and you can decide what -- if anything -- needs to be altered. 

In particular, if you think the "problems" I'm finding are not that bad, and that any solution would be too expensive/difficult/etc then feel free to leave things as they are.

In this particular case, my observation is that having small holes for the SMAs makes it hard to insert some modules because the modules must be rotated down, and the clearances are tight. So, if a PCB-mount D-type has moved slightly it can be hard to insert the module. 

Making the SMA cutouts larger would help. We could also consider having just one large cutout for each RF module: the cutout would be a rectangle with a rounded edge, and the RF modules would have a matching protrusion (a bit like FMC front panels). But, maybe that's too complex, would not work/look good in practice?

Anyway, do what you think will be best!

closed 2018-05-01 by gkasprow