sinara-hw/Booster #7: Booster: production specs

hartytp, 2018-04-23:
Once we've finished the design and moved into full-on production mode, we will need a proper specification for what the user will receive, including things like:
- Accuracy of the forward and reverse power measurements
- Accuracy of the gain, gain flatness and channel-channel variations
- P1dB
- etc

hartytp, 2018-09-06:
It would be great if someone could take the test data I've posted and collate it into a data sheet for this amplifier...

At this point, they're very well tested, so we can put together quite a nice data sheet.

hartytp, 2019-05-27:
Closing this issue on the grounds that it applies essentially equally to every part of Sinara. I think that a lack of proper documentation is now the biggest impediment to getting more users into Sinara (that's the piece of feedback I hear quite often from people considering adopting Sinara). It would be *great* for someone to include all the data we've taken (including things like phase noise measurements, cross-talk measurements, gain stability, output noise spectra) into a proper manual that shows how well Booster now works.

Hopefully now that there are commercial partners, this will actually happen some day...

closed 2019-05-27 by hartytp

gkasprow, 2019-05-27:
Creotech will apply for another grant for SInara commercialization. Things take a lot of time but sooner or later we will get decent documentation quality

hartytp, 2019-05-27:
Great!