sinara-hw/Booster #9: Booster: filters

hartytp, 2018-04-28:
This has been discussed elsewhere, but I'm creating a separate issue so that I remember to check it before we finalise the next version...

- We should add CM+DM filters on all signals entering/exiting the RF modules. Ideally, we should have filters on both sides of the RF module enclosure.
- Aim is to reduce RF cross-talk and also to reduce the SMPS spurs on the RF outputs that are currently seen at 201kHz offsets from the carrier
- Looks like there is plenty of room on the top PCB in the RF modules to add something like https://www.murata.com/en-global/products/productdetail.aspx?cate=luNoiseSupprFilteBlockType&partno=BNX026H01%23
- I'm assuming that adding these filters has negligible impact on the overall production costs, but let me know if I'm wrong about that!

gkasprow, 2018-04-29:
It's easier to add filters on the main board and stop SMPS interference from entering the board than fighting with them after they pass via LDO.
I wouldn't break continuity of the ground plane on the RF control module GND plane. I believe that better job would make low pass filter at the DC entry + additional ferrite beads on  the mainboard at the RF module socket .
The 30V SMPS modules have quite high ripples and there is only single stage LC filter used so far.

hartytp, 2018-04-29:
Okay @gkasprow, that makes sense.

My thinking was that it's very cheap to add simple LC filters inside the RF modules as well as on the main board (but maybe this is more difficult/expensive than I realise?), which would further reduce cross-talk and supply noise. The advantage of having the filtering inside the module as well as outside is that there is then very good screening (due to the module enclosures) between the filter stages, which can help with RF filters.

hartytp, 2018-05-01:
> added Murata filters at the mainboard entry. If that does not help, we can add them to the modules. They are quite expensive, we would need 32pcs * 8 EUR

Wow, that's worse than I thought! Which filters did you use?

I guess that once the PCBs arrive for the next revision, we can try selectively populating the filters to see which ones are actually needed, and remove any that don't help with the cross-talk/SMPS spurs.

hartytp, 2018-05-01:
As always, cost is my main concern here. I thought the filters would be very cheap, but if they push the price of Booster up by a noticeable amount, then maybe it's better to live with the noise.

Anyway, let's make a final decision once we've characterised the next design revision.

gkasprow, 2018-05-01:
Tom, I already did several steps to limit the interferences and crosstalk:
- added LC filter at each supply output on the mainboard
- grounded the fixing screws to the chassis
- added more fixing screws between channels
- added LC filters at the SMPS entry.
If that won't help, we can think about adding more filters. I don't want to add them blindly. At the moment the SMPS generates high ripples, we can easily filter them out using filters. LDO has poor rejection ratio above 200kHz and I'm sure we will get rid of them easily.

hartytp, 2018-05-01:
> If that won't help, we can think about adding more filters. I don't want to add them blindly. 

Sorry, my previous post wasn't clear. I wasn't suggesting adding more filters inside the RF module. It sounds like the filters you've added will be more than enough!

My suggestion was that we may find that some of the filters that you've added do not help. e.g. the pre-amp supply may not be a large source of cross-talk. In that case, we may be able to remove some of them for the final revision to save costs.

One final question: do we need CMCs/RFI filters (cheap ferrets) on the analog voltages/I2C on the d-type? Do you think that those lines are a potential source of cross-talk between modules?