sinara-hw/Booster #10: Higher power instability

hartytp, 2018-04-30:
@gkasprow measurement setup:
- Booster ch0 driven by a synth at 200MHz. RF power is either -8dBm or -11dBm input
- Measuring the output power on an RF power meter using -45dB high power attenuator between booster and the power meter. Thus, the gain of booster appears to be approx 0dB when the attenuators are included

See the attached video (apologies for filming upside down on my phone!):
- Synth starts off, meter reads -63dBm
- Turn RF on at -11dBm, the meter reads -11.3dBm. Leave that alone for a few seconds to demonstrate that Booster's output power is stable.
- Next, turn the synth to -8dBm. Meter reads -9.3dBm, as expected
- After a few seconds, Booster's output power ramps down to -10.3dBm over about 1s. Wait for a few seconds there to demonstrate that this is stable at this new output power
- Now, turn the synth back to -11dBm. Output power is now -12.4dBm. After a few seconds, the output power increases to -11.4dBm without me doing anything at all. Now wait for a few more seconds to show that Booster's output power is stable at this level.
- Now (around 00:30) turn synth back to -8dBm. Again, the power is -9.3dBm, but decreases by about 1dB after a second or so and stays there.
- One more time, I go to -11dBm input and see the same as before.
- Finally, I show a scope with traces on the 5V and 28V power lines. I toggle between -8dBm and -11dBm on the synth. A small glitch can be seen on the 28V line when I toggle between the two, but other than that there is nothing to be seen. So, it seems that this issue is not to do with these power lines.

Are you able to reproduce this? Are there any other tests you would like me to perform?

Thanks

hartytp, 2018-04-30:
https://drive.google.com/open?id=18rjKjgueQRWgvrGoNTXqZU5HJNA7Z3WH

hartytp, 2018-04-30:
The measurement I made on the power supply rails with a scope was not very precise. But, I would have thought it was good enough for to catch issue large enough to cause a 1dB change in the amplifier's output power.

Maybe this is some issue to do with the bias voltage?

hartytp, 2018-04-30:
Voltages on these two rails look correct: 27.9V and 4.94V. No noticeable change in the DC level when looking on a scope as I turn the RF on/off. AC coupling and looking on a finer scale, I do not see any obvious changes in the noise level as I turn the RF on/off.

So, I do not think either of these supplies are our issue here.

gkasprow, 2018-04-30:
You can also have a look at the negative bias voltage:
[image]

hartytp, 2018-04-30:
Thanks Greg! Tomorrow, I'll see if I can reproduce this with the lid removed from the RF module. If I can, I'll set something up to automatically toggle between input power setpoints every few seconds and probe all pins on the d-type. I'll let you know what I see.

hartytp, 2018-05-01:
I see the same behaviour on channel 1 as well (was previously only looking at channel 0).

NB I didn't try adding the 10uF capacitor to the negative rail yet, but based on my measurements, I don't think that's the issue. Let me know if you want me to try doing that.

hartytp, 2018-05-01:
@gkasprow when do you think that Maciej/Tomek will be able to look at this? I think we need to understand this before producing the next revision of Booster (which I want to do asap). If they can't look at it soon, then I'll do some more work myself. I think the next thing is to use the test points (which someone very kindly added) to measure the RF power at various stages in the circuit to determine where this effect originates.

gkasprow, 2018-05-01:
The University is closed till 6th May. We have bank holidays. All people are usually far away from Warsaw because we have real summer (roughly 30 degrees) since a few days.
So you can speed up the things a little. 
Once you manage to use IDC cables to get into the test points, you can identify which stage is causing the problems. To enable certain test point, one need to solder the jumper.

hartytp, 2018-05-01:
@gkasprow that's fine. 

I'll do what I can then.

I'm jealous of you having the sun. Make sure you all enjoy it :)

hartytp, 2018-05-01:
Well, I'm sure we can find a few days' slack in our timeframe, so don't feel like you have to work during the holiday on our behalf.

hartytp, 2018-05-02:
Interestingly, I don't see this behaviour on channels 2-4. There, the output power seems quite stable.

hartytp, 2018-05-02:
Nevermind, I do see this on those channels, but only at 1dB higher input powers.

hartytp, 2018-05-02:
I connected the UFL test points (thanks to whoever added those!) via 1k resistors to allow me to debug this better.
Nothing obviously wrong on P1 or P2. On P5, I see the output power increase by about 0.5dB after the RF input is turned on, but it's hard to tell if this is just a normal thermal warm up transient.
P5, I see the output power increase by 0.5dB as the overall output power decreases by 1dB....

hartytp, 2018-05-02:
Looking again here https://user-images.githubusercontent.com/21218399/39369489-5979e886-4a34-11e8-83a0-8aa87172623f.png it's interesting to note that when the overall output power decreases, the power in the second harmonic increases. That seems consistent with this being a biasing issue in some component.

hartytp, 2018-05-02:
@gkasprow hmmm...well, I'm not sure the measurements I've made have really tracked the issue down much better. I think it might be best to wait until Monday, when Maciej/Tomek can look at this properly, since they understand the circuit better than I do and are more likely to get to the bottom of this.

Let me know if you need any help reproducing this.

hartytp, 2018-05-02:
> @hartytp do you observe any form of hysteresis ?

Yes, see the post at the top of this issue, and the video I posted.

> Does it have form of time or only amplitude?

What do you mean?

gkasprow, 2018-05-02:
I mean if the hysteresis is only function of amplitude and time-invariant.
If the issue is caused by temperature, it will change with time. If it is caused by some form of positive feedback in the circuit, will depend only on the signal amplitude.

hartytp, 2018-05-02:
@gkasprow I'm still not sure exactly what you mean. Can you give me an example of a measurement I would do to test that?

gkasprow, 2018-05-02:
Do you have power detectors in your lab? Connect one to measure the input power and second to measure the output power.
Connect the power detector outputs to the XY inputs of the scope and for certain frequency change the amplitude quickly up and down (<1s). you should see a single line response. If there is a hysteresis loop, it means that it is not thermal issue.
Then change the amplitude toggle rate to a few second and repeat measurement. If you see a hysteresis loop that area depends on toggling rate, it means it is thermal issue.

hartytp, 2018-05-02:
Okay, I can do that tomorrow.

I've already done something a bit like that, but didn't understand what I saw. The behaviour is quite non-linear in both time and amplitude (which is why I don't think it's a thermal issue). Look here https://user-images.githubusercontent.com/21218399/39468229-ed208002-4d29-11e8-8de7-dcb3284b462f.jpg for the time non-linearity and here https://user-images.githubusercontent.com/21218399/39369489-5979e886-4a34-11e8-83a0-8aa87172623f.png for the amplitude non-linearity.

hartytp, 2018-05-03:
One more test:
- Start with RF off
- Then toggle RF on at -3dBm and off. Spending 5s on, then 5s off.
- Then change toggle time to 0.25s

With the fast toggle, I do not see this effect. So, it seems to be non-linear in time as well.

hartytp, 2018-05-03:
I can't think of an explanation for this...

hartytp, 2018-05-03:
If you can think of any other measurements you want me to do then let me know. Otherwise I'll stop there.

hartytp, 2018-05-03:
@gkasprow Okay, I'll look at both the main output and at the test points on the spectrum analyzer with the input driven at 200MHz near P3dB. I'll let you know if I see anything.

hartytp, 2018-05-03:
I did that, but I didn't see anything. I looked with no RF input, -11dBm RF input (well, a bit less than that due to my synth calibration errors) and -2dBm. For higher powers, I see a lot of harmonics but that's it.

I looked at the RF output and 4 test points (input, switch output, pre amp 1 output, pre amp 2 output). I did not see any large non-harmonic oscillations.

hartytp, 2018-05-03:
All measurements are 100MHz to 13.2GHz span. 100kHz RBW.

No input power applied:
[image]

Low input power (-11dBm on synth):
[image]

High input power (-2dBm on synth):
[image]

These shots are all of booster's output, but the signals I saw on the test points looked pretty similar.

hartytp, 2018-05-03:
Oops, those are in the wrong order. Anyway, I didn't see any oscillations obviously visible.

FWIW, the harmonics seem to be worst after the second pre-amp in the current design, so I guess that's what saturates first.

hartytp, 2018-05-03:
From my previous measurement I'm pretty sure this only occurs in the pa stage as I didn't see hysteresis on the TPs.

So unless there is some thermal cutput in the FET  I think this has to be an issue with the biasing of the final stage. Are you sure -1.2V is enough for 5W output power?

gkasprow, 2018-05-03:
@hartytp I'm not sure. Maciej recommended such voltage. After the weekend I will try to recreate this issue in my setup.

hartytp, 2018-05-03:
@gkasprow I checked the datasheet and Maciej's design looks correct AFAICT (although, I'd need to model it to make sure).

Hopefully you can recreate it and track it down!

Anyway, it's mysteries like this that make RF interesting.

gkasprow, 2018-05-08:
Maciej got the amplifier back and will recreate it.

hartytp, 2018-05-08:
Great. Let me know how he gets on, and if he's able to recreate the issue. I'm very curious to hear what's going on.

On a related note, looking at the FET biasing diagram in the data sheet, I do still wonder if we will need to implement active feedback on the gate voltage to stabilise the bias current.

hartytp, 2018-05-09:
How is this going? Did Maciej manage to reproduce the issue?

Note that I also saw a high voltage on the reverse power measurement with the amplifier driving a high-quality 50R load (https://github.com/sinara-hw/Booster/issues/2#issuecomment-385637632). It would be good to know what voltage you expect here.

hartytp, 2018-05-11:
any progress on this? Would it help if I shipped you our amplifier unit back?

gkasprow, 2018-05-11:
Maciej said that will send you today an email with his findings.

hartytp, 2018-05-14:
Summary of offline conversations with Maciej:
- This is reproducible at WUT
- Definitely an issue with the PA stage
- Either some issue with the RF choke inductors or with the PA itself. I don't fully understand what's going on here in either case since this kind of threshold behaviour is quite odd.
- Maciej looking into it, but away for a few days
- We'll post our Booster back to WUT for investigation and repair
- Finalising plans for Booster v1.1 to begin production asap once testing is finished on Booster v1.0

hartytp, 2018-05-24:
@gkasprow how is this going? Has a solution been found? Getting booster into production is absolutely my top priority atm, and these delays are causing me big problems.

hartytp, 2018-05-26:
Maciej fixed this by increasing a biasing inductor.

closed 2018-05-26 by hartytp