sinara-hw/Booster #12: Bias regulation range

gkasprow, 2018-05-02:
@wizath what negative BIAS (was -8V before) range is required for amplifiers designed by Maciej?
At the moment I implemented -1.5...-2.5V regulation range.

wizath, 2018-05-02:
At this moment we're using -1.22V, the upper limit of LDO

hartytp, 2018-05-08:
[image]

@gkasprow here is the plot from the datasheet, showing the range of voltages we need.

I'll be  interested to see what the temperature stability is like without actively controlling the bias current, but from that plot it looks like the bias current will vary quite a lot over temperature if we don't feed back on the voltage...

gkasprow, 2018-05-09:
I will add measurement of the bias voltage. I have one ADC channel which is connected to the 3.3V monitoring, I will use it to measure the bias voltage.

hartytp, 2018-05-09:
What do we need bias voltage monitoring for? I though that was set quite accurately using a DAC?

My suggestion was to monitor the bias current, and adjust the bias voltage to keep it at a set-point. I'm not an expert on this, but I think this is the standard thing to do (it's what the FET data sheet recommends). I think this will give improved thermal stability. If we use feedback then accurate knowledge of the bias voltage is not required.  

I believe that we already have all the hardware required to do this, it's just a matter of implementing it in firmware.

hartytp, 2018-06-06:
@gkasprow Did you make all the changes you want for this? Can we close this issue?

closed 2018-06-06 by gkasprow