sinara-hw/Booster #14: v1.1 schematics ready for review

gkasprow, 2018-05-02:
https://github.com/sinara-hw/Booster/releases/tag/v1.1_rc1

hartytp, 2018-05-02:
Great. I'll look at that tomorrow as well.

Once we get to the bottom of this transient issue, I think we'll be ready to begin production of v1.1.

hartytp, 2018-05-03:
Starting to look at these now. Will post comments here.

- [ ] Please can you make sure that the block diagram is the first page of all the schematics? It isn't for the control board, which is the only one I've looked at so far.
- [ ] It would be great to have a sentence or two at the top of each schematic, saying what that board does. Or, maybe an overall system block daigram, with short annotations on what the different PCBs do.

hartytp, 2018-05-03:
- [ ] Is it worth putting ferrites on the TX and RX power measurement pins? I assume these cost single digit pennies, so the cost is nothing. Without the ferrites, the capacitors that you have left space for won't be effective as filters.

hartytp, 2018-05-03:
- [ ] Is it worth at least having a pad for a ferrite + capacitor filter on the on/off line as well? Again, the thinking here is just to have the ability to filter out any RF that couples on to these control lines and then radiates out. Same for SDA and SCL, FR_Q, etc

hartytp, 2018-05-03:
- Is it worth adding a 10uF capacitor or larger on the 29V LDO output in parallel with the 1uF?

hartytp, 2018-05-03:
- Why does the schematic have so many copies of the thermal management and RF channel page? Is this some Altium thing? Can't we just have one copy of each, since all channels are identical?

gkasprow, 2018-05-03:
I can generate single copy of each sheet but then you have to know that i.e. C1A and C1B on the PCB belongs to different channels :)

hartytp, 2018-05-03:
- [ ] Is it worth having an extra LC filter stage between NV8 SMPS and the LDOs? We are quite sensitive to noise on this pin, and the LDO won't do a great job of removing SMPS spurs. Same for the 6V5 rail.

hartytp, 2018-05-03:
> I can generate single copy of each sheet but then you have to know that i.e. C1A and C1B on the PCB belongs to different channels :)

True. Well, do this however you think is best.

hartytp, 2018-05-03:
- [ ] Is it worth adding a larger SMT tantalum capacitor on the 29V (maybe also the 5V) rail inside the RF module? This will help to ensure that the amplifier has good transient performance.

hartytp, 2018-05-03:
- Why is the PA schematic called SUP CTRL?
- Why are the PA and pre-amp on separate PCBs? Wouldn't it be easier/cheaper to just combine the two designs into a single PCB?

hartytp, 2018-05-03:
- On the RF control page, what is the function of IC1A? IIUC, it resets the interlock (latches on the power detectors) when the on/off pin is high, and holds them in the reset state for approx 10us after the input has been enabled. Is that correct? If so, please can you add a short annotation explaining that.
- If that is correct, is that definitely what we want? Don't we want to reset the interlock **before** opening the switches? i.e. shouldn't the delay be between IC2 and IC1C? In the current configuration, if the interlock is reset while high input is applied, the switch will open for 10us, which could be long enough to cause damage to the power amplifier or to a sensitive load.

hartytp, 2018-05-03:
- I assume the RF boards haven't been updated yet (the attenuator and switch haven't changed). So I won't review them now.

hartytp, 2018-05-03:
@gkasprow Okay, I've completed my review on everything apart from the RF PCBs. Overall, it looks really good, so thanks for all the work.

Usual caveats: I've looked over it, but there are some parts I don't fully understand, so I'm not able to pick up all bugs, just give you my impressions. Also, feel free to ignore my suggestions/make counter suggestions if you don't agree with me.

gkasprow, 2018-05-09:
- [ ] add bias voltage monitoring

gkasprow, 2018-06-04:
LDOs don't like ceramic 10uF caps (very low ESR), so only 47uF tantalum on each RF module will be implementeted.

gkasprow, 2018-06-04:
I will add some comments on the schematics later on, the priority is to push it to production today.

hartytp, 2018-06-04:
> I will add some comments on the schematics later on, the priority is to push it to production today.

Great!

> LDOs don't like ceramic 10uF caps (very low ESR), so only 47uF tantalum on each RF module will be implementeted.

Yes, although usually this is okay if there is a ferrite bead between the LDO and the capacitor.

I would not expect tantalumns to do much at RF frequencies...

gkasprow, 2018-06-04:
There is already 1uF which blocks the RF. I can add  ceramic 10uF on the RF
module.

gkasprow, 2018-06-04:
Tantalum capacitor is to provide better pulse response of the supply chain.

hartytp, 2018-06-04:
> There is already 1uF which blocks the RF.

:+1: 

> I can add ceramic 10uF on the RF

Only if you think it's worth it. I'd stick with 100nF to 1uF. At 100MHz, ignoring ESR/ESL 1uF would be 1.6mOhm. So, in practice, the impedance is dominated by the ESR/ESL at the frequencies we care about. ESR/ESL are usually lower for smaller capacitors in a given package size, so it's not a good idea to make these caps too big.

> Tantalum capacitor is to provide better pulse response of the supply chain.

Aah, that makes sense. 

All sounds like a good plan!

wizath, 2018-06-04:
@gkasprow Could you connect I2C MUX reset pin to MCU? Since there is possibility that broken amp channel breaks the communication.

gkasprow, 2018-06-04:
OK, I was just sending files for board manufacturing :)

gkasprow, 2018-06-04:
I connected it to PB14. I added a pullup as well. Now you can reset it prior any I2C transfer.

hartytp, 2018-06-06:
Moved remaining points to #16

closed 2018-06-06 by hartytp