sinara-hw/Booster #15: Input coupler

hartytp, 2018-05-23:
The power meters used on Booster are not very accurate for low powers.

[image]

Currently, 0dBm input, which is well above Psat, gives -20dBm at the power detector. So, the measurement accuracy will be quite poor without careful calibration.

Might be worth considering using a -10dB coupler. Pick one with decently tight tolerance to avoid degrading the measurement accuracy.

gkasprow, 2018-05-23:
@hartytp OK OK. Let's do this calibration then. We can automate it and make the amplifier firmware do it automatically for certain power levels and loads. We can connect  all channels at the time to known power level, connect known load with characterized reflection coefficient and just normalize the readouts from all DACs at once. With such scenarios calibration would be simple and could be repeated at any time in the lab.

hartytp, 2018-05-23:
@gkasprow sorry, I'm not explaining myself clearly, and I think we're talking past each other...

*I do not want to calibrate each amplifier module unless we **really** have to*. My priority is to keep this design as cheap and simple as possible.

All I want is:
1. Maciej to double check that the circuit is optimised to provide good accuracy without calibration. e.g. 
- have we used reasonably accurate power detectors and couplers? Or, are there better options for a small cost increase?
- Have we used reasonably accurate resistors to form attenuators in the measurement paths?
- Have we set signal levels to be appropriate to give accurate measurements? (for the input power measurement, I don't think we have)
- etc
2. Maciej to provide an estimate of the measurement accuracy for his circuit without calibration. Based on the data sheet numbers I gave above, I expect that this will be fine for our needs.

In any case, for the initial version of the firmware we will definitely use a fixed V-dBm conversion with no per-module calibration.

hartytp, 2018-05-24:
AFAICT, the output power measurement/interlock is all that the users will care about.

hartytp, 2018-05-25:
Does anyone object to scrapping the input power measurement/interlock (cc @dtcallcock)?

Note that if we scrap the input power interlock then it will be possible to enable the amplifier with an excessive input power applied. The output power interlock is quite fast and will ensure that the high power won't be applied for more than a few ns, so I think this won't actually be a problem...

hartytp, 2018-05-25:
Similarly, does anyone object to scrapping the reverse power measurement in order to simplify the design? AFAICT, this is not required to protect the amplifier from damage.

As a user, I'm not too fused about having automated reverse power measurements. Does anyone care about this (I think it was @dtcallcock who originally requested it, but I can't remember the use-case), or can we scrap it?

dtcallcock, 2018-05-25:
> I think it was @dtcallcock who originally requested it, but I can't remember the use-case

It's a good diagnostic of problems with AOMs.  It's a nice-to-have at best though.

hartytp, 2018-05-26:
I see, you mean as a quick sanity check to test whether an AOM has died? Not for accurate measurements of reflected power, right?

In that case (assuming reverse power interlocks aren't required to protect the amp) let's scrap the reverse power interlock, but keep the power measurement for diagnostics. We can add a note to the data sheet that the reverse power measurement accuracy is bad for loads with high return loss due to the uncalibrated power detectors.

hartytp, 2018-06-06:
Greg, assuming Maciej has checked/is happy with the input coupler and, more generally, how the power detectors are wired up, please can you close this issue?

closed 2018-06-14 by hartytp