sinara-hw/Booster #16: Schematics cosmetics/docs/etc

hartytp, 2018-06-06:
- [ ] All remaining items from https://github.com/sinara-hw/sinara/issues/452
- [ ] It would be great to have a sentence or two at the top of each schematic, saying what that board does. Or, maybe an overall system block daigram, with short annotations on what the different PCBs do.
- [ ] Is it worth adding a 10uF capacitor or larger on the 29V LDO output in parallel with the 1uF? Has this been done already?
- [ ] Why is the PA schematic called SUP CTRL?
- [ ] Why are the PA and pre-amp on separate PCBs? Wouldn't it be easier/cheaper to just combine the two designs into a single PCB?
- [ ] add bias voltage monitoring
- [ ] On the RF control page, what is the function of IC1A? IIUC, it  resets the interlock (latches on the power detectors) when the on/off  pin is high, and holds them in the reset state for approx 10us after the  input has been enabled. Is that correct? If so, please can you add a short annotation explaining that. If that is correct, is that definitely what we want? Don't we want to reset the interlock before opening the switches? i.e. shouldn't the delay be between IC2 and IC1C?  In the current configuration, if the interlock is reset while high input is applied, the switch will open for 10us, which could be long  enough to cause damage to the power amplifier or to a sensitive load.
- [ ] Can we rename the ADL_Q signal to something like IN_Q to be more consistent with the FR_Q?
- [ ] on the RF control board, is there a silk screen layer with annotations for the TPs? I didn't see one in the PDF.
- [ ] on the RF control board, the rendering of the PCB looks a bit odd. Are we looking through the metal work?

hartytp, 2018-06-10:
@gkasprow what's the timeline for getting this finished? I'd really like to get Booster off my to do list!

Again, if you feel that some of the issues raised here aren't worth the effort then feel free to implement them. Either way, let's get a high-quality set of build files uploaded and finish this project off.

gkasprow, 2018-06-11:
@hartytp  is such block schematic sufficient?
[image]

hartytp, 2018-06-11:
The resolution of your post is a bit too low to see the details. But, from what I can tell: yes, that looks beautiful!

gkasprow, 2018-06-11:
Higher resolution:
[Booster_Block_Schematic.pdf](https://github.com/sinara-hw/Booster/files/2089782/Booster_Block_Schematic.pdf)

hartytp, 2018-06-11:
That's really nice.

"filtration" should be filtering. But, other than that, no comments.

gkasprow, 2018-06-14:
PA schematic is called SUP CTRL because it is the upper board that delivers supply and controls the lower board.
We have two RF boards that are mounted separately inside of the module because it's easier due to angled SMA connector. They are produced together as one board so it does not affect its price.
The test points are not necessary since we have big 2.54 grid connector. I added silk screen with signal names.
[image]

hartytp, 2018-06-15:
Nice! Thanks for doing all that.

As always, with the remaining items, feel free to only implement the ones that you thing add sufficient value to be worth the effort...

hartytp, 2018-06-15:
@gkasprow Did you post the updated schematics for the 3 RF module PCBs? I only saw the main board and the IFC board.

gkasprow, 2018-06-15:
Not yet, Maciej did not sent them to me - he is updating documentaion. The difference between v1.1 is cosmetic - just coil and some capacitor value. I'll post it once I get it.

hartytp, 2018-06-16:
@gkasprow okay, thanks. I don't think you uploaded the new supplemental control board yet (it's not in the latest release).

One other question: didn't we agree to scatter some more filtering ferrite beads + capacitors on all the signals into/out of the RF modules (e.g. the ADC signals, I2C etc) to suppress EMI and ensure that we achieve a very low cross-talk figure? I didn't see those in the schematics you uploaded. Or, did you decide that they won't be necessary? (Or, are they on the supplemental control board?)

gkasprow, 2018-06-16:
@hartytp I added missing files.
The capacitors (C67, C73, C79, C80, C81,C82) placeholders were added to the I2C, control and ADC lines on the main board.. Ferrite beads were added as well (L5...L10) on module control board.

hartytp, 2018-06-16:
PS Thanks again for the really high-quality schematics. They are now very clear and look great!

hartytp, 2018-10-20:
@gkasprow if possible, please can you have a look at the remaining issues here before I do my review? If you think some things aren't worth doing, then it's not a problem, just say so. Let's just make sure we do the important things.

gkasprow, 2018-10-20:
we don't need negative voltage monitoring, we don't have free ADC channels. So I'm closing it.

closed 2018-10-20 by gkasprow

hartytp, 2018-10-20:
thanks. FWIW, there are a few things left from https://github.com/sinara-hw/sinara/issues/452 but maybe we don't feel like they're worth doing.

gkasprow, 2018-10-20:
done.