sinara-hw/Booster #18: Booster: gain stability

hartytp, 2018-06-14:
Currently, the power FET is biased at constant gate voltage, rather than constant drain current. This is probably fine, but could lead to worse gain drifts

- [ ] Measure the gain stability over time/temperature.
  - Turn on transients
  - output power as a function of duty cycle
  - gain of one channel when all other channels run at max power vs all other channels off
  - stick a hot air gun in the air intake to change the chassis temperature

If we decide this needs to be improved then we should consider constant current biasing. This can should be able to be done in software with no hardware changes, but would need to check for spurs etc.

hartytp, 2018-06-14:
- [ ] @gkasprow should we add a filter on the output of the DACs that control the gate bias source? This will help to ensure that a software control loop does not add noise to the outputs as the DAC voltage changes. The simplest would be to solder a capacitor between vadj and ground. 
- NB if a software control loop turns out to add spurs, we can probably fix that with a simple sigma-delta modulator scheme

hartytp, 2018-06-14:
> The DAC controls negative LDO which output is additionally filtered.

Is that something that was added on the new schematics? Looking at the schematic on the releases page, I don't see any filtering between IC15 (DAC) and IC13 (negative LDO). If this has been added since those schematics were made then great (although, does that mean that the filter is not in the prototype unit that has already been produced?)!

gkasprow, 2018-06-14:
I'm just updating the documentation.
At the output of the LDO we have:
RC filter, then FB, another FB and 1uF in the module, then 4.7uF on RF board, then 38R, 100n and 680pF
So I think it's more than sufficient

hartytp, 2018-06-15:
@gkasprow true.

Okay, well in that case, I can't see any reason why doing a slow (kHz) feedback loop onto the amplifier current in software on the microprocessor wouldn't work.

hartytp, 2018-07-10:
cf https://github.com/sinara-hw/Booster/issues/19#issuecomment-403815448

closed 2018-07-10 by hartytp