sinara-hw/Booster #21: Booster firmware

hartytp, 2018-06-22:
@gkasprow @wizath I have updated the wiki (https://github.com/sinara-hw/Booster/wiki) with (what I hope is) a clear and complete specification for what the firmware should do. Please read this over carefully.

If there are any things you think should be changed then let me know asap. Otherwise, please ensure that the firmware on Booster exactly and completely implements the specification there before shipping it back to us.

Thanks!

hartytp, 2018-06-27:
NB added factory default reset command to specification on the wiki.

gkasprow, 2018-06-27:
Do you mean calibration values entered during factory calibration?

hartytp, 2018-06-27:
No, I was more thinking about the amplifier bias voltage, interlock threshold etc.

The calibration is not accessible via the ethernet interface and should not be affected by any ethernet commands.

gkasprow, 2018-06-27:
that's good. otherwise we would have to store 2 sets of coefficients, have dedicated commands.

hartytp, 2018-06-28:
@wizath I've done some tidying up on the software interface section of the Wiki for Booster.

Can you double check that this is all correct and fill in any missing parts, please?

In particular:
- [ ] What is the acceptable range of bias voltages for normal operation? Maciej should determine that from his circuit (but it's nice to have the option of decreasing the bias current a bit for applications that don't need so much output RF power)
- [ ] What does the UART interface do? IIRC, there are some diagnostics printed to it. There is also some way of querying the device MAC address. What are the serial port settings (baud rate etc)?

hartytp, 2018-06-28:
@wizath one other question, just because I'm curious: why does it take 200ms to switch on each channel when the interlock reset button is pressed? Presumably programming the DACs over I2C is quite quick. Is this the time it takes for the regulators to stabilise? (The 200ms isn't a problem, I'm just curious to understand where it comes from).

gkasprow, 2018-06-28:
The speed is 115200, 8, n, 1. The time was set arbitrary, can be shorter.
USB UART prints readout of all ADCs,  CPU IOs

hartytp, 2018-06-28:
Thanks.

> The time was set arbitrary, can be shorter.

Let's make this as short as we feel comfortable with. I'm impatient

> USB UART prints readout of all ADCs, CPU IOs

Continuously? Or, does the user have to ask for it? Can you post an example output here? What about the commands to read the MAC address?

gkasprow, 2018-06-28:
yes, every second. It's good for debugging right now. We can modify and make it on request.
@wizath please paste example output

wizath, 2018-06-29:
It's something like this:

[image]

wizath, 2018-06-29:
Lowered the reset wait to 10ms, working fine

hartytp, 2018-07-04:
@wizath 
- Please can you confirm that the description of the firmware on the [Wiki](https://github.com/sinara-hw/booster/wiki) is accurate. Are all commands implemented exactly as described and tested? Do the buttons and LEDs act as described?
- Please can you confirm what the allowed range of bias voltages (`CHAnnel:BIAS <channel> <voltage>`) is?
- Other than printing diagnostic information once per second, what commands are supported on the UART? e.g. what is the command to query the MAC + IP address? Are there any other commands that can be used on the UART?

wizath, 2018-07-09:
> Please can you confirm what the allowed range of bias voltages (CHAnnel:BIAS <channel> <voltage>) is?

I still got no information from Maciej, I will ping him once more

> Other than printing diagnostic information once per second, what commands are supported on the UART? e.g. what is the command to query the MAC + IP address? Are there any other commands that can be used on the UART?

I'm currently revamping the CLI, so it will include more debug/diagnostic commands. I'll update the wiki when it's finished

> please can you document the device calibration procedure used via the UART

(Interlock) Calibration process will require setting the desired signal on input/output and then using the calibration function from the CLI, rest will be automatic. 

We're currently working on power measurement calibration, It will involve similar steps as the interlock calibration.

hartytp, 2018-07-10:
- [ ] @wizath please can you document the fan controller settings on the Wiki? IIRC, the fans tun on at a low speed (what percentage) and stay there until the temperature hits 35C. Then the fan speed becomes proportional to temperature and reaches 100% at 85C. Is that correct?

hartytp, 2018-07-10:
@gkasprow what temperature does the chassis hit when all fans are running at Psat?

wizath, 2018-07-10:
> @wizath please can you document the fan controller settings on the Wiki? IIRC, the fans tun on at a low speed (what percentage) and stay there until the temperature hits 35C. Then the fan speed becomes proportional to temperature and reaches 100% at 85C. Is that correct?

On power-up then do initial spin, then if temperature is < 35 deg, they will go off. If the temperature rises above 35 deg, they will spin linearly from 26% (minimum speed without annoying 'clicks') to 100% at 80 degrees.

hartytp, 2018-07-10:
Can I suggest a couple of tweaks to that:
- 35deg is quite close to the ambient temperature this could run at in a rack, which could lead to the fans frequently turning on/off. Would it be better to say that when the temperature is <=25C the fans go to 30% (let's keep it safely away from 26% so we don't get the clicks on some units)
- let's then have it go to 100% at 65C. So, `fan_speed(temp) = 0.3 + min(0, max(1, (temp-25C)/40C)) * 0.7` so, for example, with all units running we might expect a 10C increase in temperature from an ambient of 30C. So, something like 40deg, which gives 56% fan speed. How does that sound @gkasprow is that too high a fan speed for full load in 25C ambient?

gkasprow, 2018-07-10:
Let's try it with full load first and see if this is sufficient. It's easy to modify anyway.

hartytp, 2018-07-10:
> Let's try it with full load first and see if this is sufficient. It's easy to modify anyway.

Okay. My thinking was just that we should keep the fan control quite tight to optimise the temperature stability in the chassis.

hartytp, 2018-07-17:
@gkasprow @wizath once you're happy that the firmware correctly and completely implements the specification on the Wiki, please close this issue.

hartytp, 2018-07-17:
> > Please can you confirm what the allowed range of bias voltages (CHAnnel:BIAS ) is?
> I still got no information from Maciej, I will ping him once more

@wizath Did you hear back from Maciej on this yet? It's important that this command has input validation to stop users programming a dangerous bias voltage, which could damage the FET.

wizath, 2018-07-17:
There is hardware circuit onboard which limits the voltage from something like -0.72V to -1.43, so we can adjust within full range of DAC without worry of damaging the amp.

hartytp, 2018-07-17:
Okay. How does the DAC voltage related to the FET bias voltage? Is there a divider somewhere, or does the DAC voltage directly control the FET voltage?

wizath, 2018-07-17:
DAC range (0 - 3.3V) controls the LDO range (-1.7V ~ -3.4V) then it's just 38R:47R resistor divider.

closed 2018-09-12 by hartytp