sinara-hw/Booster #22: Booster 28V LDO

hartytp, 2018-06-26:
@gkasprow some questions about the +29V LDO on booster:
- generally, why are we using such a high-current arrangement (beefy LDO + external pass BJT) to supply the +29V? Even a 0.25A limit on that rail would supply 7.25W per channel, which seems like plenty (the current design is something like 150mA max). Unnecessarily putting such a powerful regulator here makes it much easier for Booster to self-destruct. Additionally, the current arrangement is a little complex and could be simplified, again reducing the room for design bugs
- Where does the design used here come from? There seem to be quite a few more passives than in the reference design.
- Am I correct in thinking that when Booster self-destructed in my lab, these LDOs died? If so, how could that have happened, since they have internal thermal and over current protection?
- One possible destruction mechanism would be related to thermal runaway in the power BJT causing more current to leak from the base to collector. If the LDO went into thermal shutdown then such leakage would have to go through R12 (56k). It wouldn't take much leakage current to fully turn the transistor on and lead to destruction.
- I'd prefer not to have to design a fully robust power transistor design if we don't need to. Let's keep it as simple as possible and use a nice 30V LDO with integrated pass transistor and all protection handled internally.

What do you think?

hartytp, 2018-06-26:
Can we replace the pass transistor with a wire and use the regulator as a 0.25A regulator?

If we don't do that, please can you double check the design to look for potential failure modes and check that we haven't missed anything.

gkasprow, 2018-06-26:
I will add current limiting resistor and see how it behaves when shorted to GND.

hartytp, 2018-06-26:
@gkasprow sounds like a good idea. Let's make sure that this arrangement is not prone to self destruction, or replace it with a simpler regulator with integrated pass transistor.

hartytp, 2018-06-26:
> I will add current limiting resistor and see how it behaves when shorted to GND.

NB make sure that you do this test with the LDO/BJT at elevated temperature to simulate a case where the fans fail. My guess is that this was a thermal runaway in the power BJT.

Generally though, the more I think about it, the less I like having a 75W regulator here when 10W would be sufficient. It makes it far to easy for things to break.

gkasprow, 2018-06-26:
I fully share your point. Adding 0.25Ohm resistor between in 1 and 7 pin will limit the current to 0.4A and to 80mA when shorted to GND.
The problem was with too high current set in current configuration. The regulator had 2.5A of max current and roughly 300mA of current when shorted to GND. This caused 10W of thermal dissipation on bipolar transistor which quickly caused it to trip and short its terminals which caused LDO controller and power FET to fail. When we limit the current to 0.4A with foldback current of 80mA the dissipated power will be 2.4W which such transistor will handle.
We can also add electromechanical switch that will cut the main power when temperature inside exceeds 80 degrees. I's low cost and reliable solution.

hartytp, 2018-06-26:
@gkasprow cool, that all sounds good.

gkasprow, 2018-06-26:
This is how I debug the amplifiers:
[image]

gkasprow, 2018-06-27:
@hartytp I managed to limit the current to 0.5A but the power stage cannot start and the LDO is kept in short circuit state with current limited to 67mA. When the power stage is cold, it needs roughly 250mA to start. Then the current drops down to 160mA.
I installed 0.16R resistor and it starts but the current limit is 600mA.
We can leave it as it is or change the way the power stage is initialised.
Now we first apply desired bias voltage, then 5V and then 30V.
We can lower the LDO current and modify sequence and apply much higher negative bias voltage, then 5V , then 30V and then desired bias voltage. But then if the power stage for a moment exceeds let's say 400mA it will enter short circuit mode without possibility of leaving it without power cycle.

hartytp, 2018-06-27:
@gkasprow what do you think the best approach to use would be for this?

gkasprow, 2018-06-27:
The simpler, the better. Just set 600mA current limit.

gkasprow, 2018-06-27:
I added simple and robust protection. When fans fail it will cut the power after exceeding 80 degrees.
[image]

hartytp, 2018-06-27:
Sounds good!

Although, if the fans fail, the heat source will be mainly around the PA modules. So, by the time the IEC end of the enclosure gets to 80C, those amplifiers are going to be well over their max rating. So, on reflection, maybe this kind of simple protection isn't so effective?

gkasprow, 2018-06-27:
without forced convection the chassis will heat in more or less uniform way since the heat won't have a way to escape.

hartytp, 2018-06-28:
Okay. In any case, a thermal switch is cheap and probably a good idea.

gkasprow, 2018-09-12:
@hartytp did Technosystem install the thermal switch?

hartytp, 2018-09-12:
@gkasprow I didn't check.

The point of https://github.com/sinara-hw/Booster/issues/102 was to make sure that these features were added to the design/BOM. If they are already added then please close that issue

gkasprow, 2018-09-12:
they were added to mechanical BOM.

hartytp, 2018-09-12:
Perfect!