sinara-hw/Booster #25: errata

hartytp, 2018-07-10:
@gkasprow please can you post a list of all fixes you've applied during your bug testing, please?

gkasprow, 2018-07-10:
Main board:
- added current limiting resistor 0.18Ohm instead of embedded 0.025Ohm between pin 7 and pad 8 of U1(A,B,C..), disconnect pin 8.
- remove resistors in Vbias LDO: R87(A,B,C...) = 27k

RFPA_OUT
- removed 2 feedback resistors in OA, R32, R31
- shorted threshold regulation divider, R24 = 0R
- added 56R in parallel to C4 to change Vbias range from 1.7....3.2 to 1...1.8
- removed EEPROMIC (U8)

RFPA_PREAMP
- shorted threshold regulation divider, R29 = 0R
- removed feedback resistor in OA, R31

hartytp, 2018-07-10:
:+1: thanks!

wizath, 2018-07-10:
Main board:
- Neg voltage:
-- R87 - 22k
-- R86 - 10k
- 5V LDO
-- R10 - 12k
-- R13 - 3k9
- 30V LDO
-- R8 - 16k
-- R9 - 1k6

closed 2018-10-20 by gkasprow