sinara-hw/Booster #26: Booster full power tests

gkasprow, 2018-07-11:
@hartytp We run it at full power already for a few hours.
When I started measuring spectrum, I noticed sth strange. With power greater than 20dBm I noticed that the noise floor gets increased by 40 dB, but for exactly +- 100MHz. 1MHz above or below it’s fine
WTF?
Then I realized that the BW of the analyzer is 200MHz!
I simply saturated the analyzer.
After I added another 20dB attenuator, it looks much better. All measurements are done via 40dB attenuator.
Below are plot for all channels.

Ch0, 10MHz span
PWR = 35dBm
[image]

CH0, PWR = 20dBm
[image]

CH1:
[image]

CH2:
[image]

CH3:
[image]

Ch4:
[image]

CH6:
[image]

CH6 with harmonics, span = 1GHz. There is broadcast visible due to unshielded splitter.
[image]

CH7:
[image]

Ch7, 1MHz span
[image]

gkasprow, 2018-07-11:
@wizath 
I noticed that when all channels operate with full power, one cannot switch the amplifiers on.
One can switch off, but when wants to switch on, some channels are OK, some get interlock activated.
When I reduce the power to let's say 30dBm, I can switch them all on.
We will investigate it tomorrow.

hartytp, 2018-07-11:
@gkasprow thanks a lot for posting that! Looks good. So:
- no visible spurs at the 80dBc level, even with all channels saturated :+1:
- we hit a (spectrum analyzer?) noise floor of something like -80dBc/1MHz = -140dBc/Hz before 100kHz. By 1MHz offset it's more like -150dBc/Hz. :+1:

So, all in all, that looks excellent!

> I noticed that when all channels operate with full power, one cannot switch the amplifiers on.
> One can switch off, but when wants to switch on, some channels are OK, some get interlock activated.
> When I reduce the power to let's say 30dBm, I can switch them all on.
> We will investigate it tomorrow.

I'm glad you're putting booster through its paces and doing a thorough job of catching these bugs! Hopefully this is some sequencing/current limit issue that can be fixed easily from software.

wizath, 2018-07-12:
After testing from cold to warm:
Input: 16.7dBm -> splitter -> 8 channels
26 degrees - 1.15 V RMS
33 degrees - 1.13 V RMS
37 degrees - 1.13 V RMS

Couldn't get any hotter, fans keep it cool at about 35 degrees, running on the lowest possible level.

hartytp, 2018-07-12:
Wow, impressive. So, thats 0.15dB over the full operating range.

Nice.

Did you sort out the startup issue.

wizath, 2018-07-12:
It was only led issue, everything worked ok beneath this.

hartytp, 2018-07-12:
Cool. Sounds like everything is working really nicely. Can't wait to get my hands on it ;)

Thanks for all your great work on this project!

closed 2018-07-17 by hartytp