sinara-hw/Booster #27: Power readout calibration

wizath, 2018-07-16:
I'm struggling with calibration of power measurements atm, so far I've used simple linear approximation between two points, so it's simple and fast.

Used this table for converting the values:
http://wera.cen.uni-hamburg.de/DBM.shtml

Calibration process is done as follow:
* set power to 0dBm, gather ADC readout
* set power to (maximum) 30~ dBm, gather ADC readout
* calculate the linear coefficients and save them to eeprom

**Input power (dBm)**|**Measurement (scope) mV**|**Power level (scope /w 20db att)**|**Power level (RF output)**|**ADC Readout**|**Calculated Value**
:-----:|:-----:|:-----:|:-----:|:-----:|:-----:
-37.9|22.3|-20 dB|0 dB|823|-0.0535714286
-32.8|39.7|-15 dB|5 dB|1119|5.2321428571
-27.6|70.7|-10 dB|10 dB|1415|10.5178571429
-22.6|125|-5 dB|15 dB|1691|15.4464285714
-17.8|224|0 dB|20 dB|1949|20.0535714286
-12.4|397|5 dB|25 dB|2231|25.0892857143
-6.5|711|10 dB|30 dB|2535|30.5178571429
1.5|1160|~14 dB|34~ dB|2803|35.3035714286

This was calibrated using scope for reading out voltage from amplifier, so it is not very accurate. But the results are quite good. @hartytp what do you think about it?

hartytp, 2018-07-16:
Looks good to me: the worst-case error is about 1.3dB error at saturation, which is fine.

FWIW, if you took the higher power calibration point at something like +33dBm it would probably give an even better result.

gkasprow, 2018-07-17:
I was not very happy with accuracy of the input interlock. To improve it we changed the divider to use full DAC range. @wizath did test today, it is much better but I don't remember the numbers.
Another issue which I noticed today.
After I replaced one output stage which didn't manage to deliver full power, the small signal gain is similar, but I have an impression that this channel behaves in different way than others - the signal is distorted differently. The other channels outputs when running with full power (35dBm) gets clipped from upper side, but in the one I replaced it is clipped from bottom side. Moreover, I get much higher max output from this channel.
So it's quite possible that during recent failure caused by broken  DC/DCregulator, the MOSFETs also suffered when they were forced to conduct very high current without gate voltage. They were not completely damaged but probably lost some characteristics. The channel I fixed also has much different bias voltage than other channels. But this can be explained by different series of the output stage transistor.
Luckily I still have 4 output stages so can replace them when needed.

hartytp, 2018-07-17:
> I was not very happy with accuracy of the input interlock. To improve it we changed the divider to use full DAC range. @wizath did test today, it is much better but I don't remember the numbers.

:+1:

I see, so that improvement is due to the impedance of the divider combined with the power detector current draw? In that case, we should either remove the divider for the next revision or add an OpAmp. Once you've decided which approach you think is best, please add an item to the errata.

> After I replaced one output stage which didn't manage to deliver full power, the small signal gain is similar, but I have an impression that this channel behaves in different way than others - the [...]
So it's quite possible that during recent failure caused by broken DC/DCregulator, the MOSFETs also suffered when they were forced to conduct very high current without gate voltage. They were not completely damaged but probably lost some characteristics. The channel I fixed also has much different bias voltage than other channels. But this can be explained by different series of the output stage transistor.
Luckily I still have 4 output stages so can replace them when needed.

Good catch!

Does the bias voltage on the one you replaced match the data sheet more closely? Do you have enough FETs to replace all channels? Might be quicker than testing carefully to see which ones have a subtle fault.

gkasprow, 2018-07-17:
@hartytp now the impedance is less than 100Ohm so combined with 20uA of current gives 2mV of drop. Tests with other modules will show. PCB batch for next version is already done so don't want to add modifications more than just component values.
Tomorrow I will replace the power stages in the channels that have the lowest max output power.

hartytp, 2018-07-17:
Sounds good

closed 2018-08-16 by hartytp

hartytp, 2018-08-17:
@gkasprow I should have checked before closing, but: were the relevant component changes added to the new schematic and implemented in the new boards?