sinara-hw/Booster #28: Input interlock instability

wizath, 2018-07-17:
I'm getting trouble setting reliable value of the input interlock. 

At the point of 1 dB, DAC voltage after calibration is ranging from about 0.01V to 0.02V. This causes the input interlock to go on much lower power levels than desired value, with the maximum difference about 1.5dB worst case.

hartytp, 2018-07-17:
> At the point of 1 dB, DAC voltage after calibration is ranging from about 0.01V to 0.02V

Do you mean 1dB output power? Or, 1dB compression?

wizath, 2018-07-17:
1 dB input power

hartytp, 2018-07-17:
Worth adding a buffer after the divider for the next revision?

wizath, 2018-07-17:
After adding this resistor divider, interlock calibrated at 1dB input power is going off at about >= 0.6dB.

gkasprow, 2018-07-17:
It may be worth adding the amplifier before the detector...

hartytp, 2018-07-17:
> After adding this resistor divider, interlock calibrated at 1dB input power is going off at about >= 0.6dB.

Okay, so that's now an error of <0.5dB, which is fine.

hartytp, 2018-09-11:
@gkasprow were all these changes implemented in the schematic? If so, please close this issue...

gkasprow, 2018-09-12:
Technosystem was instructed to mount 0R resistor instead of R29 and R24. So the divider is not used. The documentation is not updated.

hartytp, 2018-09-12:
Thanks. That's what I thought based on the DAC values we saw when testing the interlock.

Please close this issue once the schematic is updated.

gkasprow, 2018-10-20:
we moved to external comparator so it is not relevant any more.

closed 2018-10-20 by gkasprow