sinara-hw/Booster #29: bias current setup

gkasprow, 2018-07-19:
@hartytp I replaced 6 output stages and we observe that the maximum output voltage got increased by 10..15% comparing with not modified channels.
What is more important, on some channels the drain current felt to roughly 80mA with -0.9V bias voltage 
So when we set the same current for each channel, i.e. 100mA, we get very different max power value per channel.  Previously we worked at 160mA. we will modify the bias divider to be able to set lower values.
I'm just curious why the gate thresholds changed so much with new components. Was it just another series of components?
And related question is if we should set the same current value for all channels or just optimise the max power and adjust the current correspondingly .

hartytp, 2018-07-19:
> I'm just curious why the gate thresholds changed so much with new components.

Were the old ones damaged due to running too much current through them?

hartytp, 2018-07-19:
> And related question is if we should set the same current value for all channels or just optimise the max power and adjust the current correspondingly .

I'm don't think I understand the trade offs here well enough to answer that. Presumably you want each channel to give the same gain as well?

gkasprow, 2018-07-19:
The gain is different story. To equalise the gains, one tunes one resistor in attenuator for low level signal.

hartytp, 2018-07-19:
@gkasprow Can you ask someone with more experience like Maciej or Sammer what the appropriate solution to this is, please?

FWIW though:
- I don't care it the Psat changes by 15% from channel to channel so long as we're always comfortably > 4W at saturation
- All the FET data sheets I've seen recommend adjusting the gate voltage to achieve a desired drain current. So, my guess is that's probably what we should do here
- adjusting the bias voltage to set a bias current seems like the nicest thing to do. For example, the firmware can have an autocalibration mode where it measures the gate voltage required to achieve the set current before it opens the RF switch. Doing anything involving Psat makes the calibration proceduce more complex

gkasprow, 2018-07-19:
At the moment the firmware simply adjust the bias in such way to get desired drain current so we will go this way. I simply don't want to surprise you with non-standard features :)

hartytp, 2018-07-19:
>  I simply don't want to surprise you with non-standard features :)

Which non-standard feature do you mean?

gkasprow, 2018-07-20:
I mean the way how we calibrate them. One can tune gain for low signal or at the compression point, we can adjust the current to get highest output level or maximum efficiency, etc.

hartytp, 2018-07-20:
@gkasprow I'll be happy for you to do whatever your RF guys like Sammer/Maciej think is best.

My guess would be that the "standard" thing to do is:
- tune the gain at small signals
- tune the gate voltage to give the same bias current on all amplifiers

I don't mind if the saturation output power/gain varies a small amount from channel to channel (that's quite normal for this kind of amplifier IIRC).