sinara-hw/Booster #31: UART interface

hartytp, 2018-07-27:
I want to query Booster's MAC address via the USB UART.

I've connected to it using `miniterm` and see the following printed every so often:

```
␛[2J
PGOOD: 1
FAN SPEED: 0 %
AVG TEMP: 35.14 MAX:34.50
CHANNELS INFO
==============================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	1	1	1	1	1	1	1	1	
TXPWR [V]	0.04	0.12	0.05	0.05	0.05	0.05	0.08	0.05	
RFLPWR [V]	0.08	0.04	0.05	0.04	0.06	0.10	0.04	0.04	
TXPWR [dB]	-12.15	-11.83	-13.35	-14.04	-12.59	-14.20	-12.71	-12.65	
RFLPWR [dB]	-14.90	-11.61	-15.25	-14.98	-11.78	-14.87	-13.02	-13.79	
P30V [A]	0.022	0.076	0.081	0.072	0.047	0.029	0.063	0.091	
P6V0 [A]	0.004	0.015	0.016	0.014	0.009	0.006	0.013	0.018	
PN8V0 [mA]	0.094	0.325	0.346	0.307	0.197	0.122	0.270	0.387	
ON		0	0	0	0	0	0	0	0	
SON		0	0	0	0	0	0	0	0	
IINT		0	0	0	0	0	0	0	0	
OINT		0	0	0	0	0	0	0	0	
SINT		0	0	0	0	0	0	0	0	
SINTE		0	0	0	0	0	0	0	0	
SINTV		0	0	0	0	0	0	0	0	
OVC		0	0	0	0	0	0	0	0	
ADC1		63	198	73	88	85	75	130	76	
ADC2		126	63	80	65	98	170	69	61	
DAC1		110	78	105	120	52	120	1500	146	
DAC2		1021	1015	1009	1010	954	1050	1043	1090	
SCALE1		55	53	55	55	56	56	55	57	
OFFSET1		731	825	807	860	790	870	829	797	
BIASCAL		135	1635	1515	90	105	1695	3180	390	
LTEMP		33.50	33.00	33.25	34.50	34.00	35.50	34.00	34.50	
RTEMP		32.25	33.50	33.50	33.00	34.25	33.00	34.50	33.00	
==============================================================================
```

when I type anything (including `help`) I see the message: `Command not recognised.  Enter 'help' to view a list of available commands.

Any suggestions? (For debugging, it might be better to print messages like "Command \"{}\" ..." where {} is the input received from the user. Echoing unrecognized input back like that makes it easier for users to see what they're doing wrong).

gkasprow, 2018-07-27:
@wizath please help

wizath, 2018-07-27:
Two ways:
- quickly type ifconfig to get the ip address / MAC of the board before screen clears itself
- type stop, then the status messages should be gone (also, the interlock task will be paused so remember to restart it), then type ifconfig to get the info, then type start to resume the tasks

hartytp, 2018-07-27:
That doesn't work. Is your code fussy about line endings?

hartytp, 2018-07-27:
Also, what is `␛[2J`?

hartytp, 2018-07-27:
@wizath okay, well it's probably some serial port setting. Can you post the settings I should be using on the wiki, please?

hartytp, 2018-07-27:
Okay, thanks. I need the MAC address asap to get this on the network and start testing it.

closed 2018-09-12 by hartytp