sinara-hw/Booster #32: UART interface tweaks

hartytp, 2018-07-27:
- It should not be picky about case or line endings
- it should not print anything unless asked to apart from error messages and a start up message; if users want regular updates then they should regularly call some status query command
- please don't use clear screen characters for the status queries
- when commands are not recognized, please include the received command as part of the error message to aid debugging

hartytp, 2018-07-31:
I think the current debug interface makes use of special characters, such as clear screen. That makes it a bit fragile. e.g. here is what it looks like with minicom on Ubuntu:

[image]

hartytp, 2018-08-01:
Also, it would be nice to have some help info about the commands (e.g. what parameters they take). so either something like `cmd --help` to get info about the command, or just print a more detailed help message when a command is run with no parameters/incorrect number of parameters

hartytp, 2018-09-12:
@wizath I have some odd behaviour with the current UART interface. @jochenwolf3 will remember the details better than I do, but it was something like:
- when I enter a command I first see a `command not recognised error`
- then I press enter again and the command works

wizath, 2018-09-12:
Yep that's the **feature** of current CLI lib. I almost finished writing new general purpose CLI library, so I'll switch when everything will be tested.

hartytp, 2018-09-12:
*shudder*

okay, sounds like a good plan.

closed 2018-10-18 by wizath