sinara-hw/Booster #33: MAC address reads 0

hartytp, 2018-07-27:
Got UART interface running using minicom with default settings (would need to cross-check against miniterm to see what the issue was)

```
stop
ifconfig
[log] ===== W5500 NET CONF : Static =====
[log]   MAC:    00:00:00:00:00:00
[log]   IP:     0.0.0.0
[log]   GW:     0.0.0.0
[log]   SN:     0.0.0.0
[log] =======================================
```

So, MAC address is 0.

wizath, 2018-07-27:
Interesting.. is the Ethernet cable plugged in?

hartytp, 2018-07-27:
No. Should it be?

wizath, 2018-07-27:
You can try connecting Ethernet cable first

hartytp, 2018-07-27:
Okay. That works.

So:
- [ ] we should be able to query the MAC address without an ethernet cable present if possible. (Or, at the very least, this "feature" should be documented)

wizath, 2018-07-27:
If you want to set static IP addressing, you should type:
ipconfig <ipaddress> <gw address> <netmask> into the cli.

wizath, 2018-07-27:
Yes, I'm going to refractor wiki, just need a while to get some time to do it.

hartytp, 2018-07-27:
:+1: Thanks

hartytp, 2018-07-31:
@wizath is DHCP enabled by default? I'm having some trouble getting it to pick up an IP address automatically.

wizath, 2018-07-31:
DHCP is enabled by default. You can stop the status messages (great idea that they should be disabled by default) and then connect eth cable and watch for DHCP debug msgs.

wizath, 2018-07-31:
It should print something if DHCP fails after several tries.

hartytp, 2018-07-31:
> DHCP is enabled by default. You can stop the status messages (great idea that they should be disabled by default) and then connect eth cable and watch for DHCP debug msgs.

Okay, I don't think I saw anything on the UART. I'll double check that in the morning.

hartytp, 2018-07-31:
Is there any way I can do this remotely? I have UART access, but I can't physically power cycle the device.

wizath, 2018-07-31:
You can type `reboot` to soft restart device. But this VCP based console takes some time to initialize, so you anyway could miss these printouts.

hartytp, 2018-07-31:
> But this VCP based console takes some time to initialize, so you anyway could miss these printouts.

Yes, I noticed that.

> You can type reboot to soft restart device

thanks! Sorry, now I think of it, I had seen that in the help menu (which is a really nice feature of the UART interface!)

hartytp, 2018-07-31:
So, any suggestions for how to catch the DHCP errors if the VCP takes a while to start?

wizath, 2018-07-31:
I'm planning to implement something like boot log, where errors/initialization statuses will be held, something like linux `dmesg`

hartytp, 2018-07-31:
:+1:

In the mean time, I'll check that it's not some issue with the DHCP here...

hartytp, 2018-08-01:
> I'm planning to implement something like boot log, where errors/initialization statuses will be held, something like linux dmesg

Let's try to keep the complexity of the firmware to an absolute minimum. e.g. I don't want to find that the device crashes every few months because a log takes up all the memory

hartytp, 2018-08-01:
So, I rebooted the device from the VCP to see the startup messages:
```
[log] DHCP failed after 4 retries
                                 [log] Loaded static IP settings
                                                                [log] ===== W5500 NET CONF : Static =====
[log]   MAC:    00:08:DC:AB:CD:EF
[log]   IP:     192.168.1.10
[log]   GW:     192.168.1.1
[log]   SN:     255.255.255.0
[log] =======================================
```

any ideas what could cause DHCP to fail?

wizath, 2018-08-01:
> any ideas what could cause DHCP to fail?

Interesting, i'll investigate it today.

hartytp, 2018-08-01:
```
[log] ===== W5500 NET CONF : Static =====
[log]   MAC:    00:08:DC:00:00:00
[log]   IP:     0.0.0.0
[log]   GW:     0.0.0.0
[log]   SN:     0.0.0.0
[log] =======================================
Command not recognised.  Enter 'help' to view a list of available commands.
[log] ===== W5500 NET CONF : Static =====
[log]   MAC:    00:08:DC:00:00:00
[log]   IP:     0.0.0.0
[log]   GW:     0.0.0.0
[log]   SN:     0.0.0.0
[log] =======================================
[log] DHCP failed after 4 retries
                                 [log] Loaded static IP settings
                                                                [log] ===== W5500 NET CONF : Static =====
[log]   MAC:    00:08:DC:AB:CD:EF
[log]   IP:     192.168.1.10
[log]   GW:     192.168.1.1
[log]   SN:     255.255.255.0
[log] =======================================
```

hartytp, 2018-08-01:
Why is the MAC different before the DHCP fails? Our network has MAC filtering...

wizath, 2018-08-01:
Looks like the settings are applied too late. Could you try to restart dhcp, by typing `ifconfig dhcp`. It will restart DHCP task, but hopefully the settings (along with MAC address) will remain.

hartytp, 2018-08-01:
@wizath thanks. 

I tried that. No DHCP errors, but I `ifconfig` now returns
```
[log] =======================================
[log] ===== W5500 NET CONF : Static =====
[log]   MAC:    00:08:DC:AB:CD:EF
[log]   IP:     0.0.0.0
[log]   GW:     0.0.0.0
[log]   SN:     255.255.255.0
[log] =======================================
```

hartytp, 2018-08-01:
```
Re/Starting DHCP clientCommand not recognised.  Enter 'help' to view a list of available commands.
Command not recognised.  Enter 'help' to view a list of available commands.
[log] ===== W5500 NET CONF : Static =====
[log]   MAC:    00:08:DC:AB:CD:EF
[log]   IP:     0.0.0.0
[log]   GW:     0.0.0.0
[log]   SN:     255.255.255.0
[log] =======================================
[log] ===== W5500 NET CONF : Static =====
[log]   MAC:    00:08:DC:AB:CD:EF
[log]   IP:     0.0.0.0
[log]   GW:     0.0.0.0
[log]   SN:     255.255.255.0
[log] =======================================
[log] ===== W5500 NET CONF : Static =====
[log]   MAC:    00:08:DC:AB:CD:EF
[log]   IP:     0.0.0.0
[log]   GW:     0.0.0.0
[log]   SN:     255.255.255.0
[log] =======================================
```

So, no DHCP errors, but no IP address either...

wizath, 2018-08-01:
At least MAC address is consistent. I'll try to investigate it, but at this moment I don't have any board to test (waiting for new revision).

hartytp, 2018-08-01:
It seems like the DHCP process isn't restarting properly, since I don't get any errors and I don't end up with a static IP. Well, either that or it is hanging.

When will you get the new revision? ATM the ethernet interface isn't much use since DHCP doesn't work and I would have to reassign the static IP each time I power cycle the device.

hartytp, 2018-08-16:
DHCP now works after a firmware update.

Thanks @wizath !

closed 2018-08-16 by hartytp