sinara-hw/Booster #35: Booster testing take II

hartytp, 2018-07-30:
Small-signal S21 measurements. Pin=-20dBm

[image]

Comments:
- I'm plotting 50MHz to 1GHz. The top end of that spectrum is outside the amps specification, just plotted for interest
- gain flatness looks good (as before) on all channels apart from channel 4, where the matching seems to be off
- gain is too high for basically all channels (we agreed to aim for 40dB at the center of the gain profile IIRC). I assume that this is due to the replacement of FETs, without retuning the gain. That's fine for this prototype unit, but obviously wouldn't be okay for a production unit

gkasprow, 2018-07-30:
It looks like in channel 4 there is wrong coupling capacitor mounted at some stage...

hartytp, 2018-07-30:
Yes, I assume it's something like that. It doesn't seem like a design error, just some issue with assembly/tuning.

hartytp, 2018-07-30:
Well, assuming it isn't that some capacitor has died or something nasty like that...

gkasprow, 2018-07-30:
We should have done full tests with VNA after all. I didn't expect that something like this could happen.

hartytp, 2018-07-30:
small-signal cross talk:
- -20dBm into channel 0, all other inputs 50Ohm terminated
- measure RF power out of other channels

[image]

- no channel to channel dependence on cross-talk
- below about 200MHz, I'm struggling to measure it (cross-talk is <100dB on most channels)
- bu 400MHz, the cross talk is still < 70dBc

previous measurement, before increasing the size of the RF chokes on the main 28V power rail:

[image]

hartytp, 2018-07-30:
Same, but signal applied to channel 4

[image]

gkasprow, 2018-07-30:
@hartytp yes, post it. I want to make sure I fully understand what is going on.

hartytp, 2018-07-30:
@gkasprow okay, I'll do that when I'm done with other tests

hartytp, 2018-08-13:
Using a synth + spectrum analyser to measure IIP3 + saturation
- f = 200MHz
- looking at channel 0
- gain = 45.8dB
- P1dB = 28.9dB
- P2dB = 32.0dB
- P3dB > 5W
- IIP3 = -2.9dBm

[image]

hartytp, 2018-08-13:
looks good!

hartytp, 2018-08-13:
Same for channel 1, which had lower gain for some reason. I don't see any need to look at any other channels for now.
- gain=42.4dB
- P1dB=26.2dB
- P2dB=30.7dB
- IIP3 = 1.7dBm

[image]

hartytp, 2018-08-14:
[image]

- Channel 0 running at roughly max power at 200MHz
- all other channels enabled but with no RF (NB current consumption should not change much depending on the output RF level IIRC)
- start logging just after turning the amplifier on, so this shows the warmup transient
- monitor the temperatures via the VCP interface. All channels start at around 25C. They heat up to something like 40C, then the fans cool them back down to about 30C
- the temperature coefficient is thus something like 0.2dB/20C = 0.01dB/C (0.2%/C) which is good!

hartytp, 2018-08-14:
The odd transient there is due to the fans not turning on until the units reach 35C. Would it be better to keep the fans running at min speed, even at lower temperatures to minimize this transient?

hartytp, 2018-08-14:
- use a hot air gun to heat one channel from 30C to 74C as measured by the UART (apologies for the scorched labels :) )
- power changed from 35.8dBm to 35.0dBm. That gives a temp co of roughly 0.02dB/C, so in rough agreement with the previous measurement. 

[image]

hartytp, 2018-08-14:
[image]

After warming up, the power stability looks to be very good indeed!

hartytp, 2018-08-14:
NB oops, x and y axis labels are swapped for all these figures ...

hartytp, 2018-08-14:
Another look at the warmup transient: 

In a 1C stable lab, the steady-state stability after warmup is 0.01dB or better. Good

[image]

hartytp, 2018-08-14:
Now, turn the RF synth off, wait for the amplifier to cool down then turn it on again (but, don't turn the amplifier into standby mode duri
[image]
[image]

So, we see a transient of something like 0.2dB (5% power), which lasts for a few minutes. Some of that may be the synth...

ng this time).

hartytp, 2018-08-15:
Final drift measurement. Same, but this time at 1W of RF.
- start logging a few minutes after turning booster on
- see usual startup transient where gain decreases by 0.1dB or so as Booster heats up, then increases again when the fans come on
- at the start, I have 1W of RF output, then turn the RF source off for a while (keep booster bias current on) then reactivate the RF source
- transients due to switching the source on/off are at the quantisation limit of my V3500A power meter.

Good!

hartytp, 2018-08-15:
[image]

hartytp, 2018-08-15:
Large-signal gain measured with -9dBm input (so 5W out with 46dB of gain):
[image]

NB channel 1 also has less small-signal gain, so I guess this is just an error in the tuning. Other than that, this all looks good!

hartytp, 2018-08-15:
Looked at transients on a scope
- switching the RF on/off with a MCL switch (ZASWA2-50DR-FT+)
- input RF is -9dBm
- rectify using a 423B, 50R terminated and followed by a 30MHz low pass filter (SLP-30+) and some further 100MHz digital filtering
- FS is approx -30mV, so 1% in power is 0.3mV

[image]
[image]

hartytp, 2018-08-15:
[image]

hartytp, 2018-08-15:
[image]

hartytp, 2018-08-15:
[image]

hartytp, 2018-08-15:
- The ns-level stuff could well be the RF switch (I've seen things like that from one of those switches before)
- otherwise, there looks to be a couple of % (a bit under 1mV) transient on a us-time scale and then maybe another percent or two on the 100ms timescale. From the previous measurement, there is then another percent or two over seconds/minutes.

Given the previous measurements, I would expect the transients to be a lot smaller for 1W output powers. 

Broadly, however, this seems consistent with my experience with other PAs at similar power levels from manufacturers like MCL.

@wizath that's about all I can do until I get new FW...

gkasprow, 2018-08-15:
@hartytp the transients are well known and caused protection activation. I mitigated it by adding small delay. They seem to be caused by internal capacitances of the RF switch.
That's why Maciej replaced switch with reflective one and additional attenuator to get rid of them. The glitches were not present in very first prototype designed by Tomasz.

hartytp, 2018-08-15:
> @hartytp the transients are well known and caused protection activation. I mitigated it by adding small delay. They seem to be caused by internal capacitances of the RF switch.
That's why Maciej replaced switch with reflective one and additional attenuator to get rid of them. The glitches were not present in very first prototype designed by Tomasz.

You mean these transients? 
https://github.com/sinara-hw/Booster/issues/35#issuecomment-413260278

Those aren't anything to do with Booster's interlock (which isn't used in this measurement). The only active switch in this design is the MCL switch outside Booster, which I use to turn my RF source on / off for the measurement. 

I think those transients are just from the MCL switch I'm using.

hartytp, 2018-08-17:
Booster v1.1 getting ready to ship...

[image]

hartytp, 2018-08-17:
@gkasprow just to check, but do the new modules have D-shaped holes for the SMA cables? During testing, another SMA came loose on my Booster v1.0 and was quite time-consuming to fix.

gkasprow, 2018-08-17:
Yes.

hartytp, 2018-08-17:
Cool, thanks :)

closed 2018-09-03 by hartytp

hartytp, 2018-09-05:
@gkasprow I'm curious: the amplifier modules in that phot look golden, but the ones I have are silver. What's the difference?

gkasprow, 2018-09-05:
I'm surprised by that. I never saw golden one.
This may be some side effect of nickel-plaing.

gkasprow, 2018-09-05:
Oh, no, it was me who tool this picture. It was automatic white balance. The camera always sets the balance to get grey colour in average. If you have one dominating intensive background colour, it will screw up the auto-balance mechanism. That's why some colours are wrong .It works in every camera with automatic WB setting. That's why it's better to set the illumination type manually to have good color representation.

gkasprow, 2018-09-05:
So they are the same amplifiers and they changed colour after crossing the border :)

hartytp, 2018-09-05:
Makes sense. I was wondering what the finish was. Is the nickel plating to improve the conductivity, and hence improve the screening?

gkasprow, 2018-09-05:
It is nickel plating. Aluminium tends to oxidise and after some time electrical contact gets worse. In our case it is not critical, but it simply looks good :) It was Technosystem initiative and I had nothing against it  Shielding is ensured mainly by densely populated M3 screws.

hartytp, 2018-09-05:
> It is nickel plating. Aluminium tends to oxidise and after some time electrical contact gets worse. 

Yes, I think it was the correct decision. And it does look *great*.