sinara-hw/Booster #36: Booster: metastability

hartytp, 2018-07-30:
@gkasprow something concerning happened during my gain measurements

- I plugged my VNA into one channel of booster and measured the gain. It was a bit lower than I expected
- After a minute or so, I pressed the standby button to turn all channels off, then pressed it again to turn them on
- after the channel came back up, it had the gain I expected (see plot)
- I haven't been able to reproduce this issue. Any ideas?

[image]

gkasprow, 2018-07-30:
Didn't you damage your attenuator in the meantime ? :)

hartytp, 2018-07-30:
> Didn't you damage your attenuator in the meantime ? :)

No. It's a 30W attenuator, with 100mW going into it. Moreover, the change in output power happened without me touching my measurement apparatus. I then remeasured the gain on other channels and found the same results I'd measured previously.

gkasprow, 2018-07-30:
Please switch it off, let it cool and repeat. It looks like poor soldering in the gain-matching attenuator since the effect does not depend on frequency

wizath, 2018-07-30:
DAC voltages are consistent through channel power cycle. I've observed this issue some days ago - even after calibrating the current, it sometimes differs from the set value. We can double check if the negative voltage is set properly during channel init.

hartytp, 2018-07-30:
> I've observed this issue some days ago - even after calibrating the current, it sometimes differs from the set value. 

Do you mean that you observed the gain sometimes being different when the channel starts up (i.e. the same thing I observed)?

wizath, 2018-07-30:
Not really the gain, just the current measurements were a bit off sometimes on single channel.

hartytp, 2018-07-30:
Did this happen on all channels, or was there just one dodgy channel?

How much was the current off by? Enough to explain the gain changes I've seen?

gkasprow, 2018-07-30:
@wizath was it the same channel?

wizath, 2018-07-30:
Which channel number is bad? I think it was 4th

hartytp, 2018-07-30:
Channel 5 was the problem

hartytp, 2018-07-31:
Here is the diagnostic output at startup

```
PGOOD: 1
FAN SPEED: 0 %
AVG TEMP: 27.52 MAX:30.00
CHANNELS INFO
==============================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	1	1	1	1	1	1	1	1	
TXPWR [V]	0.04	0.12	0.05	0.05	0.06	0.05	0.07	0.05	
RFLPWR [V]	0.08	0.04	0.05	0.04	0.04	0.09	0.05	0.04	
TXPWR [dB]	-12.02	-11.89	-13.18	-14.15	-12.34	-14.00	-13.02	-12.53	
RFLPWR [dB]	-14.77	-11.67	-15.08	-15.10	-11.52	-14.66	-13.35	-13.66	
P30V [A]	0.118	0.159	0.164	0.144	0.169	0.134	0.162	0.167	
P6V0 [A]	0.024	0.032	0.033	0.029	0.034	0.027	0.032	0.033	
PN8V0 [mA]	0.501	0.681	0.696	0.613	0.714	0.572	0.688	0.712	
ON		1	1	1	1	1	1	1	1	
SON		1	1	1	1	1	1	1	1	
IINT		0	0	0	0	0	0	0	0	
OINT		0	0	0	0	0	0	0	0	
SINT		0	0	0	0	0	0	0	0	
SINTE		0	0	0	0	0	0	0	0	
SINTV		0	0	0	0	0	0	0	0	
OVC		0	0	0	0	0	0	0	0	
ADC1		70	195	82	82	99	86	113	83	
ADC2		130	71	86	72	72	141	79	66	
DAC1		110	78	105	120	52	120	1500	146	
DAC2		1021	1015	1009	1010	954	1050	1043	1090	
SCALE1		55	53	55	55	56	56	55	57	
OFFSET1		731	825	807	860	790	870	829	797	
BIASCAL		135	1635	1515	90	105	1695	3180	390	
LTEMP		26.50	28.25	28.50	27.00	25.00	26.25	28.00	27.25	
RTEMP		25.25	28.25	29.50	28.50	26.25	26.50	30.00	27.25	
============================================
```

hartytp, 2018-07-31:
@wizath what is PN8V0?
- Why `P` when it's a current? Why not `I`? :)
- IIRC, N8V0 is the bias voltage for the FET, so I can't believe it has >0.5A draw. Am I correct in thinking that it's a typo and that `PN8V0` is really `I28V0`?
- If that's correct, why is there so much variance in the bias current? Didn't you tune the bias voltage to give the same bias current on each amplifier channel?

hartytp, 2018-07-31:
Power cycled this and looked again:
```
PGOOD: 1
FAN SPEED: 0 %
AVG TEMP: 30.50 MAX:31.00
CHANNELS INFO
==============================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	1	1	1	1	1	1	1	1	
TXPWR [V]	0.04	0.12	0.05	0.05	0.06	0.05	0.07	0.05	
RFLPWR [V]	0.08	0.04	0.05	0.04	0.05	0.09	0.05	0.04	
TXPWR [dB]	-12.02	-11.85	-13.18	-14.07	-12.36	-14.00	-12.91	-12.51	
RFLPWR [dB]	-14.77	-11.63	-15.08	-15.02	-11.54	-14.66	-13.23	-13.64	
P30V [A]	0.111	0.158	0.162	0.142	0.161	0.126	0.159	0.164	
P6V0 [A]	0.022	0.032	0.032	0.028	0.033	0.025	0.032	0.033	
PN8V0 [mA]	0.473	0.673	0.694	0.605	0.701	0.538	0.678	0.696	
ON		1	1	1	1	1	1	1	1	
SON		1	1	1	1	1	1	1	1	
IINT		0	0	0	0	0	0	0	0	
OINT		0	0	0	0	0	0	0	0	
SINT		0	0	0	0	0	0	0	0	
SINTE		0	0	0	0	0	0	0	0	
SINTV		0	0	0	0	0	0	0	0	
OVC		0	0	0	0	0	0	0	0	
ADC1		70	197	82	86	98	86	119	84	
ADC2		131	71	86	73	76	147	80	66	
DAC1		110	78	105	120	52	120	1500	146	
DAC2		1021	1015	1009	1010	954	1050	1043	1090	
SCALE1		55	53	55	55	56	56	55	57	
OFFSET1		731	825	807	860	790	870	829	797	
BIASCAL		135	1635	1515	90	105	1695	3180	390	
LTEMP		26.00	29.25	30.00	29.25	27.00	28.50	31.50	29.25	
RTEMP		27.50	29.50	31.00	29.00	27.50	27.00	30.00	29.25	
==============================================================================
```

So the currents look quite different

hartytp, 2018-07-31:
The fluctuations in the `Pn8V0` reading are surprisingly large. e.g. Cht goes between 0.582 ish and 0.62, which seems like a lot

hartytp, 2018-07-31:
Also, even when the device is in standby mode, the current draw on tha trail is 0.4A for one channel

hartytp, 2018-07-31:
Other than the above observations, I don't see anything obviously wrong with the diagnostics. @gkasprow @wizath is there anything in particular I should be looking out for? If not, I'll put the VNA back in and see if I can reproduce my previous observation of gain instability

hartytp, 2018-08-01:
I'll see if I can reproduce this. In the mean time, @wizath please could you answer my questions above e.g.
- what is `PN8V0`?
- how large do you expect the fluctuations on these readings to be?
- why is there such a larger difference between the `Pn8V0` reading from channel to channel?

wizath, 2018-08-01:
> what is PN8V0?
It's the current measurement on the negative voltage LDO.

> how large do you expect the fluctuations on these readings to be?
Since it's not calibrated, the accuracy is based on 12-bit ADC and 1% resistor.

> why is there such a larger difference between the Pn8V0 reading from channel to channel?
That could be because they're calibrated to draw about the same amount of current each and negative voltage that is set varies highly between channels.

hartytp, 2018-08-01:
@wizath Can you double check that woth Greg, please. I thought the NV8 rail was only for biasing the FET so the current should be close to zero. Are you sure it's not the 28V0 current?

gkasprow, 2018-08-01:
There is resistive divider of several tens of Ohms, so the current consumption is several mA.

hartytp, 2018-08-01:
@wizath apologies: i'd missed the 'm' there and thought it was in amps.

In that case, how do I read the 28V bias current? That's most important for me in terms of checking for start up issues.

hartytp, 2018-08-01:
@gkasprow how did you calibrate the FET gate voltage? IIRC you were going to calibrate it to give a fixed bias current, is that what you ended up doing? If so, what was the target current? Why is the current on channel 1 so much lower than the other channels?

hartytp, 2018-08-01:
One other question:

when the device is active, I see the following current draws
```
PGOOD: 1
FAN SPEED: 0 %
AVG TEMP: 30.50 MAX:32.25
CHANNELS INFO
==============================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	1	1	1	1	1	1	1	1	
TXPWR [V]	0.92	0.12	0.05	0.05	0.06	0.05	0.07	0.05	
RFLPWR [V]	0.62	0.04	0.05	0.04	0.05	0.10	0.05	0.04	
TXPWR [dB]	14.00	-11.89	-13.18	-14.07	-12.32	-14.02	-12.87	-12.54	
RFLPWR [dB]	12.75	-11.67	-15.08	-15.02	-11.50	-14.68	-13.19	-13.68	
P30V [A]	0.133	0.157	0.161	0.139	0.140	0.114	0.157	0.162	
P6V0 [A]	0.028	0.031	0.032	0.028	0.028	0.023	0.031	0.032	
PN8V0 [mA]	0.600	0.668	0.683	0.595	0.600	0.483	0.668	0.688	
ON		1	1	1	1	1	1	1	1	
SON		1	1	1	1	1	1	1	1	
IINT		0	0	0	0	0	0	0	0	
OINT		0	0	0	0	0	0	0	0	
SINT		0	0	0	0	0	0	0	0	
SINTE		0	0	0	0	0	0	0	0	
SINTV		0	0	0	0	0	0	0	0	
OVC		0	0	0	0	0	0	0	0	
ADC1		1501	195	82	86	100	85	121	82	
ADC2		1011	72	86	73	80	159	83	65	
DAC1		110	78	105	120	52	120	1500	146	
DAC2		1021	1015	1009	1010	954	1050	1043	1090	
SCALE1		55	53	55	55	56	56	55	57	
OFFSET1		731	825	807	860	790	870	829	797	
BIASCAL		135	1635	1515	90	105	1695	3180	390	
LTEMP		29.00	28.00	29.25	29.25	28.00	31.50	31.00	31.00	
RTEMP		31.00	30.50	30.25	28.00	28.25	29.00	32.25	30.00	
==============================================================================
```
when its in standby I see
```
================================================
␛[2J
PGOOD: 1
FAN SPEED: 0 %
AVG TEMP: 31.35 MAX:29.25
CHANNELS INFO
==============================================================================
		#0	#1	#2	#3	#4	#5	#6	#7
DETECTED	1	1	1	1	1	1	1	1	
TXPWR [V]	0.04	0.11	0.04	0.05	0.11	0.05	0.07	0.05	
RFLPWR [V]	0.08	0.04	0.05	0.04	0.04	0.09	0.04	0.04	
TXPWR [dB]	-12.16	-12.04	-13.35	-14.25	-10.77	-14.20	-13.04	-12.63	
RFLPWR [dB]	-14.92	-11.81	-15.25	-15.22	-9.89	-14.87	-13.37	-13.77	
P30V [A]	0.022	0.078	0.082	0.074	0.053	0.030	0.065	0.091	
P6V0 [A]	0.004	0.016	0.016	0.015	0.011	0.006	0.013	0.018	
PN8V0 [mA]	0.094	0.330	0.351	0.314	0.221	0.127	0.278	0.387	
ON		0	0	0	0	0	0	0	0	
SON		0	0	0	0	0	0	0	0	
IINT		0	0	0	0	0	1	0	0	
OINT		0	0	0	0	0	0	0	0	
SINT		0	0	0	0	0	0	0	0	
SINTE		0	0	0	0	0	0	0	0	
SINTV		0	0	0	0	0	0	0	0	
OVC		0	0	0	0	0	0	0	0	
ADC1		62	187	73	76	187	75	112	77	
ADC2		124	63	80	64	69	150	72	60	
DAC1		110	78	105	120	52	120	1500	146	
DAC2		1021	1015	1009	1010	954	1050	1043	1090	
SCALE1		55	53	55	55	56	56	55	57	
OFFSET1		731	825	807	860	790	870	829	797	
BIASCAL		135	1635	1515	90	105	1695	3180	390	
LTEMP		30.25	28.00	28.00	29.50	28.00	31.25	31.50	31.00	
RTEMP		28.00	29.25	28.00	27.00	28.25	29.00	29.00	29.00	
==============================================================================
```

Why is the 30V current draw so high in standby for some channels? It seems like there is only a factor of 2 between normal operation and standby mode. Is this just ADC offsets? If so, how accurate is this measurement?

dallcock, 2018-08-01:
> AVG TEMP: 31.35 MAX:29.25

This looks fishy.

hartytp, 2018-08-06:
Oh dear, have we lost David?

hartytp, 2018-08-06:
https://github.com/sinara-hw/Booster/issues/36#issuecomment-409612201

@wizath

jordens, 2018-08-06:
@dtcallcock is also metastable now

hartytp, 2018-08-06:
which, I'd suggest, is an improvement from his days here