sinara-hw/Booster #41: interlock threshold resolution

hartytp, 2018-08-01:
let's make it 0.1dB (currently 1dB) that way, it's a bit better than the measurement accuracy

wizath, 2018-08-29:
Is it better to save software interlock setting to EEPROM every time it is changed? Or create new command specifically to save them?

hartytp, 2018-08-29:
Personally, I'd make them automatically save. Otherwise, it's easy for the user to forget to call the save command, and be surprised that their settings are lost when the power is cut.

But, I don't really mind about this, since it's easy for me to implement automatic saving on my PC-side python.

hartytp, 2018-08-29:
As always, the most important thing is that the description on the Wiki should match the behavior of the firmware! I'm less fussed about whether this is achieved by altering the Wiki or by altering the firmware

wizath, 2018-08-29:
I'll keep them to save every time. Closing

closed 2018-08-29 by wizath