sinara-hw/Booster #42: no life

hartytp, 2018-08-14:
@gkasprow this morning, Booster won't start up. As before, I see a blue LED on the 12V SMPS and 2 green LEDs on two of the other rails (P30V and the one to the right of it, but the silk-screen is illegible because of vias).

What should I do?

hartytp, 2018-08-14:
The standby button doesn't do anything any more :(

gkasprow, 2018-08-14:
Is there 3V3 present? Does the LMZ14203 regulator work?

hartytp, 2018-08-14:
3V3 LED is off, so I guess not 3V3.

What do you want me to check on the LMZ14203?

hartytp, 2018-08-14:
LMZ14203:
- Vin=12V
- Ven=4.8V
- Vo = 6.4V

hartytp, 2018-08-14:
I see something like 30mV on the big capacitor, so I guess the 3V3 regulator is not starting correctly.

hartytp, 2018-08-14:
With a pen on the diode to make a connection, I see 3V4 on the capacitor, which I guess is within spec. 

Some of the LEDs are retina-burningly bright, while others glow at a more civilised level. Is this some anti-tamper feature? ;)

hartytp, 2018-08-14:
[image]
[image]
[image]
[image]

hartytp, 2018-08-14:
It looks like a simple dry joint, so I'll go ahead and resolder it unless you object @gkasprow?

closed 2018-08-14 by hartytp

gkasprow, 2018-08-14:
Sure, solder it. LEDs have wrong resistors so they shine like crazy. I fixed it in current release.