sinara-hw/Booster #44: Interlock LEDs

hartytp, 2018-08-14:
- I heated one Booster channel with a hot air gun to about 60C
- the yellow overload LED cam on and the RF turned off
- I thought we agreed that temperature would trigger the RED "ERROR" LED, didn't we?

@marmeladapk

wizath, 2018-08-14:
They should go red after 80 degrees, could you check on the console which error is present?

hartytp, 2018-08-14:
hmmm...okay. Will do.

wizath, 2018-08-14:
Honestly - I've never tested this feature, even after long time temperature never exceeded 50 degrees.

hartytp, 2018-08-14:
Just heated it up again and didn't see this. Maybe the interlock threshold drifted a bit with power.

I'll have a look at that later on.

hartytp, 2018-08-14:
Okay, never mind. I do get a red LED when 80C is exceeded. I had the power within 0.3dB of the interlock threshold, so that was probably my problem

closed 2018-08-14 by hartytp