sinara-hw/Booster #47: fan speed control

hartytp, 2018-08-15:
I think it would make more sense if the fans never switched off fully.

The current behavior leads to an odd warm up transient, whereby the gain decreases slowly as the unit heats up, but then suddenly decreases a bit when the fans turn on. If the fans were always on, the size of the transient would be smaller, and there would not be this odd overshoot.

hartytp, 2019-03-06:
Looking at our v1.3 Booster the fans do not turn on without RF applied, but the VCP reports channel temperatures of around 40C. This seems quite hot to not have any fans on.

cjbe, 2019-03-06:
@wizath with this v1.3 the fans are still not on under load, despite the average temperature pushing 50 deg. C - is this what you expect?
```
PGOOD: 1
FAN SPEED: 0 %
AVG TEMP: 49.00 CURRENT: 49.00
CHANNELS INFO
==============================================================================
                #0      #1      #2      #3      #4      #5      #6      #7
DETECTED        1       1       1       1       1       1       1       1
HWID            B2:13   B1:CF   90:A2   69:E0   6D:76   56:81   3C:9C   39:6B
TXPWR [V]       1.86    1.92    1.88    1.87    1.94    1.91    1.93    1.93
RFLPWR [V]      0.37    0.50    0.51    0.47    0.45    0.30    0.42    0.18
TXPWR [dB]      30.56   29.80   30.12   30.67   30.73   30.16   30.22   30.68
RFLPWR [dB]     32.87   33.44   33.95   32.11   34.56   32.79   31.75   34.10
I30V [A]        0.164   0.156   0.162   0.160   0.161   0.169   0.173   0.163
I6V0 [A]        0.270   0.254   0.279   0.500   0.262   0.259   0.254   0.270
5V0MP [V]       4.985   4.973   4.963   4.974   4.959   4.962   4.948   4.973
ON              1       1       1       1       1       1       1       1
SON             1       1       1       1       1       1       1       1
IINT            0       0       0       0       0       0       0       0
OINT            0       0       0       0       0       0       0       0
SINT            0       0       0       0       0       0       0       0
OVC             0       0       0       0       0       0       0       0
ADC1            3041    3149    3085    3070    3170    3129    3169    3160
ADC2            602     818     840     769     739     491     682     291
DAC1            0       0       255     0       0       0       0       0
DAC2            3489    3706    255     3483    3534    3571    3595    3545
SCALE1          81      88      86      88      84      87      85      87
OFFSET1         566     527     495     371     589     505     600     491
BIASCAL         1811    1960    1690    1700    1712    1687    1657    1756
HWIS            -224.79 -360.67 -238.83 -222.79 -244.34 -263.46 -276.45 -249.60
HWIO            1869.67 2999.90 1986.44 1853.08 2032.29 2191.37 2299.37 2076.09
LTEMP           43.00   46.25   47.50   47.50   48.00   47.00   46.25   44.00
RTEMP           45.25   45.00   47.00   47.00   49.00   48.00   45.00   46.25
==============================================================================
```

hartytp, 2019-03-06:
hmmm...how hot can this get before we need to shut it down?

hartytp, 2019-03-06:
@wizath any idea why the Rx power is reported as being higher than the Tx power? (I assume that @cjbe has terminated all channels, so presumably this is a calibration issue).

wizath, 2019-03-06:
I'll push you update tomorrow with fixed 30% fan speed if any channel is enabled. I'm in process of testing new PID control that hopefully will be working better than this averaging thing

hartytp, 2019-03-06:
Great. With the current firmware, will the fans ever turn on? Is it safe to leave this over night as it is?

wizath, 2019-03-06:
I would play it safe and disable it, since fans were working fine when I was testing channels - they seem like they have their own will to work... Thermal cutoff is set to 80 degrees, a little bit to high regarding current working temperatures with new bias current but I don't know how FETs will behave.

hartytp, 2019-03-06:
The ghosts are back...

hartytp, 2019-03-06:
@cjbe are you okay to shut this down remotely?

hartytp, 2019-03-06:
Cool :) that will be great data to see.

It would be great if he can test the board carefully. e.g.
- connect channel's input A to a precision 10k resistor. Use channel B to sweep the board temperature and record the "temperature" measured on channel A. This tells us the device temp co
- again, using a 10k resistor instead of a thermistor, run the board open loop and look at the change in temperature reading as a function of TEC current driven into a resistive load. This tells us about measurement errors induced by TEC current
- put a 10k thermistor on a 5m screened cable and put near a noise source. See if this adds noise to the temperature measurement
- temperature stabilize a block of metal carefully leave for a day or two while also monitoring a witness thermistor. We should see much better than 1mK noise/drift (may need 2-layer temperature control if the ambient is not quite stable).

I think those are all the important measurements to make to verify the board's performance, but obviously any other measurements you can think of would be great.

hartytp, 2019-03-19:
As discussed in #172 let's reduce the minimum fan speed a bit (less noise, longer fan lifetime). Should be possible now since the new hw rev has better fans.

wizath, 2019-05-29:
fixed

closed 2019-05-29 by wizath