sinara-hw/Booster #49: Booster RF PA schematic errata

gkasprow, 2018-08-18:
_From @dhslichter on December 27, 2017 20:33_

@thomasRF some errata for the schematics (moved from #251):

Preamp protection:
- [x] Input SMA should appear where it's currently labeled just TP1.
- [x] Part numbers are invalid/garbled for many ICs. Amps and power detector IC6, IC7, IC8, IC9 all have part numbers that don't seem to correspond to any real parts that Google knows about  For example, power detector is HMC7832 and switch is BBWA5673 -- neither of these are actual real part numbers for any part that exists AFAICT.  What are the actual part numbers for all these?  
- [x] CP2 needs a part number
- [x] Notation should be made about the purpose and connectivity of the signals on J4 and J5, since they are routed to a separate board (PA_Channel_SUP_CTRL).
- [x] Any critical inductors and capacitors in the amplification chain whose properties are important (particular voltage/current ratings, dielectric material for caps, core material for inductors, etc) should be called out.

Power stage:
- [x] IC3 part number doesn't correspond to a part Google knows about (MMWX1)?
- [x] Notations should be made for voltage/current/power ratings of passive components as appropriate.
- [x] J3 should have a notation as to its purpose since it is connected to another board not in this schematic.

Output protection:
- [x] CP1 needs a part number
- [x] IC1, IC2 again have an invalid part number
- [x] Notations need to be made about the purposes of the signals on J1, since they go to another board.

Edit: I can see correct part numbers as LibRefs in Altium when I go to make a BOM, but they need to be correct in the comments on the schematic.

_Copied from original issue: sinara-hw/sinara#452_

gkasprow, 2018-08-18:
_From @hartytp on December 27, 2017 21:21_

Adding to Daniel's points:

Control PCB: 
- [x] Please add a power budget. This should include the maximum power that each RF amplifier module can consume from each supply rail.
  - What limits the current on the supply rails? The supplies themselves? the d-types? the PCB routing (maximum tolerable voltage drop on the PCB?)? The overall max power dissipated in the chassis?
- [x] Power sequencing
- [x] Table or something giving the pin-out for the RF module to control PCB D-sub connectors. This should include a description of how each pin is used each pin (e.g. "Forward power (?V/W, differential)" or whatever), as well as the max current per power rail pin, etc

RF module schematics:
- [ ] Add a power budget

gkasprow, 2018-08-18:
_From @hartytp on December 27, 2017 21:22_

- [x] Pease check that all the part numbers in the Altium file are correct. For example, when I looked at the Atlium file before, it looked like the capacitance labelled on the schematic didn't always agree with the actual component part number when I opened up the component properties in Altium. This looks like a simple copy and paste error, but it's important to get it right. Similarly, *please check that the BOM matches the schematic.*

gkasprow, 2018-08-18:
_From @hartytp on December 27, 2017 21:24_

- [x] Translate schematic annotations to English
- [x] Tidy up schematic: move overview to the first page; update layout pages (currently they show many unrouted nets; etc... @gkasprow can you make this as beautiful as all the EEM schematics, please?
- [x] Zero-index all channels

gkasprow, 2018-08-18:
_From @hartytp on December 27, 2017 21:39_

Oh, and:
- [x] Please ensure that all schematic (and other design) files are well-organised in a sensible file structure, with sensible names (use the prefix "Booster")
- [x] Schematic should have the title "Booster RF Power Amplifier" and a version number

gkasprow, 2018-08-18:
_From @hartytp on January 6, 2018 22:2_

- Please can you add a table to the schematic (ideally on the front page) with a brief summary of important info for the RF PA, including:
  - [x] Gain (e.g. "Gain = 40dB ± 1.5dB")
  - [x] 3dB small-signal bandwidth
  - [ ] Output power at 1dB compression and output power at saturation
  - [x] Input power at 1dB saturation, and absolute max input power without damage
  - [ ] DC current draw

gkasprow, 2018-08-18:
_From @hartytp on April 23, 2018 12:11_

@gkasprow Please close if this has all been done.

gkasprow, 2018-08-18:
_From @hartytp on May 26, 2018 9:57_

@gkasprow Please can you close this issue once this is done?

NB schematic/cosmetic issues aren't crucial and can wait until after the next batch goes to production.

gkasprow, 2018-08-18:
_From @hartytp on June 6, 2018 8:17_

@gkasprow Now that booster v2.0 has gone to production, do you have time to finish this issue off? I'd really like to close all these Booster issues down and freeze the design/schematics.

closed 2018-10-20 by gkasprow

hartytp, 2018-10-20:
:fireworks: thanks for doing all this!