sinara-hw/Booster #50: Oscillation

hartytp, 2018-08-29:
I'm seeing an oscillation on the output of one channel at about 785MHz, +33dBm output. This is with the input 50Ohm terminated and the output connected directly to a high-power SMA terminator.

The LEDs on the FP indicate that the interlock has tripped.

hartytp, 2018-08-29:
I'll investigate the cause of this, but maybe we should change the firmware so that tripping the interlock results in the PA bias current being shut off.

hartytp, 2018-08-29:
[image]

UART log with the device in standby mode.

[image]
UART log with the device active

wizath, 2018-08-29:
Weird, I2C seems to be broken since there are no measurements from temperature and current sensors.

wizath, 2018-08-29:
But the calibration values are loaded, so it's even more confusing.

hartytp, 2018-08-29:
Any suggestions, or should I just ship it back to WUT for investigation?

hartytp, 2018-08-29:
It could just be another joint (this unit has had a lot of rework done on it and is getting a little fragile).

However, it does *really* concern me that the current interlock isn't preventing full-power from being applied to the load continuously in this fault condition. Whatever the cause of this, we absolutely must ensure that the bias current is cut off in case of error.

wizath, 2018-08-29:
Try to power cycle and see if it's getting any measurements, since it is weird that every channel is not accessible by I2C.

You mean shutting down bias current after tripping the interlock?

hartytp, 2018-08-29:
Anyway, AFAICT that channel is just dead, since this happens whatever I do (power cycling it, changing input + output connections).

> You mean shutting down bias current after tripping the interlock?

Yes, that seems to be the most reliable solution (unless you have any other suggestions)

hartytp, 2018-08-29:
I2C seems fine after power cycling. 

[image]

hartytp, 2018-08-29:
NB When I had the I2C issue, I'd pressed the standby button very qucikly after powering the unit on. Not sure if that made any difference...

hartytp, 2018-08-29:
Same after another power cycle...

[image]

hartytp, 2018-08-29:
So, the dead channel is drawing way too much current from the 30V rail. I thought we had a software interlock that should prevent that. Why hasn't that tripped in this case?

Some of the channels are also drawing surprising amounts of current on the N8V0 rail.

Channels 3 and 6 also seem to have relatively high output powers, even with their inputs terminated. So, they may also be misbehaving

hartytp, 2018-08-29:
Some of the channels also seem to be drawing too much current on the 6V0 rail.

wizath, 2018-08-29:
I'm most concerned about 7th channel drawing 0.7 A from 30 V supply

hartytp, 2018-08-29:
Yes, so it seems that this unit has died. I'll post it back to you after lunch...

gkasprow, 2018-08-29:
oscillations occur when the input cable goes too close to the PA stage. You can open the module and space it apart with some non-conductive material. Normally we glue it to the wall.

hartytp, 2018-08-29:
> oscillations occur when the input cable goes too close to the PA stage. You can open the module and space it apart with some non-conductive material. Normally we glue it to the wall.

Aah, that makes sense!

There is a reason that channel 2 (the channel that oscillates) is missing from this plot: https://github.com/sinara-hw/Booster/issues/35#issuecomment-413223627

One of my students tightened the SMA up with a torque spanner instead of fingers when adding a 50R terminator. When I removed it, the SMA cable rotated, which was probably the cause of the oscillation. So, I'll have a go at fixing that myself.

hartytp, 2018-08-29:
Still, even if that is the cause of the oscillations, there are several things that concern me:
1. Does the firmware actually implement the specification that we agreed (see the Wiki)? It appears not, since there is no software interlock on the current (this should probably apply to all supply rails). Are there other issues? 
2. Please can we ensure that I2C errors, unrealistic temperature/ADC readings, etc trigger an error condition that shuts down the channel
3. Why did the oscillation of one unit cause I2C to break?
4. If there is a 600mA current limit on each channel for the 30V supply, why is one channel drawing nearly 800mA?
5. From the UART traces above, it seems that several other channels are acting up apart from channel 2. Why did channel 2 oscillation break those other channels?

hartytp, 2018-08-29:
@gkasprow The glue used in the unit I have doesn't seem that robust and came off quite easily.

We could live with slightly lower gain (my requirement is for Urukul to be able to saturate the PA). Would reducing the gain a bit by adding some more attenuation fix this?

Otherwise, is it worth adding some better mounting in the next revision? Maybe a screw + zip tie to secure the cable and hold it away from the PA FET?

hartytp, 2018-08-29:
@gkasprow fixed the oscillations by moving the cable, here is the UART log now...

[image]

Comments:
- even with its input and output 50R terminated, channel 2 now has "IINT" high. I guess that means that the input interlock is broken or something?
- channel 7 is still drawing nearly 800mA (24W!) of 30V current.

hartytp, 2018-08-29:
Also, channel 0 shows almost no current draw on the 6V0 rail

hartytp, 2018-08-29:
Should I post the unit back to you?

wizath, 2018-08-29:
> Does the firmware actually implement the specification that we agreed (see the Wiki)? It appears not, since there is no software interlock on the current (this should probably apply to all supply rails). Are there other issues?

Current limits were mentioned couple of times, but probably just were forgotten in favor of other issues. As for now - each channel is calibrated at around 160 mA on 30 V rail on standby (no signal applied). We have window comparator on ADC, so without proper range of about 140 - 500 mA we can shut down channel for safety (since low current draw could provide information that amplifier is broken).

Bias voltage current varies on calibration factor (since there are very large differences between channels).

Next thing - reverse power interlock is listed on the Wiki. Do we need user-adjustable value for that kind of interlock? Or fixed value will be fine (what level?).

> Please can we ensure that I2C errors, unrealistic temperature/ADC readings, etc trigger an error condition that shuts down the channel

Since this issue happened twice in about two months, I'll add switch to disable channel if the temperature reading will jump from _expected_ value (since broken channel tell that temperature is 1 degree). I don't know the cause why I2C could die in a middle of operation (because calibration values were loaded and channel could be powered).

hartytp, 2018-08-29:
>  Current limits were mentioned couple of times, but probably just were forgotten in favor of other issues.

Understandable. Not a problem, so long as they are implemented before we start connecting Booster to sensitive loads.

> We have window comparator on ADC, so without proper range of about 140 - 500 mA we can shut down channel for safety (since low current draw could provide information that amplifier is broken).

:+1: 

> Bias voltage current varies on calibration factor (since there are very large differences between channels).

That's fine. AFAICT, interlocking the current is enough.

> Next thing - reverse power interlock is listed on the Wiki. Do we need user-adjustable value for that kind of interlock? Or fixed value will be fine (what level?).

Fixed-level is fine for the reverse power interlock -- that should be what's on the Wiki already, but let me know if you spot anything else.

The reverse power interlock is only intended to protect the amplifier itself, not the loads. So, this is something that Maciej should choose -- make it something which we are confident will protect the amplifier from damage.

> Since this issue happened twice in about two months, I'll add switch to disable channel if the temperature reading will jump from expected value (since broken channel tell that temperature is 1 [...]

Great, thanks!

Am I right in thinking that the new booster channels all have EEPROMs with identities? Is it worth reading back the identity EEPROM before enabling the channels or clearing the interlock status? Then we can trigger an error if the identities that are returned are not valid. That might be a good way of ensuring that I2C works before enabling the channels.

wizath, 2018-08-29:
> Am I right in thinking that the new booster channels all have EEPROMs with identities? Is it worth reading back the identity EEPROM before enabling the channels or clearing the interlock status? [...]

After powering up software detects available channels based on presence of temperature sensor, EEPROM memory and DAC - if they're online, channel can be enabled, otherwise they aren't initialized at all and could not be powered up.

hartytp, 2018-08-29:
Is it worth repeating this check every time the interlock is cleared and every time the unit is switched into/out of standby mode?

hartytp, 2018-09-01:
did you receive the v1.0 unit back yet? Hopefully it should have been delivered on Friday.

@gkasprow The main priority now needs to be taking all the failed units we have and understanding why they failed. Once that is done, we can press on with manufacture of the next batch.

gkasprow, 2018-09-01:
I'm on holidays, next week can help remotely only.

hartytp, 2018-09-01:
Good, I hope you're going somewhere nice!

The firmware @wizath sent me implements all the critical functionality I need to finish testing booster and start using it in my lab. We can finish fixing the last few issues when your back.

Have a good break!

hartytp, 2018-09-11:
Closing this issue for now, but let's open a new issue for anything related to fixing this damaged booster unit

closed 2018-09-11 by hartytp