sinara-hw/Booster #52: All channels should turn on automatically when the amplifier is powered on

hartytp, 2018-08-29:
There is no need to shut the channels down. Note that the interlocks should make it safe to automatically power all channels at startup.

cjbe, 2019-02-06:
@wizath could you implement this?

cjbe, 2019-03-22:
@wizath it would be very useful if you could implement this.

wizath, 2019-03-22:
https://github.com/sinara-hw/Booster/issues/98#issuecomment-475599581

Added, channels will be enabled shortly after power-up

cjbe, 2019-03-22:
@wizath after your fix for this sometimes the channel LEDs do not match the state of the channel. E.g. after pressing "standby" twice there are no LEDs on for channels 3-7 despite them being enabled according to the firmware.

wizath, 2019-03-22:
You've got first or second version :thinking: ?

cjbe, 2019-03-22:
I believe the second version: `built Mar 22 2019 15:07:47`

wizath, 2019-03-22:
Hmm, weird bug, I thought that fix was working after seeing the same problem :sad:

hartytp, 2019-04-09:
Is this fixed? If so, please close. Thanks!

wizath, 2019-04-09:
I'll check it last time and close

hartytp, 2019-04-10:
This seems to work fine now.

closed 2019-04-10 by hartytp