sinara-hw/Booster #53: Respond to all commands via remote interface

hartytp, 2018-08-29:
From the Wiki

```
Booster responds to all commands. Commands containing a ? query Booster for information, other commands are acknowledged with the response OK. An error string is returned in case of a programming error, such as a malformed command
```

This makes it easier to detect errors over the remote interface

wizath, 2018-08-29:
That's fixed in incoming release.

`*IDN?
ISEPW,RFPA,1.0,10-17`
`*IDN
[scpi] **ERROR: -113, "Undefined header"`

hartytp, 2018-08-29:
Thanks.

What does `INT:POW 0 30` return? It's useful if it returns something (even just a blank line or `OK`) so that we can check for errors.

hartytp, 2018-09-03:
NB this is still not implemented. `INT:POW? 0,1` does not return anything. This makes it tricky to detect error conditions (how long should one wait for an error message?). A default response of `\n` or `okay` or something like that would be better than no response IMHO.

hartytp, 2018-09-03:
Also, I get the impression that there is a limited Rx buffer, which can overflow if too many commands are sent too quickly. Having some kind of response allows users to ensure that each command is dealt with before the next one is sent

hartytp, 2018-09-04:
@wizath another comment about this: currently, Booster can currently issue more than one error message in response to a command. Again, this makes it tricky to write a robust driver. For example, if command *a* gets two errors in response, but the user only reads one error message then there is still an error message left in the buffer. if the user now issues command *b* then they will incorrectly interpret the second error message as a response to command *b*. Since there is no EOM in the interface, the user has no way to check for more than one error message from *a* apart from adding a quite large timeout (say, 1s) which is not acceptable in many cases.

It would be nice to have some unambiguous way for a user to know that a command has executed and that they have read all response from the device. There are a few ways of doing this (e.g. use a blank line as an EOM) and I don't mind which one you use...

wizath, 2018-09-04:
Yes, I've noticed that - I think one error is sufficient

wizath, 2018-09-04:
Fixed, only first error is returned, rest is discarded.

hartytp, 2018-10-10:
@wizath was this fixed already, can we clsoe it?