sinara-hw/Booster #55: secure input RF coax

hartytp, 2018-08-29:
If this moves too close to the PA FET it can cause the amplifier to oscillate (destructively?). Currently, it's secured via glue, but I'm concerned that this won't be enough (particularly at higher temperatures). Is it worth adding a better way of fixing this in place, such as a zip tie to anchor it to the enclosure side wall?

hartytp, 2018-08-31:
NB this coax moving seems to be a real problem in all the current units. I've added hot glue to some of them to hold it out of the way, but I think we need to find a more robust solution going forwards.

gkasprow, 2018-10-20:
We will add stand-off with some shelf to keep cable reliably 10mm above the board.

closed 2018-10-20 by gkasprow

hartytp, 2018-10-22:
@gkasprow I would have done this by adding a small screw to the side wall and zip-typing the coax to that. But, if your way is easier/cheaper then it sounds like a good solution.