sinara-hw/Booster #56: Booster v1.1 testing

hartytp, 2018-08-31:
Booster v1.1 here and looking really nice. The mechanics feel vastly improved from the initial version.

One minor niggle is that it would be slightly nicer if the screws on the FP were CS rather than cap head. Otherwise, it looks great. 

Also, the lid was a bit tricky to remove because it fits so snugly (had to stick a screwdriver through the fan holes to push it out). However, once I did get it off I was rewarded by seeing how beautiful it all looks inside. I'm really impressed.

hartytp, 2018-08-31:
From a mechanical standpoint: I've done some quick disassembly (removed an amplifier module lid, removed an entire module) and I'm really struggling to find things I don't like about it. Very well done.

hartytp, 2018-08-31:
Actually, one other thing: the (M3?) internal CSK Phillips  screws were a bit tricky to remove. Not sure if they were threadlocked or just done up very tightly, but some of them started to strip. Using cap head screws (and maybe even increasing the screw size a touch) would probably make life easier.

hartytp, 2018-08-31:
Also, channel 0 was a little tricky to remount. I had to remove the lid to get the D-type to remate with the PCB.

hartytp, 2018-08-31:
Related to the previous comment, the amplifier mounting feels a little over-defined due to the use of CSK screws to hold the amplifier channels down. Might be better to use cap head screws with a generous clearance hole to give some slack.

hartytp, 2018-08-31:
@gkasprow @marmeladapk here is the UART output when I enable it: 

[image]

- channel 2 has a very low bias current, and the interlock seems to trip straight away
- channel 6 has a very high bias current
- channel 7 has a low bias current

any ideas?

wizath, 2018-08-31:
Did you swap any module?

hartytp, 2018-08-31:
I don't believe so! I wrote numbers on them before removing, so the worst case would presumably be if I had swapped 7 with 0, 6 with 1, etc. However, looking at the bias current measurements, I don't think that would explain what we see. Do you?

wizath, 2018-08-31:
I've observed that calibration is channel specific - e.g if you calibrate module in slot 1 then insert it in slot 2 it won't be the same

hartytp, 2018-08-31:
However, maybe we should make the UART report the unique ID of each channel, and store the channel ID along with the calibration data, to allow this kind of think to be checked.

hartytp, 2018-08-31:
Okay, let me try swapping them around and see what happens...

wizath, 2018-08-31:
I'll add ID reporting

hartytp, 2018-08-31:
hmm...in fact, I can't have swapped the channels around. I can tell by the stickers that TechnoSytem put on them (the stickers bridge the gap beween adjacent amplifier channels, so if you swap the channels around the stickers aren't in the right places any more). So, I'm pretty confident that they are all in the same places that I received them in..

hartytp, 2018-08-31:
[image]

See this photo. If I swap 7 with 0 then the stickers don't bridge the units correctly

PS these electro polished RF modules are so shiny I feel like I need to be in a clean room to touch them

wizath, 2018-08-31:
Owwww, they could swapped them....

hartytp, 2018-08-31:
> Owwww, they could swapped them....

? how?

hartytp, 2018-08-31:
Did you record the channel IDs when you calibrated them? that would allow us to check.

wizath, 2018-08-31:
They took it from me to assembly them, and probably swapped them all. But its simple fix - I'll write you the details about current calibration since it's super easy

hartytp, 2018-08-31:
Great, thanks! Can you post the details on the Wiki, please?

Do we also need to recalibrate the Tx/Rx power measurements?

wizath, 2018-08-31:
No, I don't think power measurement recalibration is needed.

You can connect 50 Ohm terminator to input or leave it unconnected. Then you type `stop` in console to stop status messages from appearing (the're disabled by default in new version). Then you simply call `calbias <ch number> <desired current>` and wait if the process will catch the selected value.

hartytp, 2018-08-31:
> You can connect 50 Ohm terminator to input or leave it unconnected. Then you type stop in console to stop status messages from appearing (the're disabled by default in new version). Then you simply call calbias <ch number> <desired current> and wait if the process will catch the selected value.

Nice! is the bias current in A or mA? i.e. is the calibration process you did `calbias 0 0.16`?

I'm glad you implemented this nicely -- at some point I'd like to play with running these amplifiers at a lower bias current to reduce the power consumption at the expense of less gain and lower P1dB. This will make that process very easy

wizath, 2018-08-31:
in mA's, `calbias 0 160`

wizath, 2018-08-31:
Anyway, I think that this option should be available to end-user since current calibration can't really be done properly in production.

hartytp, 2018-08-31:
> Anyway, I think that this option should be available to end-user since current calibration can't really be done properly in production.

Why not?

hartytp, 2018-08-31:
@wizath for channel 2, calibration seems to have failed. I see `DAC 15 CURRENT 120` Have you seen that before?

> Anyway, I think that this option should be available to end-user since current calibration can't really be done properly in production.

If the calibration alogrithm were improved a bit (large steps first, then smaller steps) so that it was a bit faster then the amplifier could just autocal on power up...

hartytp, 2018-08-31:
Channel 7 calibration also fails:
`DAC 15 CURRENT 146`

wizath, 2018-08-31:
> > Anyway, I think that this option should be available to end-user since current calibration can't really be done properly in production.

> Why not?

If we calibrate whole set (box + 8 modules) it's fine, but if user wants to replace one unit, it must be calibrated again after inserting it to the slot - because AFAICT current calibration is slot and module dependant.

hartytp, 2018-08-31:
> That's unfortunate - we need to better set the voltage, by using the potentiometer on module, so there is more range to calibrate on (since calibration values on some modules are very close to the DAC limits).

Okay, maybe we need to adjust these limits for the next revision.

hartytp, 2018-08-31:
> You can adjust the current, by using potentiometer on RFPA_PA_OUT board, then recalibrate.

Where is that? Do you have a photo?

wizath, 2018-08-31:
[image]
It's the POT1

hartytp, 2018-08-31:
Okay, I'll look at that. Any advice as to how I should tweak it (which direction, how much)?

hartytp, 2018-08-31:
FWIW, in that channel, the coax was resting on the PA chip. I'll put some hot glue to hold it away.

wizath, 2018-08-31:
Oh, I forgot - another command is `bias <channel> <value 0-4095>` which enable you to control bias voltage. You can set it to 0, then turn the potentiometer to lower the voltage (if there is not enough current). You can measure negative voltage on the pot's leg close to C2 silk. It should be around -0.8-1 V, then reapply calibration.

wizath, 2018-08-31:
Next batch will be calibrated to the center of DAC range, so there will be more room to adjust the current.

hartytp, 2018-08-31:
> Oh, I forgot - another command is bias <channel> <value 0-4095> which enable you to control bias voltage. You can set it to 0, then turn the potentiometer to lower the voltage (if there is not [...]

I don't have a wiring harness to give me access to the pot while I do that. I could make something but it would take time

wizath, 2018-08-31:
Oh, I forgot about top board blocking the way. I'll tell you which way to turn to increase current in a second

wizath, 2018-08-31:
Maciej suggested that top board should have hole for screwdriver to adjust the pot.

hartytp, 2018-08-31:
> Maciej suggested that top board should have hole for screwdriver to adjust the pot.

:+1: and a note on the silk screen indicating which direction to turn for more juice

we should also consider changing the dividers to give a large range of bias currents without a pot. Or, even scrap the pot (do we need it when the DAC is 16-bit?)

wizath, 2018-08-31:
[image]

So you turn in direction of "-" to get more current, and opposite to get less current. Be gentle, and turn only several degrees. DAC is 12 bit

hartytp, 2018-08-31:
> DAC is 12 bit

Would it be easier to use a higher-resolution DAC in the next revision and scrap the pot? One fewer thing that has to be adjusted by hand.

hartytp, 2018-08-31:
Thanks. That works

hartytp, 2018-08-31:
Any idea why channel 2 interlock trips if the input isn't 50R terminated? No other channels do that.

hartytp, 2018-08-31:
[image]

hartytp, 2018-08-31:
Okay, thanks.

One more thing. Look at this:

[image]

why is channel 7 so hot?

hartytp, 2018-08-31:
hmm...unless it's the hot glue I put in to hold the cable down keeping the sensor warm

hartytp, 2018-08-31:
Okay, cool. Cross-talk is better than 100dBc up to 400MHz. Nice! I'll post proper data on that later.

So, that more or less concludes all the measurements of analog performance I need to make. Performance is as good/better than I would have expected/hoped for by every measure! Well done to everyone involved.

Just need to check the interlock + power measurements and I can start using this in the lab.

hartytp, 2018-09-03:
Gain for all channels:
[image]

- channel to channel variation in gain is about +-0.5dB so, within spec
- between 50MHz 450MHz the gain variation is just over +-1dB so, again within spec

hartytp, 2018-09-03:
NB output power for that measurement was approx 1W (-10dBm + 40dB) so, while not a P1dB measurement, it's a good indication of the cross-talk at high output power...

hartytp, 2018-09-04:
Some quick stress testing:
- turn Booster on/off a couple of dozen times with various gaps between on/off. done with rf applied to all channels (0dBm/all outputs 50R terminated) and with no RF applied.

hartytp, 2018-09-04:
Quick test of output power interlock using RF at 200MHz. Interlock set to +37.0dBm. Actually trips at +35.7dBm. This is consistent with https://github.com/sinara-hw/Booster/issues/27#issue-341533805 where the power measures about +1.3dB to high on the power detector. So, that seems to work within spec! Good!

wizath, 2018-09-04:
HW interlock is set at 36 dBm

hartytp, 2018-09-04:
So, do you mean that the output power interlock is currently implemented in software, and that the hardware interlock (via the comparator) has a fixed threshold of 36dBm? I would much prefer it calling the `INT:POW` command also adjusted the hardware interlock (that way, the interlock can respond very fast to protect sensitive loads)

hartytp, 2018-09-04:
Spent some time trying to break Booster by power cycling and turning it in/out of standby mode with and without input power (at varying levels) applied. So far, it all seems to be very robust and we don't seem to have managed to break anything. Great!

NB we always kept the outputs 50R terminated, as the reverse power interlock hasn't been implemented yet

hartytp, 2018-09-04:
Looking for issues, such as PSU issues, when all channels are fully loaded. Running all outputs at 4W, 200MHz and looking at RF on a spectrum analyser.

[image]

[image]

hartytp, 2018-09-04:
Same but very narrow span. Peak is 35dBm. See some spurs at -100dBc ish level. Will check whether they are on the synth or not.

[image]

[image]

hartytp, 2018-09-04:
Looked at large signal cross-talk by driving ch 3 at 4W, 200MHz. All other channels at 201MHz, same RF power. The spur (combined cross-talk) is -92dBc. So, cross-talk with all channels loaded at max power is roughly as expected from the previous measurement. Good!

hartytp, 2018-09-04:
Here is the synth on the spectrum analyzer.Carrier is -5dBm. Noise floor is a bit worse for this measurement, but at least the major spurs we saw were due to the synth, so I currently have no evidence of any spurs on Booster's output. We'll check this more carefully at some point in the future with a phase noise meter if we find the time.

hartytp, 2018-09-05:
A different kind of cross-talk check:
- looking for slow thermal cross-talk or things like ground plane/power line sag when other channels are switched on
- ran channel 3 at 4W, 200MHz and let warm up while monitoring output power via a V3500A power meter
- once that had stabilized, we applied the same RF power to all the other channels and continued logging
- I don't see anything I can measure on a seconds timescale. On a timescale of about 1 hour, I see a 0.5dB decrease in the output power of channel 3. This is consistent with a change in temperature of a couple of degrees due to the other channels heating up a bit.

So, that looks good.

hartytp, 2018-09-05:
Edit: fixed previous post -- the change in output power of channel 3 when all other channels are run at max power is 0.05dB, *not* 0.5dB (and that drift takes something like an hour). i.e. there is almost no thermal cross-talk to speak of!

hartytp, 2018-09-05:
Also checked for fast glitches pm channel 3 caused by turning all other channels on, by looking with a 423B diode detector (see transient measurements I posted previously for the setup). Wasn't able to see anything.

So, the channel-channel isolation really is awesome!

Jochenwolf3, 2018-09-05:
Further to post https://github.com/sinara-hw/Booster/issues/56#issuecomment-418664006, here is the power drift of a single channel driven at 200 MHz. Initially, only that channel is being driven to output 4W. After 40 min, all other channels are being driven to output 4W, too. 
After an hour, the resulting drift is just 0.05 dBm, so pretty good!
[image]

[17 more replies]

closed 2018-09-11 by hartytp