sinara-hw/Booster #57: Small mechanical issues

hartytp, 2018-08-31:
Overall I'm very happy with the units provided by TechnoSystems. Some minor feedback:
- The front-panel cap head screws look a bit ugly. Counterbore or countersunk screws would look much better. If we go for CSK then we need to watch out for potentially overdefining the mechanics
- the lid was a little tricky to remove. partially because it's a very snug fit and partially because there is no way to get a finger/screwdriver under it. I ended up having to stick a screwdriver through the fan holes to get the lid off!
- the lids for the amplifier modules seem to collide a bit with the front wall of the chassis. I think this is because it's a very snug fit and the tolerance on those PCB-mount D-types isn't perfect. Not sure if there is an easy way of leaving a small gap between the front of the lid and the front of the chassis
- channel 0 is still hard to mount as the PCB-mount D-type wasn't in quite the right place. not sure if there is an easy fix to it. generally, the channels get easier to mount as the channel number increases. In the end, I was only able to mount channel 0 by removing the lid, and then fitting the lid after it had mated with the PCB-mount D-type
- for internal screws, please avoid using CSK screws where not necessary, as they lead to overdefined mechanics. e.g. do the screws that connect the amplifier modules to the mounting bar need to be CSK? When I assembled this, it felt like there was a lot of force on the screws due to things being overdefined
- please try to use allen key screws instead of phillips and don't make internal screws smaller than necessary

hartytp, 2018-08-31:
See also https://github.com/sinara-hw/Booster/issues/55

hartytp, 2018-09-06:
One other thought, but is it worth adding a few external tapped holes on the amplifier modules that can be used for mounting them "standalone" i.e. outside the main Booster chassis?

These amplifier modules turned out better than I expected and I can see a few cases where it would be nice to use just one or two units outside the chassis. We would supply power via the D-type and either override the interlock by pin strapping or hook up some microprocessor for control

gkasprow, 2018-09-06:
One can use holes that fix the lid. Just use longer screws. We can make tiny board with just regulators + power entry. We can also mount dc/dc converters to make it working with single supply. Otherwise such module can be easily damaged by missing negative power

hartytp, 2018-09-11:
When mounted in a rack, the Booster unit sags a bit under its own weight, meaning that it would bump into the unit mounted directly below it. See this photo...

[image]

gkasprow, 2018-09-11:
We can make front panel slightly thicker on the sides only. This won't affect box dimensions.

hartytp, 2018-09-11:
also https://github.com/sinara-hw/Booster/issues/92
and https://github.com/sinara-hw/Booster/issues/91

hartytp, 2018-09-11:
more packaging than mechanical, but:https://github.com/sinara-hw/Booster/issues/89

gkasprow, 2018-09-11:
Pawel found the reason why the right-most amplifier is more difficult to install. The connector on PCB is slightly misplaced. We will fix it in next release.

hartytp, 2018-09-12:
good catch

hartytp, 2018-09-12:
good catch

hartytp, 2018-09-14:
I suspect that the issues we had after heating a Booster module up were all due to mechanical stress (the temperature sensors reported that the modules never got above 40C so it's unlikely to be thermal damage/anything melting).
 
We really need to fix this for the next version. Do you thin it's worth adding slots in the control PCB to release stress, and then letting each module screw directly into the main PCB via the D-type jack screws?

gkasprow, 2018-09-15:
I will carefully inspect the modules on Tuesday. There can be several issues:
- too short pins that connect two boards. This would explain why you need to play with them to make them working
- shorts to chassis
- contamination of contacts during cleaning after assembly
- cold solder joints 
We did not observe such behaviour with previous modules.

hartytp, 2018-09-15:
Thanks Greg! Do you have a Booster v1.1 with all 8 channels populated? I think that may be important for reproducing this issue since there is most mechanical stress when all channels are mounted. You could also try taking them out and replacing a few times and maybe heating some channels up with a hot air gun. Anything that causes mechanical stress may show up these issues

hartytp, 2018-10-20:
Is this all done now? If so I think we're ready for review.

gkasprow, 2018-10-20:
I'm just finishing last cosmetic things on PCBs..

gkasprow, 2018-10-20:
@hartytp I published rev1.3.
Main board is finished.
RF and suppl. boards need some cosmetics.

hartytp, 2018-10-21:
Great! Is the mechanical stuff in this issue sorted? If so ease close. Thanks.

hartytp, 2018-11-05:
@gkasprow is this all fixed? If so, please close.

hartytp, 2019-03-07:
also to check are the input coax fixing and the D-sub support https://github.com/sinara-hw/Booster/issues/137#issuecomment-454726446

hartytp, 2019-03-08:
Apart from the Dsub height question and the missing D-shaped SMA cutouts, this all looks good!

closed 2019-03-08 by hartytp

hartytp, 2019-03-08:
all fits together beautifully. Nice job all.

hartytp, 2019-03-08:
One thing though: when we took it to pieces to tighten the SMAs, we noticed that the supplemental control board on one channel is not screwed into the module (no screws inserted). OOps...

gkasprow, 2019-03-08:
The screws are not really important. The chassis is designed in such way that board is squeezed between the box and the lid. The only function is maintaining electrical contact. If you don't see any performance penalty, it means that they are not really necessary. However we did not decide to scrap them yet, so they should have been installed.