sinara-hw/Booster #60: UART add longer docstrings

hartytp, 2018-08-31:
The UART interface has nice docstrings for all commands (when one types `help`). These are great! But, they don't give any information about any of the commands. e.g. the `calbias` command does not say that it takes two arguments, and that the bias current is in `mA`. Is it worth making this help message more verbose? It's probably better to make this self-documenting than to document it on the Wiki!

We could also consider adding a `help?` command to the ethernet interface, which prints a similar help message with a list of ethernet commands, and their parameters. This would replace the documentation on the Wiki

wizath, 2018-10-25:
I've added `help <command>` to display more detailed information for specified command.

closed 2018-10-25 by wizath

hartytp, 2018-10-25:
Nice! Thank you!