sinara-hw/Booster #67: reverse power interlock

hartytp, 2018-09-03:
hasn't been implemented yet AFAICT

hartytp, 2018-09-03:
@wizath we definitely need to implement this before producing the next hardware version, as I want to check that the hardware for this actually works without any unexpected issues

hartytp, 2018-09-04:
This is currently my top priority for Booster, as I have no idea how much reverse power Booster can take without damage/what the stability circle is for the PA (wouldn't want a mismatched load to cause it to oscillate).

hartytp, 2018-09-11:
@wizath this is our next top priority since we are using resonant AOMs, which can easily have a very low return loss if the RF frequency is set incorrectly.
- at the moment there is no specification for what reverse power Booster can take, or how fast the interlock needs to be (is high reverse power okay for ms? or can it only be ns?)
- the interlock isn't even implemented in software afaict
- can we reduce the averaging time to make the interlock respond faster. I assume that if we average for a few ms the noise is already quite low

wizath, 2018-09-11:
At 500 samples, we've got measurement each 270~ ms
At 50 samples, we've got measurement each 27~ ms
At 10 samples, we've got measurement each 6~ ms

Adding a few ms for task switch, we could get around 10-20 ms response time for 10 samples.

hartytp, 2018-09-11:
> Adding a few ms for task switch, we could get around 10-20 ms response time for 10 samples.

Sounds good to me, but does Maciej think it's fast enough to protect Booster in case of full reverse power?

hartytp, 2018-10-20:
- [ ] document what reverse power level the interlock trips at and how long it takes to trip

wizath, 2018-10-24:
As I was checking reverse power measurements, with disconnected cable we get equal readings between TX and RFL power. We could use that to disable channel if RFL power matches TX with some margin.

hartytp, 2018-10-24:
@wizath I'm happy for you to do this however you/@gkasprow/Maciej think would be best.

FWIW though, my guess is that it's better to have an interlock on the absolute reflected power instead of the return loss (TX-RX). The only point of this interlock is to prevent Booster from dangerous levels of reverse power. Very low return loss can be tolerated if the absolute reflected power is low (so long as this doesn't cause the amplifier to oscillate, but that should be caught by the forward power interlock, so it's not really a problem).

closed 2018-10-31 by wizath

hartytp, 2018-10-31:
:fireworks: (but please make sure that this is correctly documented in the Wiki. e.g. what power level does the interlock trip at?).