sinara-hw/Booster #68: bias current interlock

hartytp, 2018-09-03:
needs to be implemented in firmware.

should disable a channel if its bias current is too high or too low. The threshold for this interlock should take into account the normal variation in bias current over temperature (I can measure this with a hot air gun if necessary), so that it only trips if the bias current is outside the normal operating range.

hartytp, 2018-10-24:
We need to be a bit careful about this: due to amplifier non-linearity, the bias current changes with the output power. One one channel I was looking at, the bias current changes from 160mA with no signal applied to 350mA at 36dBm output power. We need to make sure that this doesn't trip a current limit

wizath, 2018-10-26:
ATM overcurrent protection is set at 600mA. I can lower it to about 400-500 mA's but need to know maximum current consumption with signal.

hartytp, 2018-10-27:
That sounds right. At max rf power the modules can draw close to 400mA with some variation between units and we don't want the interlock to trip accidentally.

600mA for a few ms won't damage anything. The interlock is nice since the user gets a clear error + red led so knows what the problem is.

Should the interlock also trip if the current is too low? That way the user gets a red LED if the FET dies and stops drawing power. Or is that a waste of time?

wizath, 2018-10-30:
Checked hardware threshold for detecting too low current - direct detection via ADC thresholds is not reliable enough, I'll leave only overcurrent protection ATM

closed 2018-10-30 by wizath

hartytp, 2018-10-30:
Cool, thanks!

hartytp, 2018-10-30:
> Checked hardware threshold for detecting too low current - direct detection via ADC thresholds is not reliable enough, I'll leave only overcurrent protection ATM

I see, so it's hard to check if the current is << 160mA. So, currently the firmware only monitors the current being >600mA. If the current is <<160mA the user is not alerted. Is that correct? (That's fine, just wanted to check)

wizath, 2018-10-30:
It's not so hard, but when ADC is enabled to detect lower threshold at around 100mA's it gives so many false positives at startup sequence. I'll try to come up with something more clever after I'll fix the SPI/Ethernet bug.

tl;dr yes, the only protection at this moment is after exceeding 600 mA current.

hartytp, 2018-10-30:
> It's not so hard, but when ADC is enabled to detect lower  threshold at around 100mA's it gives so many false positives at startup  sequence. I'll try to come up with something more clever after I'll fix  the SPI/Ethernet bug.

I see, yes. We'd need to only enable this after startup to avoid false positives.

Well, maybe don't bother then. The users will know something is wrong when there is no gain. Once that happens, they can check the UART...

I'm keen to finish this off and don't want to delay things with too many additional features -- there is always some improvement one can make, but at some point one has to call projects finish...