sinara-hw/Booster #69: SCPI error

hartytp, 2018-09-03:
writing `b'INT:POW 0 1.0\n' to the ethernet port gives the error '[scpi] **error: -101, "invalid character"'` after that, I get `connection refused` on subsequent attempts to open a connection to the device until I power cycle it.

Thoughts @wizath?

hartytp, 2018-09-03:
@wizath NB I can't do any more testing until this is fixed, so this is a top priority for me atm

hartytp, 2018-09-03:
`meas:tem?` gives an undefined header error...

hartytp, 2018-09-03:
Still, the unit seeming to crash is an issue...

wizath, 2018-09-03:
Strange, will check tommorow but I've tested it for some time and never encountered something like this.

wizath, 2018-09-04:
```
*INT:POW? 0,1
[scpi] **ERROR: -101, "Invalid character"
[scpi] **ERROR: -113, "Undefined header"
```

```
INT:POW 0 1.0
[scpi] **ERROR: -101, "Invalid character"
[scpi] **ERROR: -101, "Invalid character"
[scpi] **ERROR: -101, "Invalid character"
```
Everything still works after these commands. Could you post a code snippet that causes the error?

hartytp, 2018-09-04:
I think the problem is actually related to either sending the commands too fast, or to not reading the response back. Here is a WIP version of a Booster driver: https://hastebin.com/isimutilic.py

running this with the `time.sleep(0.3)` on line 45 commented out, I see
```
(artiq-dev) ion@ion-OptiPlex-990:~/scratch/artiq$ python -m oxart.devices.booster.driver
Traceback (most recent call last):
  File "/home/ion/anaconda3/envs/artiq-dev/lib/python3.5/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/ion/anaconda3/envs/artiq-dev/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/ion/scratch/oxart/oxart/devices/booster/driver.py", line 165, in <module>
    main()
  File "/home/ion/scratch/oxart/oxart/devices/booster/driver.py", line 158, in main
    print(','.join([str(pa.get_output_interlock(ch)) for ch in range(8)]))
  File "/home/ion/scratch/oxart/oxart/devices/booster/driver.py", line 158, in <listcomp>
    print(','.join([str(pa.get_output_interlock(ch)) for ch in range(8)]))
  File "/home/ion/scratch/oxart/oxart/devices/booster/driver.py", line 88, in get_output_interlock
    return float(self._cmd(channel, "INT:POW?"))
  File "/home/ion/scratch/oxart/oxart/devices/booster/driver.py", line 54, in _cmd
    cmd, response))
Exception: Unrecognised response to INT:POW?: '[scpi] **error: -101, "invalid character"'
(artiq-dev) ion@ion-OptiPlex-990:~/scratch/artiq$ python -m oxart.devices.booster.driver
Traceback (most recent call last):
  File "/home/ion/anaconda3/envs/artiq-dev/lib/python3.5/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/ion/anaconda3/envs/artiq-dev/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/ion/scratch/oxart/oxart/devices/booster/driver.py", line 165, in <module>
    main()
  File "/home/ion/scratch/oxart/oxart/devices/booster/driver.py", line 156, in main
    pa = Booster("10.255.6.216")
  File "/home/ion/scratch/oxart/oxart/devices/booster/driver.py", line 22, in __init__
    self.sock.connect((ip_addr, 5000))
ConnectionRefusedError: [Errno 111] Connection refused
```

I cannot open a new connection to the device without power cycling it. If I uncomment that `time.sleep(0.3)` then I no longer get the connection refused errors...

wizath, 2018-09-04:
Found an issue to resolve that 'Connection refused' calls, still testing the performance.

hartytp, 2018-09-04:
Nice. Thanks

closed 2018-09-12 by hartytp