sinara-hw/Booster #70: ethernet interface slow

hartytp, 2018-09-03:
Currently, the interlock status query `int:stat?` seems to take 200ms to respond. Why so slow?

gkasprow, 2018-09-04:
It's probably related to power detector readout and averaging + multiple I2C access. @wizath please confirm.

hartytp, 2018-09-04:
@gkasprow:
- even if that's the case, I'm surprised that it's really as slow as 200ms per operation
- the command `int:stat?` should just return the current interlock status register in the CPU. No measurements are needed for that. 
- even for power measurement commands, it's probably fine to return the (cached) result of the last measurement.
- taking 1s to check the interlock status of all channels is a bit annoying for the user

wizath, 2018-09-04:
I'm worried that the SCPI library is taking so much time parsing input query. SW is returning status from memory so it should be quick - I'm trying to optimize this down.

hartytp, 2018-09-04:
@wizath makes sense (although I have no idea how it can possibly take that long!).

Anyway, let's not worry about this for the time being. Most important is getting the interlocks finished.

wizath, 2018-10-30:
Found issue, will fix shortly

wizath, 2018-11-09:
Fixed, response time is about 5 ms~

closed 2018-11-09 by wizath