sinara-hw/Booster #71: scpi interface does not match the wiki

hartytp, 2018-09-03:
The current scpi interface is close to the description on the wiki, but not an exact match. e.g. if no interlock has tripped, `int:stat? 0` returns `0`, not `okay` (as on the wiki). That behavior is absolutely fine (probably better than the description on the wiki) but the fact that they don't match is a problem, as writing a driver for booster becomes trial and error to reverse engineer the protocol

wizath, 2018-09-03:
To help you with reverse engineering - chan:enab without parameters will enable all channels, with simillar fashion for other commands.

hartytp, 2018-09-03:
Thanks. If it's not much more work, what do you think about removing the commands list from the wiki, and adding a detailed "help" function to booster? That way, it's easier to track changes across firmware versions. NB this needs some clear way to indicate the end of the message if '\n' characters are used in the help string...

wizath, 2018-09-03:
That's the plan, but at first I would like to replace that CLI library that is weird to work with.

hartytp, 2018-09-03:
okay

hartytp, 2018-09-03:
sounds like a good plan...

hartytp, 2018-09-04:
I'll put SCPI command testing on hold until this issue is resolved (as I'm not having much luck guessing how the interface works).

hartytp, 2018-09-10:
@wizath is readout of the forward power implemented yet? What's the command for that?

Also, is readout of the module temperatures implemented yet? If so, what's the command?

Thanks!

wizath, 2018-09-10:
```
MEASure:TEMPerature?`
MEASure:OUTput?
MEASure:REVerse?
```
If you leave it without arguments it will return values for all available channels - if you specify channel e.g `meas:temp? 1` you'll get only one value. There are so many tasks that I always forget to update the wiki

hartytp, 2018-09-10:
@wizath Great, thanks! @jochen.wolf3 can you integrate that into our Booster driver tomorrow, please?

> There are so many tasks that I always forget to update the wiki

Don't worry, I know how busy you've been. Anyway, it makes sense to sort out all the tricky things first and then do the wiki.

wizath, 2018-10-30:
Fixed docs for SCPI commands, also added CLI commands definitions. @hartytp you can check and close if you are satisfied. Should we leave BIAS command for SCPI interface?

hartytp, 2018-10-30:
what do you mean by 'blank'?

wizath, 2018-10-30:
eg. if you type `CHAN:ENAB` argument will be _blank_ and will enable all channels.

hartytp, 2018-10-30:
aah, I see.

Personally, I'd prefer not to have the blank option. I prefer users to have to explicitly enable all channels by adding an "all", as it avoids accidents.

wizath, 2019-01-14:
Can we use something like 'chan:enab 255' instead of ALL? Since after acquiring the argument it gets popped from stack and it's tricky to check if it's the string or number.

hartytp, 2019-01-14:
Okay, a bit mask here is fine. (1 for channel 0, 3 for channels 1 and 2, etc).

hartytp, 2019-01-14:
But, let's make sure this is consistent across the API

wizath, 2019-01-14:
OK, I'll make this change to every function that uses channel numbering

hartytp, 2019-01-31:
> while will enable all of them.

missing word there?

hartytp, 2019-01-31:
FWIW, it's you don't have to implement this the way I suggested on the Wiki -- a bit mask is fine if that's easier. The thing that *is* crucial is: the documentation/wiki must match the implementation; inputs must be validated appropriately.

wizath, 2019-01-31:
> missing word there?

word "all"

I thought that bitmask would be inconsistent between all commands, since for example **int:pow** requires channel number <0-7> and it would be better to keep all commands using channel number instead of bitmask to avoid confusion?

hartytp, 2019-01-31:
> word "all"

Thought so, just wanted to check.

> I thought that bitmask would be inconsistent between commands, since (for example) int:pow requires channel number <0-7> and it would be better to keep all commands using channel number instead of bitmask to avoid confusion?

Agreed, using channel number/all is better. AFIACT, it's also more common in SCPI implementations for T&M kit.

I just meant that if it was problematic to implement that then we'd be fine with (correctly implemented/validated/documented) bitmasks. If you've already done it the better way then that's great.

wizath, 2019-04-15:
https://github.com/sinara-hw/Booster/wiki/Remote-programming-interface

hartytp, 2019-04-15:
@wizath thanks a lot!

A few comments
1. please can you check that the high-level descriptions, such as https://github.com/sinara-hw/Booster/wiki#features, https://github.com/sinara-hw/Booster/wiki#interlocks and https://github.com/sinara-hw/Booster/wiki#front-panel-switches are accurate? I think they're wrong now since we changed some details of the firmware behaviour. Basically, please re-read the home page and make sure that all info is correct.
2. Is there anything in this section that's not in the new page you added? https://github.com/sinara-hw/Booster/wiki#remote-programming-interface If this is all covered in the new section, can you delete it from the main page and add a link on the side bar?
3. Let's delete the legacy commands description https://github.com/sinara-hw/Booster/wiki#legacy-command-description
4. Let's move firmware update to a separate page and just add a link to the side bar
5. Is there any info here https://github.com/sinara-hw/Booster/wiki#legacy-command-description that isn't in the VCP help menu? i.e. does the VCP help detail all commands and what arguments they take? If so, please remove the content from the Wiki and replace it with details on how to connect to the VCP and how to access help/command info via the VCP interface.

hartytp, 2019-04-15:
Also, if anyone has a nicer photo of a working booster unit (@klickverbot ?) then can they update the one on the Wiki please? Thanks!

hartytp, 2019-05-08:
Nominally fixed. I'll test once our next batch of Boosters arrive and post any problems I encounter as new issues.

Thanks @wizath!

closed 2019-05-08 by hartytp