sinara-hw/Booster #72: input validation

hartytp, 2018-09-04:
all commands must perform input validation (as specified on the wiki).

For example, currently setting the output power interlock to -1dBm actually puts it to 255dBm, which could obviously be pretty catastrophic. If the interlock is too low, or above 5W, an error should be returned and the interlock should not be adjusted.

All channel numbers need to be validated as well and, ideally, a helpful error string should be returned...

hartytp, 2018-09-04:
- well, maybe allow the interlock to be a dB or two above 5W, so that 5W output power is reliably possible without interlocks tripping
- aah, currently setting the interlock to >=39dBm also puts it to 255dBm without error!

wizath, 2018-09-04:
To clarify - ATM valid interlock value is between 0 dB and 39 dB. Value 255 means that interlock is disabled - as you can see in status printout. I thought that 255 is better for representing disabled interlock than 0. Anyway, we can replace number for something like 'Disabled' if that suits you.

hartytp, 2018-09-04:
@wizath IMHO, it should not be possible to ever disable the interlock, as this is required to protect the PA from being damaged. The max value for the interlock should be chosen so that the interlock does not trip when the devices is operated at max RF output power (5W IIRC, but check with Maciej), but also sufficiently low that the max output power is within the safe range for the circuit. 39dBm sounds right for this.

hartytp, 2018-09-04:
However, beyond that, I still think that the current behavior is extremely dangerous. If the user tired so set the interlock to -1dBm, and that disabled the interlock without giving them an error message, that could easily lead to a sensitive load being damaged. Attempting to set the interlock to an invalid number should leave the value unchanged, but just return an error (as described on the wiki)

wizath, 2018-09-04:
But for protection there is always hardware interlock that is set to 36 dBm ATM

wizath, 2018-09-04:
There is HW based interlock. If you want to keep the SW one enabled what should be default value for it?

hartytp, 2018-09-04:
> There is HW based interlock fixed at 36 dBm. If you want to keep the SW one enabled what should be default value for it?

So, the behavior should be:
- The hardware interlock has a user-adjustable value, configured by the `INT:POW` command. This command always saves the value to EEPROM, and the value is restored on device power up
- As redundancy (in case the hw interlock fails for some reason) we can also keep the software interlock (although, that's not crucial if it's not simple to do). This should use exactly the same threshold as the hw interlock, and be configured in the same way
- it is not possible to disable either interlock, the most that can be done is to set them to +39dBm (or, some other safe value)
- `INT:POW` does input validation, and returns an error without adjusting the interlock value in case of errors (such as a power setting that is too high/too low)

If you want to scrap the SW interlock, that's fine too (but, please make sure you document what is actually implemented).

My next job is to test the interlocks carefully, so this is my top priority atm!

hartytp, 2018-09-04:
Okay, well I'll put interlock testing on hold until this issue is resolved.

hartytp, 2018-09-04:
> It's hard to set HW interlock for any power level you desire, since every module has different values for specified power level. That's why we calibrate interlock at fixed point to ensure accuracy. We "could" guess the DAC values based on power readout calibration, but I don't know how accurate that would be.

Okay, I'd misunderstood that. I thought that after calibration, the power detector's comparator was as accurate as the power detector's measurement output. Do you think that the comparator output is still less accurate after calibration?

closed 2018-10-10 by wizath