sinara-hw/Booster #73: output power hardware interlock

hartytp, 2018-09-04:
IMHO the main function of the hardware interlock is to protect the load, not the amplifier. So, if it has a fixed value then it's not much use.

If the comparator isn't as accurate as the voltage output after calibration then we have two options for the next hardware revision: (a) scrap the hardware interlock; (b) add an external comparator

I'd prefer to go for (b) as I think that having a fast hardware interlock is very valuable to protect sensitive loads.

In the mean time, I'd be curious to test out how accurate the hardware interlock is after calibration using the current hardware revision to see if this is really a problem!

wizath, 2018-09-04:
I can give you option to set hardware interlock based on power measurement calibration and you will see how accurate it is

hartytp, 2018-09-04:
Perfect! Give me firmware that does that and I'll test it.

If the hardware interlock isn't accurate enough then we can fix it in the next revision by using an external comparator instead of the comparator in the DAC. That way, the hardware interlock should be as accurate as the SW interlock, shouldn't it?

hartytp, 2018-09-04:
One other thing I need to do is to check how fast the software interlock is. Currently, the SCPI interface seems very slow (about 200ms to read a status register). Hopefully, that's just the SCPI interface and not an indication of how long a device measurement cycle takes. However, if the software interlocks only update once per 200ms then I think it's a real problem -- that's enough time that sensitive loads could quite easily be damaged by having excessive power applied.

hartytp, 2018-09-05:
I've measured the response time for the current software output power interlock. It varies between ?ms and ?ms.

1. Why is that so slow?? Even for a software interlock, that seems extremely slow. When we previously discussed this, I thought we agreed that the software interlocks would be much faster than that
2. I do not believe that the software interlock is fast enough to protect sensitive loads from damage. As such, it seems that the hardware interlock is essential -- even if it requires hardware modifications
3. We should check this with Maciej, but is this software interlock fast enough to protect the amplifier from excessive reverse power? I suspect that it could be damaged within 200ms. Do we need to add a hardware interlock for the reverse power as well?

hartytp, 2018-09-05:
Edit: added numbers to the previous post -- yes, that's a 0.5 second response time we've observed in some cases!

Note that this is for a CW signal. As noted elsewhere, the interlock does not seem to work properly for pulsed signals (which will be our main use case).

wizath, 2018-09-05:
That's because ADC gather 500 samples and then averages them to one. That was very old information and it just worked, because nobody seemed to care about that. At this moment it takes about 500 ms for one measurement - we're sampling at 1 ms intervals. Can change it to whatever suits you - just two defines to make it quicker.

hartytp, 2018-09-05:
Okay, good!

What's the RMS noise on each measurement? How many measurements do we need to average to get an accuracy of approx 0.25dB?

hartytp, 2018-09-05:
Having said that, I'd still like to hear from Maciej (or someone else who's thought about the RF) about how fast the reverse power interlock needs to be. Depending on what the damage mechanism is, I can imagine a few ms still being quite slow (e.g. if damage is caused by breakdown in the power FET it will be quite fast).Are we *sure* that a software interlock is fast enough for this?

hartytp, 2018-09-05:
Also, given the duty-cycle issues we've seen, I still think that a fast-responding hardware interlock would be vastly preferable to the software interlock for protecting sensitive loads...

wizath, 2018-09-05:
The problem with hardware interlock is that voltage at threshold pin (VIN−) is not the same as output from power detector (VRMS). Since the VIN characteristics are quite linear - we could try to do two-point calibration for output power hardware interlock and then set it's value accordingly.

[image]

hartytp, 2018-09-05:
> The problem with hardware interlock is that voltage at threshold pin (VIN−) is not the same as output from power detector (VRMS). Since the VIN characteristics are quite linear - we could try to do two-point calibration for output power hardware interlock and then set it's value accordingly.

Yes, I think we need to do two separate 2-point calibrations for each power detector: one for the analog output and one for the threshold pin. (For some reason, I thought that we were already doing that, but I must have misremembered something)

Can you send me firmware which implements that calibration routine, please and I'll give it a go. Thanks!!
Well, either that or add an external comparator in the next revision.

hartytp, 2018-09-05:
Anyway, once the code is written, it's not any more work to calibrate both the threshold and Vrms output at manufacture time than to only calibrate one of them...

wizath, 2018-09-06:
Another issue with setting the HW Interlock is that ATM threshold for 36 dBm at output interlock is something like 0.2V - so we're using very little of DAC resolution to control the interlock. 

[image]

hartytp, 2018-09-06:
@wizath good catch!

Presumably that's something we can fix with a suitable potential divider (and maybe an OpAmp buffer as well)?

hartytp, 2018-09-06:
At the moment, 2dB is something like 25 LSB on the DAC. That's not ideal, since we probably won't be able to get 0.1dB resolution. But it will be okay for the prototypes so long as we improve it for the next hardware revision.

hartytp, 2018-09-06:
@gkasprow what do you think about adding a 10:1 potential divider and an OpAmp buffer to the circuit? I can easily add this to the supplemental PCB in each RF module. The divider does not need to be more accurate than a standard 1% resistor since we will calibrate it after production anyway.

If you think that's a good idea, I can add it to all of our RF modules. However, I don't know if I have fully up to date schematics for the RF part of the circuit, so you will need to tell me what to solder where...

gkasprow, 2018-09-06:
It needs to be low value so 1uA of input current won't affect it much. On the other hand DAC has limited output current. If you add opamp, make sure that its offset is low. It's quite possible that we have to supply it also from negative source because even rail-to-rail opamps don't work well with very low output voltages.

hartytp, 2018-09-06:
@gkasprow @wizath AFAICT the firmware doesn't need to know whether we have an OpAmp + divider or not. So, how about this:
- give me a version of the firmware that has the hardware interlock in and I will test it
- if we decide that the interlock resolution/accuracy is not sufficient then I will find a suitable OpAmp + divider and add them onto our boards, and recalibrate and test the interlock

How does that sound?

hartytp, 2018-09-06:
Looking at this a bit more closely:
- 5W output power corresponds to approx 7dBm at the power detector. So, assume that the range that  the interlock needs to work over is something like -17dBm to +7dBm. Using the comparator built into the power detector will be a bit tough as the input voltage is only approx 10mV and, as you say, that's tough for an OpAmp to produce without an additional negative supply rail
- Looking again at the datasheet for the ADL5904, it seems that the Vrms output is much nicer than the Vin input -- it is still 200mV for a -30dBm power. So, I think that if we have to modify the hardware then it will be easier to add an external comparator than to add an OpAmp. 

I can easily find a cheap comparator that will work at this voltage range in a nice package like SOT23-5. I will add the comparator to the supplemental control boards, and also add a potential divider on the DAC output to improve the resolution (as well as a filter comparator). I will cut the trace to disconnect the power detector's threshold input.

I think that's the best solution. What do you think @gkasprow?

hartytp, 2018-09-06:
@gkasprow anyway, AFAICT all three choices (use the current hardware, add an OpAmp, or add a comparator) can use the same firmware. So I think the best approach is for @wizath to produce the firmware and then I can have a play with different approaches.

gkasprow, 2018-09-06:
I updated releases and sources. All files are in place.

hartytp, 2018-09-07:
Okay, lets do that. I'll open a new issue to discuss the comparator.

Since the comparator will use the same DAC as the power detector threshold input, it can use the same firmware. So, the most important thing for now is for @wizath to implement the hardware interlock firmware so that we can begin testing.

wizath, 2018-09-07:
You mean that you just want ability to change HW Interlock levels? I'll send that to you in a moment.

hartytp, 2018-09-07:
@wizath yes, that's the most important thing. Ideally, I'd like some code to allow me to calibrate the HW interlock levels as well, but that's less important...

hartytp, 2018-09-07:
Great, thanks.

A couple of questions:
- To check I understand, this command **only** adjusts the output power interlock DAC, it does not alter the software interlock. 
- This command is not yet accessible via the ethernet interface, only the VCP
- There is no automatic calibration routine yet for the interlock

Is that all correct?

wizath, 2018-09-07:
Yes, and it's valid only until power-cycle, since it's not stored in eeprom yet.

wizath, 2018-09-07:
That's just for testing if that range is enough. If it's sufficient, I'll change the rest

hartytp, 2018-09-07:
Thanks. We'll start testing now and get back to you with results.

Jochenwolf3, 2018-09-07:
Thank you for the new firmware. However, I am still unable to test the interlock. I did the following:

1. Sent 'int 1 4095' over USB.
2. Measured amplifier gain. At this point, the interlock LED is *not* indicating a tripped interlock. However, the gain is ~3 instead of ~40, i.e. the interlock has tripped.
3. Sent 'start' over USB. This then somehow updates the interlock LED on the front panel.

Both sending 'int 1 0' and 'int 1 65535'  results in the same behaviour. After power cycling the amplifier, its behaviour goes back to normal. 
Here is the USB readout after setting it to 'int 0 65535':
[image]
What do SON and OINT stand for? 
What should I do differently to set the hardware interlock and test it?

wizath, 2018-09-07:
First: 4095 is way too much for output interlock. I'm starting calibration from about 1500's and decrease value till interlock is tripped. That DAC is 12 bit only so 65535 should get 0V, so interlock will trip instantly.

If you type stop that will stop interlock task (leds updating - need to remove that). You have to type start each time. Will fix that in next release

gkasprow, 2018-09-07:
Are U sure that firmware was loaded correctly? Is there any FW ID implemented?

hartytp, 2018-09-10:
> Are U sure that firmware was loaded correctly? Is there any FW ID implemented?

Good question. @wizath is the firmware version included in the response to `*IDN?`?

hartytp, 2018-09-10:
Looking at the power detector data sheet a bit more carefully, I see another issue with the internal comparator besides it having a nasty voltage range: the internal comparator taps the signal off before the RMS conversion. That means that it does not have a well-defined mV/dB slope, but rather some power law slope that must be empirically determined on each unit. AFAICT, the comparator is really only designed to be operated at a single fixed level, set using the Vcal voltage. That doesn't match our use case well at all. So, while we probably can use it, I suspect that it will be easier to just use an external comparator on the Vrms output.

Let's aim to do that going forwards. We'll add the comparator to the supplemental control boards for the next revision and, since those boards are cheap to produce, also fit new supplemental control boards to all existing units.

wizath, 2018-09-10:
> Good question. @wizath is the firmware version included in the response to *IDN??

Added, `WUT,RFPA Booster,1.1,Sep 10 2018 12:00:35`

hartytp, 2018-09-10:
@wizath So, let's assume that fit external comparators to all existing Booster units. That will take a few weeks because of PCB lead time.

As a stop gap, we will use the internal comparators and get the users to do a fixed-level calibration. Please can you
- [ ] add output power interlock setting to the firmware via ethernet
- [ ] ensure that the output power interlock set point is saved to/loaded from EEPROM

That will be sufficient until the new boards are ready.

hartytp, 2018-09-10:
@wizath I'd like implement the interlock asap, so please let me know once you've implemented those changes so I can test them.

Thanks!

Jochenwolf3, 2018-09-10:
[image]
In the first plot above, you can see the power at which the interlock trips vs the corresponding DAC value in machine units, as well as a fit of that data. The second graph shows the difference between the two lines of the first graph.
For the fit, I have just done a two point calibration on the interlock, at 25 (=15.91 dBm) and 225 (=34.6 dBm) DAC mu. The data is fitted between those two points using `power [dBm] = 8.506 * log(DAC_value [mu])- 11.47`.
Considering my chosen resolution for determining when the interlock trips was only 0.1 dB, this fits beautifully. 
Below 16 dBm, the errors get bigger, which is fine for our use case.

hartytp, 2018-09-10:
@wizath NB I do not expect any users to use Booster below about 100mW (why would you?). So, to maximize calibration accuracy, let's always use 20dBm as the lower power level for our 2-point calibration...

Jochenwolf3, 2018-09-10:
Taking the `power [dBm] = 8.506 * ln(DAC_value [mu])- 11.47` formula of the above graph, I can calculate the resolution of the hardware interlock, i.e. how much the interlock tripping power changes upon changing the DAC value by 1:

- At 0.1 W, the resolution is 0.2 dB / mu
- At 1.0 W, the resolution is 0.07 dB / mu
- At 5.0 W, the resolution is 0.03 dB / mu

This resolution is good enough for our use case.

wizath, 2018-09-10:
@hartytp 
AFAICT by looking at my calibration sheet, interlock values differs slightly (266 - 231 @ 36 dBm, based on 25~ modules). If the accuracy is OK, we can skip the calibration part, and use @Jochenwolf3 equation for extra simplicity.

hartytp, 2018-09-10:
I don't think that will work. At 1W (which is roughly what we will run at) the slope is approx 0.1dB/LSB, so a 30 LSB change (260-230) is approx 3dB. So, I can see two options:
1. Implement a full 2-point calibration. The interlock threshold is then given by: `DAC_LSB = 10^((P_dBm + A)/B)` where A and B are floating point numbers to be determined by the calibration. 
2. Don't bother with any calibration for now. We'll implement calibration once we have the comparator. Just add a firmware command to set the raw DAC value and let users determine the correct DAC value for their chosen interlock setting.

I'd marginally prefer 1, since this will make Booster fully operational while we wait for the next hardware version. But, if that will be tricky, going for 2 is fine.

wizath, 2018-09-10:
Ok, I almost finished point 1

Jochenwolf3, 2018-09-10:
[image]
Here is the frequency dependence of the hardware interlock. Still using the same calibrated channel from above, I've set the DAC value to 131 = 30 dBm. There is a fixed offset of ~2 dB which is to do with cabling, and not important.
The main point is that over 50 - 500 MHz, the interlock tripping power changes by 2 dB, which is acceptible for our use case.

Jochenwolf3, 2018-09-10:
[image]
Looking at the transient response of the hardware interlock, above is a typical response. The yellow line is an external RF switch turning booster's input of -17 dBm at 200 MHz on at 0 ns. At about 50 ns, I'm measuring an output on my oscilloscope, and at about 100 ns, the interlock starts tripping. After another ~100 ns, the interlock is done tripping. Two things of note:
1. The overall time to trip ~150 ns. 
2. There is a large overshoot during the tripping for ~100 ns.

Repeating the above experiment ~20 times by hand, it triggered 20/20 times when 2 dB away from tripping setpoint (-19 dBm input); however, if I am only 1 dB away from the tripping setpoint, it occasionally only trips after ~1 us. This is presumably due to some noise on the input and is OK for us.

wizath, 2018-09-10:
> @wizath what's the eta for that?

Tomorrow, about lunchtime

Jochenwolf3, 2018-09-10:
[image]
Looking for glitches when resetting the interlock, above is the response when RF is applied (-20 dBm, 200 MHz), and a previously tripped interlock is reset. There is an initial analogous spike due to switching for 100 ns, then it takes another ~600 ns for the output to turn on. As this is the same size of spike as the interlock tripping, this is also not an issue for us.

hartytp, 2018-09-10:
> Tomorrow, about lunchtime

:+1: Thanks!

wizath, 2018-09-11:
[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/2370695/eth-scpi.dfu.zip)

Calibration procedure:
- Enable channel, set 20 dBm
- type `intcal 5 1 20` into CLI
- wait for operation to complete, output `Got 1-point cal val 40 pwr 20`
- Re-enable channel, set 36~ dBm
- type `intcal 5 2 36` into CLI
- wait, expected output `Got 1-point cal val 240 pwr 36`

After complete, you should see output simillar to `Got power factors 12 -200`. After calibrating, you can use command `intv <channel> <power level>`, e.g `intv 5 36.0` to set value using calculated values. Output should be like `Calculated value for pwr 35.00 = 220`

hartytp, 2018-09-11:
Amazing, thanks!

hartytp, 2018-09-11:
PS @wizath I see that you implemented a much more efficient method for calibrating the power interlock. It works really well :)

hartytp, 2018-09-11:
@wizath is this command accessible via the SCPI/ethernet interface yet? Also, is the calibration + threshold saved to EEPROM?

Jochenwolf3, 2018-09-11:
Thanks, I'm using the method now.
- Thank you for using the different stepsizes in the automatic routine for finding the DAC values, it makes it much faster!
- The LEDs on the front panel turn off after running intcal. Only after pressing standby two times do they come back on. This is slightly odd, but OK.
- I have just calibrated one of my channels with this method. It found the DAC values of 35 (=20 dBm) and 236 (=36 dBm), and gave me power factors of 9 & -145 after the first point, and 12 & -205 after the second point; all of which is in line with what I found manually. However, when setting the values to e.g. `intv 1 30`, it replies `Calculated value for pwr 30.00 = 0`. It therefore seems like there is an arithmetic error inside the `intv `function. Can you confirm?

wizath, 2018-09-11:
> is this command accessible via the SCPI/ethernet interface yet? 

Try using `int:hpow <channel> <power level>`, calculated result is printed on CLI ATM, but command works fine.

> Also, is the calibration + threshold saved to EEPROM?

Calibration values are saved in EEPROM, I'll make threshold to save in a moment.

> it replies Calculated value for pwr 30.00 = 0. It therefore seems like there is an arithmetic error inside the intvfunction. Can you confirm?

I'll check it in a moment. Have you tried other values? If you type start can you see cal values listed (fields `HWIS` and `HWIO` as for scale and offset)

Jochenwolf3, 2018-09-11:
There is the problem, the calibration is not saved; both `HWIS `and `HWIO `are -1, even after having just calibrated them.

wizath, 2018-09-11:
Please reboot your device. I'm saving values to EEPROM, but not updating them in memory. I'll fix that in a second.

wizath, 2018-09-11:
[eth-scpi.dfu.zip](https://github.com/sinara-hw/Booster/files/2371270/eth-scpi.dfu.zip)
Fixed updating calibration values in memory

Jochenwolf3, 2018-09-11:
@wizath Thanks, that fixed it. I have now calibrated the hardware interlock on all channels. `HWIS `is always 12, `HWIO `ranges from -187 to -204 between channels

hartytp, 2018-09-11:
@wizath We'll post details tomorrow, but that works really nicely, thanks!

One question: what do we do about the software power interlock? I can see two possibilities:

1. Scrap the software interlock:
- Just comment out the code for the software interlock so it's there in case we need it later
- rename `int:hpow' to 'int:pow'
2. Keep both interlocks
- Users then have access to both a slow software power interlock and a high-speed hardware interlock
- if an interlock is not needed then it can be programmed to the maximum value (something like 38dBm)
- keep both 'int:pow' and 'int:hpow' commands
- update the wiki to make it clear that we have two different interlocks and to explain their differences

Personally, I don't think that the software interlock is very useful now that we have the hardware interlock. I think that having both is too confusing for the users, and writing really good documentation to explain how/why to use them will take time. So, I'd prefer to take the first option and disable the software interlock. What do you think @wizath?

Jochenwolf3, 2018-09-12:
[image]
On a calibrated channel, I have set its hardware interlock power level to 30 dBm, i.e. `INT:HPOW 2 30`. Sweeping the frequency, the interlock trips at values from 31 to 33.
This accuracy is quite a bit worse than I got via manual calibration (https://github.com/sinara-hw/Booster/issues/73#issuecomment-419865910). 
@wizath What is the equation you are using to calculate the values? Is there any additional accuracy you can achieve via e.g. adding another significant digit to the calibration?

wizath, 2018-09-12:
I'll revise that in a second

[23 more replies]

closed 2018-09-12 by hartytp