sinara-hw/Booster #74: ERROR

hartytp, 2018-09-04:
@wizath The ERROR LEDs have just come on for channels 6 + 7. Any idea why?

Here is the UART output:

https://hastebin.com/fuluvokoza.shell

hartytp, 2018-09-04:
Pressing interlock reset clears the red LEDs. Here is a UART output after clearing...

https://hastebin.com/uqitahiwum.shell

wizath, 2018-09-04:
Interesting, but since the channels were still running I have no idea. Whole set of checks that could lit the ERROR led are on the FW version that is not yet on your device. Only thing is the temperature interlock, but it should disable the channels (and the temperatures were ok).

hartytp, 2018-09-04:
Interesting. Maybe some I2C glitch that caused the temperature interlock to trip?

hartytp, 2018-09-04:
Do you record what caused the ERROR status in firmware (i.e. latch it until the error is cleared)?

hartytp, 2018-09-04:
That would be a very good diagnostic

wizath, 2018-09-04:
ATM each error is latched until reboot / direct CLI command

hartytp, 2018-09-04:
Is the source of the error latched? i.e. can I tell if the red LED came on because of temperature or some other reason? Do you think this was just some I2C glitch?

hartytp, 2018-09-04:
@wizath am I right in thinking that there is no information in those UART traces that can help us identify what caused this error?

If so, I'll close this issue as the amplifier still works fine and I haven't seen this error condition happen again; most likely it was a glitch in the I2C bus, but we can't be sure. I'll reopen this issue if I see this again.

closed 2018-09-04 by hartytp

hartytp, 2018-09-06:
I left my Booster running overnight to check for any other I2C issues or strange glitches. But, I was greeted with a bank of green lights, so all seems fine...