sinara-hw/Booster #76: I2C review

hartytp, 2018-09-07:
@gkasprow @wizath before we begin production of Booster v1.2 please can you do a complete review of the I2C chips, including the temperature sensors?

The reason I ask for this is that a couple of times, we've seen issues which may relate to the I2C chips. For example, once the over temperature alarm tripped when the chassis was only at around 30C. I suspect that this was due to an incorrect reading from the temperature sensor (could also have been a firmware bug). I haven't been able to reproduce that issue since, so it's hard to determine what caused it.

I think we should check:
- Voltage ratings for all pins for the I2C chips, in particular the temperature sensor
- bus/converter timings
- decoupling
- load impedance (including capacitors and any filters/ferrites)
- probably also worth using a scope to probe the bus to check signal rise times and any potential interference
- any grounding issues
- etc

hartytp, 2018-09-13:
One of the things that made our mechanical issues difficult today is that once one channel has an I2C issue the whole bus goes down. So one has no easy way of identifying where the fault occurs. Is there any way we can add some better diagnostics here? Some way so that when a fault occurs we at least know which channel it occurred on.

gkasprow, 2018-09-14:
The i2c switch should be reset before accessing another channel. It was an
issue with rev1 but was fixed lafer on.

hartytp, 2018-09-14:
> The i2c switch should be reset before accessing another channel. 

That sounds like a good idea. At least that way we will know which unit had the problem.

> It was an issue with rev1 but was fixed lafer on.

What was the issue with rev1? Was the I2C switch reset not connected to the microprocessor? Is it something we can fix with an air wire?

gkasprow, 2018-09-14:
It was not connected but I fixed it. And AFAIK Jakub is reserting it with
firmwarr so I'm surprised that the bus gets locked.

wizath, 2018-09-14:
Well, that's the weird one - I'm resetting I2C switch every time I switch channels. But most concerning issue is that during temperature tests whole I2C bus died. As I was checking my protection for false readings, I can see that even if one channel fails, rest continue to work properly.

[image]

Readings from channel 5 are off the charts, so error led is lit and channel is disabled until power-cycle / CPU reset. But the remaining channels are up and running without any issues.

hartytp, 2018-09-14:
hmmm...so any ideas what could have caused the symptoms we see.

@jochenwolf3 noticed that the red LEDs on the main board did not come on for some channels when we were experiencing problems. Not sure if that's useful information?

wizath, 2018-09-14:
> noticed that the red LEDs on the main board did not come on for some channels when we were experiencing problems. Not sure if that's useful information?

Yes, I fixed that

hartytp, 2018-09-14:
I mean the SMT ones by each channel, not the ones on the FP.

hartytp, 2018-09-14:
@gkasprow any thoughts about what could cause this kind of lockup or how we debug it?

hartytp, 2018-09-15:
@gkasprow do you think this could be an issue with something else, like a supply rail or some control line being shorted/disconnected? What else could cause these symptoms?

gkasprow, 2018-09-15:
To check it, one needs to verify if enable signal was on. There are 3 such LEDs on rach channel, which LEDs do you mean ?

gkasprow, 2018-09-15:
The only red LED is connected to the output of the voltage/current
supervisor chip. The chip works autonomously and measures currents and
voltages. When programmed thresholds are exceeded, it changes INT pin level.
When the LED is off it means that current thresholds were exceeded. We
should probably change its polarity so the red LED is lit when sth wrong
happns...
@wizath what levels did U program and what action CPU does when ALERT
signal is present? Maybe the levels are too small.

gkasprow, 2018-09-15:
@hartytp are the three remaining green LEDs on on the main board for
faulty channels?

On 15 September 2018 at 21:48, Grzegorz Kasprowicz <kasprowg@gmail.com> wrote:
> The only red LED is connected to the output of the voltage/current
> supervisor chip. The chip works autonomously and measures currents and
> voltages. When programmed thresholds are exceeded, it [...]

hartytp, 2018-09-15:
> There are 3 such LEDs on rach channel, which LEDs do you mean ?
> The only red LED is connected to the output of the voltage/current supervisor chip.

Right: AFAICT, there is only 1 LED and that was the one that seemed to go off on some channels when we had this fault (no channels produce any readings on the I2C bus).

> @wizath what levels did U program and what action CPU does when ALERT signal is present? Maybe the levels are too small.

Maybe, but I don't think that explains it. This issue only seems to be mechanical since it was fixed by taking all the modules out and then reinserting it. After fixing the issue, it appeared again once we tightened up the screws. So, if a current limit was exceeded it sounds more like something being shorted.

> @hartytp are the three remaining green LEDs on on the main board for faulty channels?

I'm afraid I don't know. We didn't do too much debugging on this issue -- all the Boosters we have are needed urgently in our experiments so we don't have a spare one to do any debugging on. 

I wonder if this is related to the issue we had on the other Booster v1.1 that we have, where one channel was only detected by the firmware when the lid was placed on it.

hartytp, 2018-09-25:
@wizath did you have any luck figuring out what was going on here?

gkasprow, 2018-10-20:
Jakub was able to recreate I2C hanging with one of the lines shorted. He added periodic I2C core reset

closed 2018-10-20 by gkasprow