sinara-hw/Booster #77: Hardware interlock

hartytp, 2018-09-07:
Context:
- We want a hardware interlock for the output power in Booster instead of just having a software interlock
- the hardware interlock will be much faster responding than a software interlock, making it better at protecting sensitive loads. Let's aim for a response time of <100us as an absolute maximum. Ideally, let's go for < 10us
- the hardware interlock also means that Booster will still protect sensitive loads in case the firmware crashes
- currently, the hardware interlock is provided using the [ADL5904](http://www.analog.com/media/en/technical-documentation/data-sheets/adl5904.pdf) power detector's internal comparator. That's annoying because the required threshold voltage is quite low and the threshold input requires quite a bit of current, making it inconvenient to drivewithout a negative supply rail.

So, let's see if we can do this instead using the power detector's analog output and an external comparator. I'll post some details in this issue

hartytp, 2018-09-07:
First question: how fast is the comparator's analog output.
- Depends on the filter capacitor. The plot below is for a 900MHz carrier with 20MHz of signal BW
- we currently use 100nF, so the RMS output 10%-90% rise time is a couple of hundred us, which is too slow
- *My feeling is that the hardware interlock should be a fast interlock on the signal envelope*. There can be an additional (slower) software interlock that responds only to the average power. So, the "noise" shown in this plot isn't really noise at all, it's just showing us fast oscillations in the peak signal power, which is what we want to see (again, users who don't care about that can use the software interlock only).
- [ ] I'd suggest that we decrease Crms to something like 1nF for the next revision

[image]

hartytp, 2018-09-07:
In fact, I think I like the idea of a slow software interlock with lots of averaging and a fast hardware interlock with a short time constant. That gives the users more flexibility in how they configure the interlock

hartytp, 2018-09-07:
Next there is a gain OpAmp to amplify the Vrms signal by 2 to match it to the DAC/ADC range better. OpAmp is [LMV321M5X](http://www.ti.com/lit/ds/symlink/lmv358-n-q1.pdf)
- For G=1, the OpAmp responds to 2V in a couple of us. For G=2 it will be somewhat slower, but should still respond in <10us.
- Might be worth considering using a slightly faster OpAmp, but this isn't crucial. I will leave this decision to @gkasprow

hartytp, 2018-09-07:
Finally, on the supplemental control board:
- [ ] Cut the traces connecting the input / output overdrive signals to J6 + J9
- [ ] also cut the traces for the IN_VREF and FR_VREF signals 
- [ ] add 2x external comparator to the PCB. Doesn't need to be anything special here. As the voltage range is quite civilized (see below, but don't forget the factor of two gain from the OpAmp). NB need to make sure that the Vrms voltage can never exceed the max output voltage supplied by the DAC. Otherwise, we need to add a potential divider at the comparator input (or just remove the gain)
- [ ] connect the in_VREF, FR_VREF, input_overdrive and output_overdrive signals to the comparator

That's it!

@gkasprow please can you suggest a suitable low-cost comparator to use?
- need to check the max input voltage is okay and also that the DACs can supply the correct voltage
range
- need to check that the output voltage is compatible with the interlock logic gates (or use an open-drain output, but then be careful to check the speed of the comparator response!)
- check response time and supply rails

[image]

hartytp, 2018-09-07:
@gkasprow please can you review the above proposal and check that it will work? Once you've reviewed it, I will implement it in hardware on one channel here and test.

Please also review the other issues I have raised related to the power detector

Once this has been tested and we are happy with it, I would like to apply this fix to all current hardware we have (24 channels). Easiest way will be to ask TechnoSystem to create a new version of the supplemental control board with the fixes applied.

hartytp, 2018-09-10:
@gkasprow okay, I'll order those parts. In the next revision, let's add pads for 0R/DNF resistors so that we can easily switch between using the internal comparator and an external one.

closed 2018-09-12 by hartytp

hartytp, 2018-09-13:
@wizath with the new firmware this works perfectly, thank you!
- calibrated at 20dBm and 36dBm
- set threshold to 30dBm
- interlock threshold reliably trips at 29.6dBm so within 10%, which is absolutely fine for a safety feature
- verified that the interlock is retained over device power cycling
- added calibration instructions to the wiki

hartytp, 2018-09-13:
- testing the accuracy of the hw interlock
- determine the threshold where the interlock trips by increasing the input power in 0.1dB steps until Booster reports it has tripped via the SCPI interface
- measuring the output power that booster tripped at using an external power meter (V3500A + 30dB attenuator)
- this is all done at 200MHz, which is the same frequency I calibrated the interlock at and using the same cabling, power meter, attenuator, etc so I expect very good accuracy

[image]

hartytp, 2018-09-13:
So, that works beautifully!

Jochenwolf3, 2018-09-13:
NB IIRC the DAC resolution is equivalent to approx 0.3dB at 20dBm (the resolution decreases with decreasing setpoints)