sinara-hw/Booster #79: power detector RF termination

hartytp, 2018-09-07:
why aren't we using a shunt termination resistor on the power detector as suggested in the data sheet?

hartytp, 2018-09-12:
NB if we change this we will change the signal level at the Vrms pin. Need to check that won't cause any issues (e.g. due to smaller signal at the ADC input)

gkasprow, 2018-10-19:
The termination is there.
But the value indeed is wrong - should be 85Ohm. Now the discrete attenuator has open output.
For input coupler I changed it to 2x169R
For output coupler there is already discrete 20dB attenuator. Formally it needs 50R load.
so one needs to add 85R to keep the load impedance of the coupler = 50R.
So the R13 and R16 need to be 62R || 85R = 36R
To keep similar power level we can change attenuation to 15dB which gives values 71, 71 , 136R.
To have 50R load, we add 85R in parallel to 71R resistor which gives 39R.
Then the dividers would have 71R, 137R 39R and maintain 50R input impedance.

closed 2018-10-19 by gkasprow