sinara-hw/Booster #80: Vrms load capacitor

hartytp, 2018-09-07:
The data sheet instructs us to place a 10nF load capacitor on the Vrms output pin. We did not do that (the load is 4k7). Why not?

Note that the data sheet does not specify the purpose of this capacitor (just a filter?) and it does not specify the output impedance of the Vrms pin so it is unclear what the implications are for the circuit bandwidth. I will ask on the analog devices forum and report back.

hartytp, 2018-09-10:
My guess is that we don't want to add a large capacitor here since it will kill the detector's bandwidth. But, I'll see what ADI say about it...

hartytp, 2018-09-10:
@gkasprow here is what ADI say:

```
The Vrms pin has low output impedance.  Typically, less than 200 ohm depending on the frequency.

The suggested 10nF capacitor (reference C14 on eval. Board) at the Vrms pin is the filtering capacitor.  This capacitor can be filtering out any RF signal.  This capacitor does not affect significantly on the Vrms's pin timing, but C12 does.  The C12 is located between DECL and CRMS pins.
```

- [ ] Let's add this capacitor for the next design revision
- assuming 200R output impedance, 10nF gives a 2us rise time. This seems a little on the high side. Unless the OpAmp is going to limit the speed to something much slower than this, I'd be tempted to go for something more like 4n7 or even 1nF.

gkasprow, 2018-10-19:
changed RC at the VRMS output to 0R and 4n7

closed 2018-10-19 by gkasprow