sinara-hw/Booster #81: v1.1 errata

hartytp, 2018-09-07:
This is just a tracking issue for me to keep a list of fixes that I will apply to our v1.1 boards:
- Fix coax cable in place
- reduce power detector crms to 1nF
- populate DECL resistors
...

gkasprow, 2018-09-11:
- align D-sub connector on the main board
- add EUI-48 chip instead of IC20

wizath, 2018-09-18:
- Add I2C testpoints
- Add PGOOD testpoint

wizath, 2018-09-24:
- Add hole on supplementary board for trimpot

hartytp, 2018-09-24:
> Add hole on supplementary board for trimpot

Why do we need this trimpot? Can't we just due a DAC with good enough resolution to provide the bias voltage directly?

I'd guess that it's cheaper to pay for a 16-bit or even 18-bit I2C DAC than to pay a production engineer to adjust the trim pot. And, it's certainly more continent in case the user want's to adjust the bias current later on, since without the trimpot they can do that through software

wizath, 2018-10-02:
- Add better method to adjust negative voltage for bias current calibration

hartytp, 2018-11-05:
@gkasprow Crms for the three power detectors still has not been changed (would make the interlock extremely slow!). Have the other items here been fixed?

hartytp, 2019-01-07:
@gkasprow please can you confirm that all issues here were fixed in the latest design revision and then close this issue? Thanks!

hartytp, 2019-01-31:
@gkasprow can you confirm that these issues were all fixed in the new release and close this issue? Thanks!

gkasprow, 2019-01-31:
I will close it once I make sure the cable is properly fixed. Other items were fixed.

hartytp, 2019-01-31:
Thanks for clarifying. I'll open a specific issue for that one thing then so that it's easier to keep track of.

closed 2019-01-31 by hartytp