sinara-hw/Booster #82: Vrms OpAmp

hartytp, 2018-09-07:
Max power reaching the detector should be 8dBm. That gives a Vrms of close to 1.6V. We have an OpAmp with a gain of two after the power detector that's running from a 3V3 supply rail. Isn't that going to get a bit marginal and cause issues? Should we change the supply voltage/gain to give more margin?

Also, should we have a ferrite bead here to keep RF from getting onto the supply rail?

hartytp, 2018-09-07:
Is there any reason not to change the regulator on the supplemental control board to provide more margin here (needs someone to check what else is powered from that rail and check what voltages they can take)...

hartytp, 2018-09-10:
@gkasprow yes, that's always a good aim! A current limiting resistor also does the trick if that's easier than driving the OpAmp right to its supply rail.

- [ ] Please can you review the signal level at the OpAmp and check that there is enough headroom between the max signal level when the amp is running at 5W and the supply rail for the OpAmp and ADC to function properly? If this looks marginal then let's consider reducing the OpAmp gain a bit
- [ ] Please consider using a slightly faster OpAmp for the next revision to ensure that this doesn't limit the speed of the interlock too much
- [ ] let's also make sure we add a ferrite bead to the OpAmp supply pin to prevent RF from getting onto the supply rail (if we reduce the power detector Crms then there will be more RF reaching the amp)
Thanks

hartytp, 2018-09-10:
- [ ] also, let's double check that this OpAmp is happy driving the capacitance of the long PCB traces + ADC input. If not, it might be worth adding a small isolation resistor on its output.

gkasprow, 2018-10-19:
At the moment we use opamp with gain = 1. I can replace existing one (BW=1MHz, 1V/us) with LT6220 (70MHz, 20V/us)

gkasprow, 2018-10-19:
To make sure the trace capacitance won't be an issue, I will place 22R at the output.

closed 2018-10-19 by gkasprow