sinara-hw/Booster #83: reflected power hardware interlock

hartytp, 2018-09-07:
I'd like to hear an opinion from an RF specialist about what kind of reverse power interlock is needed to protect Booster. is the current software interlock (averaging time >100ms) enough? For example, this means that pulses of high reverse power will not trip the interlock.

If this isn't enough then we can easily add an extra DAC and comparator to the supplemental control board in the next revision. These boards are cheap, so it's easy to replace them in the existing booster units.

gkasprow, 2018-10-20:
We kept the amp output open with full power for minutes. So 100ms is sufficient to protect it.
However we can later on add DAC, comparator and DFF tot he suppl. board.

closed 2018-10-20 by gkasprow

hartytp, 2018-10-20:
> We kept the amp output open with full power for minutes

Okay, if you're convinced that the software interlock is sufficient to prevent damage to the amplifier then I'm happy.

Please make sure that the threshold for the software intlock is documented somewhere (e.g. Wiki).

> So 100ms is sufficient to protect it.

Presumably the software interlock can be configured to respond in more like 1ms without issue?

> However we can later on add DAC, comparator and DFF tot he suppl. board.

I'd like this to be the final iteration of the design if possible. If we need the DAC + comparator + DFF then let's add it now. If we don't need them then let's forget about it. Either way, we won't add them "later".

gkasprow, 2018-10-20:
OK, we can add necessary HW now. But CPU does not have any free line to get info that the interlock was triggered. It has only two lines: input overdrive, output overdrive. But one can set both of them high to tell CPU that reflected power intlock was activated.

hartytp, 2018-10-20:
> OK, we can add necessary HW now

Only if you think it's necessary (it sounds like you don't!). If you don't think it's necessary then let's not bother. I don't want to complicate the design and increase the cost without a good reason.

My point is just that we should make a final decision about what we will do for this interlock now, and do our best to freeze the design.

gkasprow, 2018-10-20:
OK, let's leave it as it is now.

hartytp, 2018-10-20:
Sounds good. Assuming the damage mechanism is thermal, and we can make the interlock trigger after 1ms or so, I think the FET will be able to handle it (the board does not run hot in normal operation thanks to the excellent fan cooling, so there is a decent amount of thermal headroom in the design).