sinara-hw/Booster #89: recieved in an unsatisfactory condition

bnichol, 2018-09-10:
When we received the booster the front panel was bent as shown in the photo
[image]

The lid had also been warped meaning that once it had been taken off it could not be put back on. There is a groove on the main body that a lip on the lid must slot into meaning that any slight deformation means the lid cannot be removed and put back on.

gkasprow, 2018-09-10:
Could it happen during transport?

hartytp, 2018-09-10:
I assume it must have.

hartytp, 2018-09-10:
Although, the box didn't look damaged so maybe not.

gkasprow, 2018-09-10:
But it looks like it happened after the lid was installed. So either during packing or shipping.

hartytp, 2018-09-10:
I assume it was one or the other.

I'd hope that TechnoSystems would have noticed if that had happened during packing, so I assume it was shipping. Looking more closely, there is some damage to the box, so maybe it was a shipping issue (or, insufficient cushioning in the packaging).

hartytp, 2018-09-11:
Photos showing minor damaging to the box this Booster was shipped in. 
[image]
[image]

hartytp, 2018-09-11:
we made this fit into one of our racks, so we won't return it. But, let's make sure that the packaging/shipping is done better next time!

closed 2018-09-11 by hartytp